jayofelony/pwnagotchi #3: [BUG] error installing in pi 02W

jagp8, 2024-01-12:
I Just put the image in an sd card and tried it with the default values and it starts well, but when launching pwnagotchi-launcher with the 2 options from /etc/pwnagotchi/config.toml uncommented (cand changing to waveshare_3) this error shows

'''
sudo pwnagotchi-launcher
INFO:root:[update] plugin loaded.
INFO:root:grid plugin loaded.
[2023-12-11 06:07:00,347] [INFO] [update] plugin loaded.
INFO:root:webcfg: Plugin loaded.
INFO:root:[Fix_Services] plugin loaded.
[2023-12-11 06:07:00,349] [INFO] grid plugin loaded.
INFO:root:Session-stats plugin loaded.
[2023-12-11 06:07:00,352] [INFO] webcfg: Plugin loaded.
[2023-12-11 06:07:00,352] [INFO] [Fix_Services] plugin loaded.
[2023-12-11 06:07:00,356] [INFO] Session-stats plugin loaded.
WARNING:root:ui.fps is 0, the display will only update for major changes
INFO:root:Got here
[2023-12-11 06:07:00,379] [WARNING] ui.fps is 0, the display will only update for major changes
INFO:root:initializing waveshare v3 display
[2023-12-11 06:07:00,380] [INFO] Got here
[2023-12-11 06:07:00,382] [INFO] initializing waveshare v3 display
Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 226, in <module>
    pwnagotchi_cli()
  File "/usr/local/bin/pwnagotchi", line 205, in pwnagotchi_cli
    display = Display(config=config, state={'name': '%s>' % pwnagotchi.name()})
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 18, in __init__
    self.init_display()
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 96, in init_display
    self._implementation.initialize()
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/waveshare3.py", line 37, in initialize
    from pwnagotchi.ui.hw.libs.waveshare.v3.epd2in13_V3 import EPD
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/v3/epd2in13_V3.py", line 32, in <module>
    from . import epdconfig
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/v3/epdconfig.py", line 148, in <module>
    implementation = JetsonNano()
                     ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/v3/epdconfig.py", line 107, in __init__
    raise RuntimeError('Cannot find sysfs_software_spi.so')
RuntimeError: Cannot find sysfs_software_spi.so

'''

jayofelony, 2024-01-12:
What are you doing with pwnagotchi-launcher, you just have to edit the config.toml and then restart the device or when ssh in type `pwnkill`

jagp8, 2024-01-12:
Yes, when I try to restart the pwnagotchi service it ends like this
---
● pwnagotchi.service - pwnagotchi Deep Reinforcement Learning instrumenting bettercap for WiFI pwning.
     Loaded: loaded (/etc/systemd/system/pwnagotchi.service; enabled; preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Mon 2023-12-11 06:14:56 CET; 23s ago
       Docs: https://pwnagotchi.ai
    Process: 970 ExecStart=/usr/bin/pwnagotchi-launcher (code=exited, status=1/FAILURE)
   Main PID: 970 (code=exited, status=1/FAILURE)
        CPU: 5.973s

Dec 11 06:14:56 pwnagotchi systemd[1]: pwnagotchi.service: Consumed 5.973s CPU time.
---

So I tried to execute the launcher to see where was it failing, that's why

jayofelony, 2024-01-12:
try:
`sudo pwnagotchi --debug`

jagp8, 2024-01-12:
It says quite the asme as the launcher

---
Kite@pwnagotchi:~ $ sudo pwnagotchi --debug
DEBUG:root:[FS] Daemonized...
DEBUG:root:[FS] Syncing /var/tmp/pwnagotchi -> /run/pwnagotchi/disk/pwnagotchi
[2023-12-11 06:37:19,096] [DEBUG] [FS] Syncing /var/tmp/pwnagotchi -> /run/pwnagotchi/disk/pwnagotchi
DEBUG:root:loading plugins from /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default - enabled: ['auto-update', 'fix_services', 'grid', 'session-stats', 'webcfg']
[2023-12-11 06:37:19,107] [DEBUG] loading plugins from /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default - enabled: ['auto-update', 'fix_services', 'grid', 'session-stats', 'webcfg']
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/auto-update.py
[2023-12-11 06:37:19,111] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/auto-update.py
DEBUG:root:loaded plugin auto-update as <auto-update.AutoUpdate object at 0x7f8c188610>
[2023-12-11 06:37:19,114] [DEBUG] loaded plugin auto-update as <auto-update.AutoUpdate object at 0x7f8c188610>
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/grid.py
[2023-12-11 06:37:19,117] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/grid.py
DEBUG:root:loaded plugin grid as <grid.Grid object at 0x7f8c189450>
[2023-12-11 06:37:19,120] [DEBUG] loaded plugin grid as <grid.Grid object at 0x7f8c189450>
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/webcfg.py
[2023-12-11 06:37:19,123] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/webcfg.py
DEBUG:root:loaded plugin webcfg as <webcfg.WebConfig object at 0x7f8c189ad0>
[2023-12-11 06:37:19,126] [DEBUG] loaded plugin webcfg as <webcfg.WebConfig object at 0x7f8c189ad0>
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/fix_services.py
[2023-12-11 06:37:19,128] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/fix_services.py
DEBUG:root:loaded plugin fix_services as <fix_services.FixServices object at 0x7f8c18abd0>
[2023-12-11 06:37:19,136] [DEBUG] loaded plugin fix_services as <fix_services.FixServices object at 0x7f8c18abd0>
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/session-stats.py
[2023-12-11 06:37:19,141] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/session-stats.py
DEBUG:root:loaded plugin session-stats as <session-stats.SessionStats object at 0x7f8c194650>
[2023-12-11 06:37:19,153] [DEBUG] loaded plugin session-stats as <session-stats.SessionStats object at 0x7f8c194650>
DEBUG:root:loading plugins from /usr/local/share/pwnagotchi/custom-plugins/ - enabled: ['auto-update', 'fix_services', 'grid', 'session-stats', 'webcfg']
[2023-12-11 06:37:19,156] [DEBUG] loading plugins from /usr/local/share/pwnagotchi/custom-plugins/ - enabled: ['auto-update', 'fix_services', 'grid', 'session-stats', 'webcfg']
INFO:root:grid plugin loaded.
[2023-12-11 06:37:19,173] [INFO] grid plugin loaded.
INFO:root:webcfg: Plugin loaded.
[2023-12-11 06:37:19,179] [INFO] webcfg: Plugin loaded.
INFO:root:[Fix_Services] plugin loaded.
INFO:root:[update] plugin loaded.
[2023-12-11 06:37:19,180] [INFO] [Fix_Services] plugin loaded.
INFO:root:Session-stats plugin loaded.
[2023-12-11 06:37:19,184] [INFO] [update] plugin loaded.
[2023-12-11 06:37:19,186] [INFO] Session-stats plugin loaded.
WARNING:root:ui.fps is 0, the display will only update for major changes
INFO:root:Got here
[2023-12-11 06:37:19,207] [WARNING] ui.fps is 0, the display will only update for major changes
INFO:root:initializing waveshare v3 display
[2023-12-11 06:37:19,208] [INFO] Got here
[2023-12-11 06:37:19,210] [INFO] initializing waveshare v3 display
Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 226, in <module>
    pwnagotchi_cli()
  File "/usr/local/bin/pwnagotchi", line 205, in pwnagotchi_cli
    display = Display(config=config, state={'name': '%s>' % pwnagotchi.name()})
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 18, in __init__
    self.init_display()
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 96, in init_display
    self._implementation.initialize()
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/waveshare3.py", line 37, in initialize
    from pwnagotchi.ui.hw.libs.waveshare.v3.epd2in13_V3 import EPD
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/v3/epd2in13_V3.py", line 32, in <module>
    from . import epdconfig
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/v3/epdconfig.py", line 148, in <module>
    implementation = JetsonNano()
                     ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/v3/epdconfig.py", line 107, in __init__
    raise RuntimeError('Cannot find sysfs_software_spi.so')
RuntimeError: Cannot find sysfs_software_spi.so

jagp8, 2024-01-12:
I've been doing some testing and the problem seems to be enabling the display with waveshare_3, disabling the display seems to work

jayofelony, 2024-01-12:
Okay, it seems I haven't merged the changes I made in bullseye repo to bookworm repo. I will do this ASAP, after work tonight perhaps.

kagehisa, 2024-01-12:
Same issue. Going to `/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/v4/epdconfig.py` and exchanging the line 
```
implementation = JetsonNano()
```

with 

```
implementation = RaspberryPi()
```

does the trick for now. Issue in waveshare Github exists but seems to be ignored...

jagp8, 2024-01-12:
It did! I changed it en v3 and it's working again, thanks!

jayofelony, 2024-01-12:
Simply lovely

kagehisa, 2024-01-12:
Hmm the update broke it again. Still getting the error:

>[2024-01-12 19:37:24,064] [INFO] initializing waveshare v2in13_V4 display
Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 226, in <module>
    pwnagotchi_cli()
  File "/usr/local/bin/pwnagotchi", line 205, in pwnagotchi_cli
    display = Display(config=config, state={'name': '%s>' % pwnagotchi.name()})
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 18, in __init__
    self.init_display()
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 216, in init_display
    self._implementation.initialize()
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/waveshare2in13_V4.py", line 36, in initialize
    from pwnagotchi.ui.hw.libs.waveshare.v2in13_V4.epd2in13_V4 import EPD
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/v2in13_V4/epd2in13_V4.py", line 32, in <module>
    from .. import epdconfig
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/epdconfig.py", line 238, in <module>
    implementation = JetsonNano()
                     ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/epdconfig.py", line 119, in __init__
    raise RuntimeError('Cannot find sysfs_software_spi.so')
RuntimeError: Cannot find sysfs_software_spi.so

Making the above mentioned change to `/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/epdconfig.py` fixes the issue. The if statement for the raspberry pi never gets true and JetsonNano is the default configuration.

jayofelony, 2024-01-12:
Yes, my bad. The webui will work after hotfix update.

jayofelony, 2024-01-12:
And in 2.6.6 there is the fix to epdconfig.

jayofelony, 2024-01-12:
2.6.6 wasnt working either.. 2.6.7 works now. 

FINALLY

kagehisa, 2024-01-13:
Confirmed! Works like a charm.

closed 2024-01-13 by jayofelony