jayofelony/pwnagotchi #13: Pwnagotchi captures handshakes of APs on the main.whitelist and deauths clients from them

spitfire, 2024-01-17:
**Describe the bug**
Despite putting my Access Points (both their SSIDs and BSSIDs) in /etc/pwnagotchi/config.toml main.whitelist section they are being reconed by Pwnagotchi - their handshakes are being captured, and the clients are being deauthenticated from them.

**To Reproduce**
Steps to reproduce the behavior:
1. put your SSIDs and BSSIDs in 
```
main.whitelist = [
 "i01",
 "a2:a2:f4:c2:ed:30",
]
```
(I have many more, this is just one AP cut out.)
2. (re)start Pwnagotchi in AUTO mode with AI enabled
3. watch your logs

**Expected behavior**
Pwnagotchi ignores said stations -  does not capture handshakes or perform any kind of attacks, especially deauths.

**Screenshots**
```
[2024-01-17 09:09:40,584] [WARNING] !!! captured new handshake on channel 6, -77 dBm: <redacted> () -> i01 [a2:a2:f4:c2:ed:30 ()] !!!

[2023-06-18 17:12:53,987] [INFO] deauthing <redacted> () from <hidden> (aa:a2:f4:c2:ed:30 ) on channel 6, -68 dBm ...
```

**Environment (please complete the following information):**
 - Pwnagotchi version: 2.7.0 (from this repo) and 2.6.4 (from [pwnagotchi-torch](https://github.com/jayofelony/pwnagotchi-torch)) (that I haven't re-imaged yet)
 - OS version: whichever was supplied with the above distributions from you
 - Type of hardware: Pi Zero 2 and Pi 3A
 - Any additional hardware used: none

jayofelony, 2024-01-17:
Does it save the handshakes in `/root/handshakes` or are they removed?

spitfire, 2024-01-17:
they are being saved and they're not empty. aircrack plugin even verifies them as valid.

jayofelony, 2024-01-17:
I am looking into the issue.

jayofelony, 2024-01-17:
You need to add both hostname AND mac address of the device you want to whitelist.

spitfire, 2024-01-17:
> You need to add both hostname AND mac address of the device you want to whitelist.

by hostname do you mean SSID (like"i01" here) and by MAC you mean BSSID ("a2:a2:f4:c2:ed:30") ? I already have them on my whitelist.

jayofelony, 2024-01-17:
Yes, that MAC is of the accesspoint you named i01?

jayofelony, 2024-01-17:
Well that should prevent it from getting deauthed, unless you made a type somewhere ofcourse

spitfire, 2024-01-17:
> Well that should prevent it from getting deauthed, unless you made a type somewhere ofcourse

I didn't. I've copied both the MAC and name from the logs, performed search on each and found a match in config file.

spitfire, 2024-01-17:
Yesterday and today, both after I updated and applied config. time was not synced, so the date stamp is wrong on one

spitfire, 2024-01-17:
Yes, like I said I've updated and applied (restarted pwnagotchi service) it.

jayofelony, 2024-01-17:
Than I don't know, there is no issue on the code itself.

closed 2024-01-17 by jayofelony

spitfire, 2024-01-17:
> ![image](https://private-user-images.githubusercontent.com/50277312/297457610-c7a0cee8-f350-4e3f-b428-75b93b8ff82b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU1MTQzNTAsIm5iZiI6MTcwNTUxNDA1MCwicGF0aCI6Ii81MDI3NzMxMi8yOTc0NTc2MTAtYzdhMGNlZTgtZjM1MC00ZTNmLWI0MjgtNzViOTNiOGZmODJiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE3VDE3NTQxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZiMWM1MDBjM2M5MmFmOGNhOGViYjgwNTExYTQyNDcwY2NmMmUzODU3NDFjZjA1MmIyZDY3MzczMmEzNjAxMTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.9Nleh6H0ny5cfQgj0ej-b8nkhSAMRNsz0HXiuiBM4d4) I will add an extra elif statement for good measure to skip any hostname/mac in whitelist
> 
> If you want you can add it yourself to `/usr/local/lib/python3.11/dist-packages/pwnagotchi/agent.py`
> 
> Released 2.7.1 with this fix, made it not check the last 2 characters of the MAC because in my private 3 accesspoints they differ from the given MAC address of the device itself somehow.

they do for me as well, that's why I've added all the variants I've noticed [3 APS * 2 frequencies (2,4 and 5gHz) * 4 SSIDs] :)
Are the MAC addresses in config case-sensitive (I see in the code you've added you're lowercasing them, and I've been converting them to lowercase myself)?

jayofelony, 2024-01-17:
Yeah always converted them to lowercase

spitfire, 2024-01-17:
It is still happening on both of the devices I'm playing around at home. Maybe there is something else wrong with my config - can I contact you somewhere else, as I don't want to post if publicly?