jayofelony/pwnagotchi #17: [BUG] Not always restarting wlan0mon interface when it "disappears"

rsun0525, 2024-01-18:
**Describe the bug**
I found that sometimes the wlan0mon isn't being reset when it disappears

**To Reproduce**
Steps to reproduce the behavior:
1. ... uploading pwnagotchi.log so you can see if there's another way to catch [it]([pwnagotchi.log](https://github.com/jayofelony/pwnagotchi-bookworm/files/13980671/pwnagotchi.log))

rsun0525, 2024-01-18:
Happened a couple more times, as far as I can tell, the only way to fix it is to reboot, after rebooting, it's work well for a while, etc. is there a way to turn on the feature again so when the pwnagotchi is sad, it will reboot itself?  It used to have that function, but I think you got rid of it.

[pwnagotchi.log](https://github.com/jayofelony/pwnagotchi-bookworm/files/13982416/pwnagotchi.log)

Ravioli5, 2024-01-19:
I'm having this problem too as well and this still doesn't change even with config.toml changes too.

jayofelony, 2024-01-19:
What version are you both using?

rsun0525, 2024-01-19:
The last time I uploaded my log file I was on version 2.7.2, now I'm running 2.7.3, if it happens again, I will upload my log again, please let me know if there's something else I should upload or set to help with the troubleshooting.

jayofelony, 2024-01-19:
And what device?

jayofelony, 2024-01-19:
And please make sure fix_services is enabled as it checks for this very error message.

rsun0525, 2024-01-19:
Yes, I see the fix_services running and fixing other times it fails, just must be some situation where it can't fix it.

Ravioli5, 2024-01-19:
This problem was here for me since 2.6.9 (first one I used in the bookworm release) and I'm using a raspberry pi 4.

jayofelony, 2024-01-20:
I cannot recreate this issue on my Pi4 or zero2w, ofcourse it will reboot sometimes if there is no other way to fix it. But I do not get it to stop fixing the wifi firmware.

rsun0525, 2024-01-20:
[pwnagotchi.log](https://github.com/jayofelony/pwnagotchi-bookworm/files/13997930/pwnagotchi.log)

Happened again, I let it sit the whole night and this morning wlan0mon still hadn't come back.  Is there any command I can run while it's down to help with the troubleshooting?

jayofelony, 2024-01-20:
`sudo monstart`

rsun0525, 2024-01-21:
I didn't realize a 2.7.3 full image existed.  I will redo my Pi in that and see how it's going.  What I found interesting it also seems like fix services stops running as I don't see it starting up anymore in the logs, so could that be the process that's "crashing", because this last time I had an issue, I also did a sudo monstart and it fixed the problem.

rsun0525, 2024-01-21:
Happened again and for some odd reason the Fix Services just doesn't seem to be running, so doing a sudo monstart brings up the interface, then I do a sudo systemctl restart pwnagotchi, then everything is good again.  Any additional logging that can be done with Fix Services to see why it "stops" working?

jayofelony, 2024-01-21:
But why would it be stopped, it is enabled by default. How many plugins are you running in total?

rsun0525, 2024-01-21:
[image]

jayofelony, 2024-01-21:
Nothing wrong there, I have no idea why fix_services would unload itself

rsun0525, 2024-01-21:
So far, each time I've seen the problem occur, the fix is the "sudo monstart", and only sometimes I need to then do a "sudo systemctl restart pwnagotchi", I will reimage it today with the latest version and see if it continues.

rsun0525, 2024-01-21:
Restarted with 2.7.3 image and I see it's already running 2.7.5, and I had to do the sudo monstart for it to fix it, I saw on the logs that the fix service is running, but just doesn't continue to run.  I've attached the latest log.
[pwnagotchi.log](https://github.com/jayofelony/pwnagotchi-bookworm/files/14002220/pwnagotchi.log)

jayofelony, 2024-01-21:
[image]
Works perfectly fine on my zero2w

Ravioli5, 2024-01-22:
pi@spark:~ $ sudo monstart
command failed: No such device (-19)
command failed: No such file or directory (-2)
wlan0mon: ERROR while getting interface flags: No such device
command failed: No such device (-19)

pi@spark:~ $ sudo monstart
command failed: Operation not supported (-95)

This is what I get when I try sudo monstart

rsun0525, 2024-01-22:
This may be a shot in the dark, but are you running this on a Raspberry Pi 4b or 400?  If so, have you upgraded it to the latest firmware?  For example:

 sudo rpi-eeprom-update
BOOTLOADER: up to date
   CURRENT: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
    LATEST: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138c0
    LATEST: 000138c0

Ravioli5, 2024-01-22:
I am on a raspi 4

rsun0525, 2024-01-22:
> pi@spark:~ $ sudo monstart command failed: No such device (-19) command failed: No such file or directory (-2) wlan0mon: ERROR while getting interface flags: No such device command failed: No such device (-19)
> 
> pi@spark:~ $ sudo monstart command failed: Operation not supported (-95)
> 
> This is what I get when I try sudo monstart after a restart when the mon is online

This tells me that it's "dead" to a point you need a reboot, etc. as the driver probably crashed so bad it can't be recovered from.

Ravioli5, 2024-01-22:
Ohhhhhhhhhhhh

Ravioli5, 2024-01-22:
Ok

Ravioli5, 2024-01-22:
sudo rpi-eeprom-update
*** UPDATE AVAILABLE ***
BOOTLOADER: update available
   CURRENT: Tue 26 Apr 10:24:28 UTC 2022 (1650968668)
    LATEST: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader-2711/default)
            Use raspi-config to change the release.
  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1

Ravioli5, 2024-01-22:
This is what I got

Ravioli5, 2024-01-22:
> This tells me that it's "dead" to a point you need a reboot, etc. as the driver probably crashed so bad it can't be recovered from.

Does this mean I have to update or reflash?

Ravioli5, 2024-01-22:
Just sudo apt update or anything else?

rsun0525, 2024-01-22:
you can just type sudo rpi-eeprom-update -a and then reboot (assuming you having all the latest patches installed through sudo apt update and sudo apt full-upgrade already done).

Ravioli5, 2024-01-22:
Ok thx I'll tell you if it works

Ravioli5, 2024-01-22:
Still doesn't but I noticed this in sudo systemctl

● ifup@wlan0.service                                                                                loaded failed failed    ifup for wlan0

jayofelony, 2024-01-22:
That is disabled, because we want full control over wlan0 with pwnagotchi.

Ravioli5, 2024-01-22:
Ok

Ravioli5, 2024-01-22:
Another thing I noticed was that the networking.service was exited, does this mean anything?

Ravioli5, 2024-01-22:
Found this in my logs...

04:58:05 war error while setting interface wlan0mon in monitor mode: No such device exists
04:58:05 !!! error while starting signaling: error while activating handle: No Such Device

Ravioli5, 2024-01-22:
pi@spark:~ $ pwnlog
04:55:03 inf pwngrid v1.10.9 starting in peer mode ...
08:39:36 inf pwngrid v1.10.9 starting in peer mode ...
08:40:33 inf pwngrid v1.10.9 starting in peer mode ...
08:41:33 inf /etc/pwnagotchi/id_rsa found
08:41:33 inf started beacon discovery and message routing (0 known peers)
08:41:33 inf peer spark@16580172af715b623a3cdad01c70626dca7967b7d86a7d3a464def675012a825 signaling is ready
08:41:33 inf pwngrid api starting on 127.0.0.1:8666 ...
08:41:34 inf peer advertisement enabled
08:42:18 err POST https://api.opwngrid.xyz/api/v1/unit/enroll (1.129026193s) invalid character 'e' looking for beginning of value
08:42:24 inf pwngrid v1.11.0 starting in peer mode ...
08:42:26 inf /etc/pwnagotchi/id_rsa found
08:42:27 inf started beacon discovery and message routing (0 known peers)
08:42:27 inf peer spark@16580172af715b623a3cdad01c70626dca7967b7d86a7d3a464def675012a825 signaling is ready
08:42:27 inf pwngrid api starting on 127.0.0.1:8666 ...
08:42:32 inf peer advertisement enabled
09:09:59 inf pwngrid v1.11.0 starting in peer mode ...
09:10:01 inf /etc/pwnagotchi/id_rsa found
09:10:01 inf interface wlan0mon is down, bringing it up ...
09:10:02 inf started beacon discovery and message routing (0 known peers)
09:10:02 inf peer spark@16580172af715b623a3cdad01c70626dca7967b7d86a7d3a464def675012a825 signaling is ready
09:10:02 inf pwngrid api starting on 127.0.0.1:8666 ...
09:10:55 inf peer advertisement enabled
11:10:25 inf pwngrid v1.11.0 starting in peer mode ...
11:10:27 inf /etc/pwnagotchi/id_rsa found
11:10:27 inf interface wlan0mon is down, bringing it up ...
11:10:27 inf started beacon discovery and message routing (0 known peers)
11:10:27 inf peer spark@16580172af715b623a3cdad01c70626dca7967b7d86a7d3a464def675012a825 signaling is ready
11:10:27 inf pwngrid api starting on 127.0.0.1:8666 ...
07:09:04 inf peer advertisement enabled
13:46:03 inf pwngrid v1.11.0 starting in peer mode ...
13:46:05 inf /etc/pwnagotchi/id_rsa found
13:46:05 war error while setting interface wlan0mon in monitor mode: No such device exists
13:46:05 inf interface wlan0mon is down, bringing it up ...
13:46:05 inf started beacon discovery and message routing (0 known peers)
13:46:05 inf peer spark@44cebfba2540bc928d3effd6c6350575c3317d14129a58554db66a047bf4c5c4 signaling is ready
13:46:05 inf pwngrid api starting on 127.0.0.1:8666 ...
11:34:37 inf peer advertisement enabled
11:35:22 err POST https://api.opwngrid.xyz/api/v1/unit/enroll (1.496038287s) invalid character 'e' looking for beginning of value
11:49:02 inf pwngrid v1.11.0 starting in peer mode ...
11:49:04 inf /etc/pwnagotchi/id_rsa found
11:49:04 war error while setting interface wlan0mon in monitor mode: No such device exists
11:49:04 !!! error while starting signaling: error while activating handle: No Such Device
13:33:02 inf pwngrid v1.11.0 starting in peer mode ...
13:33:04 inf /etc/pwnagotchi/id_rsa found
13:33:04 inf started beacon discovery and message routing (0 known peers)
13:33:04 inf peer spark@44cebfba2540bc928d3effd6c6350575c3317d14129a58554db66a047bf4c5c4 signaling is ready
13:33:04 inf pwngrid api starting on 127.0.0.1:8666 ...
13:33:11 inf peer advertisement enabled
15:10:33 inf pwngrid v1.11.0 starting in peer mode ...
15:10:35 inf /etc/pwnagotchi/id_rsa found
15:10:35 war error while setting interface wlan0mon in monitor mode: No such device exists
15:10:35 !!! error while starting signaling: error while activating handle: No Such Device
15:12:01 inf pwngrid v1.11.0 starting in peer mode ...
15:12:03 inf /etc/pwnagotchi/id_rsa found
15:12:04 inf started beacon discovery and message routing (0 known peers)
15:12:04 inf peer spark@44cebfba2540bc928d3effd6c6350575c3317d14129a58554db66a047bf4c5c4 signaling is ready
15:12:04 inf pwngrid api starting on 127.0.0.1:8666 ...
15:12:10 inf peer advertisement enabled
16:08:31 inf pwngrid v1.11.0 starting in peer mode ...
16:08:33 inf /etc/pwnagotchi/id_rsa found
16:08:33 war error while setting interface wlan0mon in monitor mode: No such device exists
16:08:33 !!! error while starting signaling: error while activating handle: No Such Device
03:56:34 inf pwngrid v1.11.0 starting in peer mode ...
03:56:36 inf /etc/pwnagotchi/id_rsa found
03:56:36 inf started beacon discovery and message routing (0 known peers)
03:56:36 inf peer spark@44cebfba2540bc928d3effd6c6350575c3317d14129a58554db66a047bf4c5c4 signaling is ready
03:56:36 inf pwngrid api starting on 127.0.0.1:8666 ...
03:56:42 inf peer advertisement enabled
05:09:46 inf pwngrid v1.11.0 starting in peer mode ...
[... 194 lines cut ...]
14:34:20 inf peer spark@44cebfba2540bc928d3effd6c6350575c3317d14129a58554db66a047bf4c5c4 signaling is ready
14:34:20 inf pwngrid api starting on 127.0.0.1:8666 ...
02:55:16 inf peer advertisement enabled
04:58:03 inf pwngrid v1.11.0 starting in peer mode ...
04:58:05 inf /etc/pwnagotchi/id_rsa found
04:58:05 war error while setting interface wlan0mon in monitor mode: No such device exists
04:58:05 !!! error while starting signaling: error while activating handle: No Such Device
04:58:57 inf pwngrid v1.11.0 starting in peer mode ...
04:58:59 inf /etc/pwnagotchi/id_rsa found
04:58:59 inf started beacon discovery and message routing (0 known peers)
04:58:59 inf peer spark@44cebfba2540bc928d3effd6c6350575c3317d14129a58554db66a047bf4c5c4 signaling is ready
04:58:59 inf pwngrid api starting on 127.0.0.1:8666 ...
04:59:31 inf peer advertisement enabled
04:59:57 inf pwngrid v1.11.0 starting in peer mode ...
[... cut ...]
04:59:59 !!! error while starting signaling: error while activating handle: No Such Device
05:00:43 inf pwngrid v1.11.0 starting in peer mode ...
05:00:45 inf /etc/pwnagotchi/id_rsa found
05:00:46 inf started beacon discovery and message routing (0 known peers)
05:00:46 inf peer spark@44cebfba2540bc928d3effd6c6350575c3317d14129a58554db66a047bf4c5c4 signaling is ready
05:00:46 inf pwngrid api starting on 127.0.0.1:8666 ...
05:12:06 inf peer advertisement enabled
^[[B^[[B^[[B^[[B^[[B^[[B^[[B^[[B^[[A^[[A^[[A^[[B
^[[

rsun0525, 2024-01-22:
If you do ifconfig -a, you see wlan0 interface?  Are you just using the internal Wi-Fi, or did you connect an external Wi-Fi adaptor to it?

jayofelony, 2024-01-22:
That error is normal.. we delay setting monitor as otherwise driver will crash immediately on zero2w.

Ravioli5, 2024-01-22:
>

> 

> That error is normal.. we delay setting monitor as otherwise driver will crash immediately on zero2w.

Oh ok

Ravioli5, 2024-01-22:
> If you do ifconfig -a, you see wlan0 interface? Are you just using the internal Wi-Fi, or did you connect an external Wi-Fi adaptor to it?

I'm using internal and see the interface

rsun0525, 2024-01-26:
I redid it with 2.7.7 image, no other upgrade, and still I have the odd situation where it stop working and doesn't automatically recover, when I get into this condition, I only need to do a "sudo monstart" and "pwnkill" and then I'm up and running.

jayofelony, 2024-01-26:
Nothing I can do about it, you are the only 2 reporting this. I cannot reproduce this issue.

rsun0525, 2024-01-26:
What's really strange is that sometimes it does work and automatically restart itself, just there must be some case where it doesn't see it as down even though it is:

[2024-01-26 08:32:51,045] [INFO] sending association frame to FiOS-D6FQA (48:5d:36:93:98:58 Verizon) on channel 11 [0 clients], -70 dBm...
[2024-01-26 08:32:51,585] [INFO] sending association frame to <hidden> (e2:db:d1:47:c4:ba ) on channel 11 [0 clients], -78 dBm...
[2024-01-26 08:32:51,620] [WARNING] [bettercap] ping error - retrying connection in 4.610791349226044 sec
[2024-01-26 08:32:51,621] [WARNING] [bettercap] can't run my request... connection to the bettercap endpoint failed...
[2024-01-26 08:32:51,624] [WARNING] [bettercap] retrying run in 0.9816867577224726 sec
[2024-01-26 08:32:52,609] [WARNING] [bettercap] can't run my request... connection to the bettercap endpoint failed...
[2024-01-26 08:32:52,609] [WARNING] [bettercap] retrying run in 1.6016602243233105 sec
[2024-01-26 08:32:54,218] [WARNING] [bettercap] can't run my request... connection to the bettercap endpoint failed...
[2024-01-26 08:32:54,219] [WARNING] [bettercap] retrying run in 4.119538450401606 sec
[2024-01-26 08:32:55,077] [ERROR] [agent:_fetch_stats] self.session: ConnectionError(MaxRetryError("HTTPConnectionPool(host='localhost', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f33f7dfd0>: Failed to establish a new connection: [Errno 111] Connection refused'))"))
[2024-01-26 08:32:56,238] [INFO] [bettercap] creating new websocket...
[2024-01-26 08:32:56,244] [WARNING] connection to the bettercap endpoint failed...
[2024-01-26 08:32:56,245] [WARNING] restarting in AUTO mode ...
[2024-01-26 08:32:59,185] [INFO] [update] plugin loaded.
[2024-01-26 08:32:59,186] [INFO] [Fix_Services] plugin loaded.
[2024-01-26 08:32:59,186] [INFO] webcfg: Plugin loaded.
[2024-01-26 08:32:59,186] [INFO] memtemp plugin loaded.
[2024-01-26 08:32:59,187] [INFO] WPA_SEC: plugin loaded

Ravioli5, 2024-01-27:
Yeah mine is working sometimes for 5 mins and other times for 30. Dunno why and it does restart sometimes too. 5 mins is the norm tho

Ravioli5, 2024-01-28:
Sudo monstart doesn't work for me but I've been using pwnkill instead.

Ravioli5, 2024-01-28:
I see some grid problems for me as well but I'm not sure how that would do anything to the wlan0mon

Ravioli5, 2024-01-28:
Ok it's way worse all of a sudden. The pwnagotchi is blind from boot and nothing is fixing it

Ravioli5, 2024-01-28:
I've been seeing this too

● spark
    State: degraded
    Units: 315 loaded (incl. loaded aliases)
     Jobs: 0 queued
   Failed: 1 units
    Since: Thu 1970-01-01 01:00:03 BST; 54 years 0 months ago
  systemd: 252.19-1~deb12u1
   CGroup: /
           ├─init.scope
           │ └─1 /sbin/init
           ├─system.slice
           │ ├─ModemManager.service
           │ │ └─652 /usr/sbin/ModemManager
           │ ├─NetworkManager.service
           │ │ └─600 /usr/sbin/NetworkManager --no-daemon
           │ ├─bettercap.service
           │ │ ├─90583 bash /usr/bin/bettercap-launcher
           │ │ └─90720 /usr/local/bin/bettercap -no-colors -caplet pwnagotchi-m>           │ ├─bluetooth.service
           │ │ └─728 /usr/libexec/bluetooth/bluetoothd --noplugin=sap,a2dp
           │ ├─cron.service
           │ │ └─516 /usr/sbin/cron -f
           │ ├─dbus.service
           │ │ └─517 /usr/bin/dbus-daemon --system --address=systemd: --nofork >           │ ├─ifup@eth0.service
           │ │ └─383 dhclient -4 -v -i -pf /run/dhclient.eth0.pid -lf /var/lib/>           │ ├─polkit.service
           │ │ └─535 /usr/lib/polkit-1/polkitd --no-debug
           │ ├─pwnagotchi.service
           │ │ ├─98575 bash /usr/bin/pwnagotchi-launcher
           │ │ └─98578 /usr/bin/python3 /usr/local/bin/pwnagotchi --manual
           │ ├─pwngrid-peer.service
           │ │ └─897 /usr/local/bin/pwngrid -keys /etc/pwnagotchi -peers /root/>           │ ├─ssh.service
           │ │ └─730 "sshd: /usr/sbin/sshd -D [listener] 0 of 10-100 startups"
           │ ├─system-getty.slice
           │ │ └─getty@tty1.service
           │ │   └─719 /sbin/agetty -o "-p -- \\u" --noclear - linux
           │ ├─systemd-journald.service
           │ │ └─251 /lib/systemd/systemd-journald
           │ ├─systemd-logind.service
           │ │ └─591 /lib/systemd/systemd-logind
           │ ├─systemd-timesyncd.service
           │ │ └─432 /lib/systemd/systemd-timesyncd
           │ └─systemd-udevd.service
           │   └─udev
           │     └─279 /lib/systemd/systemd-udevd
           └─user.slice
             └─user-1000.slice
               ├─session-2.scope
               │ ├─97628 "sshd: pi [priv]"
               │ ├─97662 "sshd: pi@pts/0"
               │ ├─97664 -bash
               │ ├─99301 sudo systemctl status
               │ ├─99304 sudo systemctl status
               │ ├─99305 systemctl status
               │ └─99306 less
               └─user@1000.service
                 └─init.scope
                   ├─97631 /lib/systemd/systemd --user
                   └─97632 "(sd-pam)"

rsun0525, 2024-01-28:
Reimaged all my Pi's to 2.7.9, I would of expected to see a problem by now, but I saw it automatically reboot the PI this time when it encountered an issue, so hopefully this has solved it.

closed 2024-01-28 by rsun0525

Ravioli5, 2024-01-29:
Hmm, I'll try this too then

Ravioli5, 2024-01-29:
I'm going to re image now as the pwnagotchi died after auto update

Ravioli5, 2024-01-29:
Everything looks fine now to me

knoxoxen, 2024-10-09:
/etc/pwnagotchi/log/pwngrid-peer.log <==
15:09:18 inf pwngrid v1.11.4 starting in peer mode ...
15:14:02 inf pwngrid v1.11.4 starting in peer mode ...
15:14:04 inf /etc/pwnagotchi/id_rsa found
15:14:05 war error while setting interface wlan0mon in monitor mode: No such device exists
15:14:05 !!! error while starting signaling: error while activating handle: No Such Device
15:14:35 inf pwngrid v1.11.4 starting in peer mode ...
15:14:37 inf /etc/pwnagotchi/id_rsa found
15:14:37 inf started beacon discovery and message routing (0 known peers)
15:14:37 inf peer pwnagotchi@3c05890d90682a7d05fa117646d416c1f8f8145d7669fd09fc9be9d5ee88857c signaling is ready
15:14:37 inf pwngrid api starting on 127.0.0.1:8666 ...
^C
pi@pwnagotchi:~ $ sudo monstart
command failed: Operation not supported (-95)
pi@pwnagotchi:~ $ pwnkill
pi@pwnagotchi:~ $ sudo monstart
command failed: Operation not supported (-95)
pi@pwnagotchi:~ $

knoxoxen, 2024-10-09:
pi@pwnagotchi:~ $ ifconfig
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether e4:5f:01:2f:eb:bc  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 2  bytes 190 (190.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2  bytes 190 (190.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.0.0.2  netmask 255.255.255.0  broadcast 10.0.0.255
        inet6 fe80::2469:58ff:fe55:4a47  prefixlen 64  scopeid 0x20<link>
        ether 26:69:58:55:4a:47  txqueuelen 1000  (Ethernet)
        RX packets 738  bytes 79411 (77.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 754  bytes 177015 (172.8 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0mon: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 1000  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dr