jayofelony/pwnagotchi #19: [BUG]Restarting into manual mode instead of auto

Ravioli5, 2024-01-19:
**Describe the bug**
When restarted, the pwnagotchi sometimes restarts in manual mode. Mildly annoying but can be changed back to auto mode. Happens every once in a while but not always.

**To Reproduce**
Steps to reproduce the behavior:
1. Restart pwnagotchi
2. Check what mode it is in
3. I used web config restart to reproduce this bug, ssh didn't do the same thing yet

**Expected behavior**
Manual mode is coming instead of auto.

**Environment (please complete the following information):**
 - 2.7.3
 - bookworm
 - raspberry pi 4
 - ethernet cable to home network?

**Additional context**
This happened with the 2.6.9 and onwards too. I'm not sure about the previous ones as I have not tested

jayofelony, 2024-01-19:
If you restarted into a certain mode from webui, you must refresh the webui after restarting in order to load the corresponding restart into "MANUAL/AUTO" button.

Ravioli5, 2024-01-19:
I click the default refresh button and I do restart the webpage. I thought this was the problem too but I made sure to do it properly but it still happens.

 I usually wouldn't restart my pwnagotchi all the time however the wlanmon is having problems as it cannot collect handshakes after a while. It gets fixed after restarting for a while then I have to restart it again.

Ravioli5, 2024-01-20:
Oh and when it shows manual mode the display is frozen so I cannot see if the pwnagotchi is working too. I use the web UI for now (display isn't going to be used for me)

jayofelony, 2024-01-20:
In MANU display does not refresh.

jayofelony, 2024-01-20:
You don't happen to have a usb cable connected to a pc as well?

Ravioli5, 2024-01-20:
No but I do use the web UI with the pwnagotchi connected to a home router. I noticed that whenever i restart with web UI it restarts into manual mode. I'm not sure why cause I didn't click the restart into manual mode but it's happening more frequently now, more like nearly every time.

Ravioli5, 2024-01-20:
It happens when the pwnagotchi is unplugged and then plugged in again too, but pwnkill has not done this yet.

jayofelony, 2024-01-20:
I use a RPi4 and do not have this issue, and it is also connected by ethernet to my modem.

Ravioli5, 2024-01-20:
Hmm I'm not sure then...

Ravioli5, 2024-01-20:
What should I do? I don't want to reflash but I won't restart this many times if I didn't get the wlan0mon problem...

Ravioli5, 2024-01-20:
Accidentally closed, sorry

jayofelony, 2024-01-20:
Do you have another sd card on which you can flash 2.7.3 to test?

Ravioli5, 2024-01-20:
I do. I'll try it and tell you what happens.

Ravioli5, 2024-01-22:
I got a possible fix. The drivers may have crashed irreparablly and an outdated bootloader so I have to fix this

Ravioli5, 2024-01-22:
Nope this did not fix it

Ravioli5, 2024-01-22:
It's worse now, it allows you only to switch to manual mode from manual mode in the web UI, I'm going to attempt to fix it for a while

jayofelony, 2024-01-22:
I don't know what is going on with your set-up, I cannot reproduce this in any way. Have you tried running without your custom config. Maybe there is a problem there?

Ravioli5, 2024-01-22:
Ok I will try...

Ravioli5, 2024-01-27:
Ok I'm having multiple web UI problems so I have to troubleshoot these for a while before I can get a actual report. I also have minor UI problems as well that didn't change with anything yet. I'll try to fix it on my own first before getting back

Ravioli5, 2024-01-29:
I'm going to re image now as the pwnagotchi died after auto update

Ravioli5, 2024-01-29:
Closed as cannot reproduce

closed 2024-01-29 by Ravioli5