jayofelony/pwnagotchi #20: [BUG]  Adding main.whitelist prevents service from starting in 2.7.3

halgski, 2024-01-19:
Any addition of main.whitelist to config.toml prevents the pwnagotchi service from starting.

**To Reproduce**
Add main.whitelist to /etc/pwnagotchi/config.toml

main.whitelist = [ 
"myNetwork"
]

**Expected behavior**

Service starts as usual:

● pwnagotchi.service - pwnagotchi Deep Reinforcement Learning instrumenting bettercap for WiFI pwning.     Loaded: loaded (/etc/systemd/system/pwnagotchi.service; enabled; preset: enabled)     Active: active (running) since Fri 2024-01-19 17:31:59 GMT; 17s ago       Docs: https://pwnagotchi.ai   Main PID: 842 (bash)      Tasks: 9        CPU: 7.563s     CGroup: /system.slice/pwnagotchi.service             ├─842 bash /usr/bin/pwnagotchi-launcher             └─844 /usr/bin/python3 /usr/local/bin/pwnagotchi --manual

**Screenshots**
With main.whitelist the service fails to start

● pwnagotchi.service - pwnagotchi Deep Reinforcement Learning instrumenting bettercap for WiFI pwning.     Loaded: loaded (/etc/systemd/system/pwnagotchi.service; enabled; preset: enabled)     Active: activating (auto-restart) (Result: exit-code) since Fri 2024-01-19 17:30:12 GMT; 21s ago       Docs: https://pwnagotchi.ai    Process: 733 ExecStart=/usr/bin/pwnagotchi-launcher (code=exited, status=1/FAILURE)   Main PID: 733 (code=exited, status=1/FAILURE)        CPU: 4.371s

**Environment (please complete the following information):**
 - 2.7.3
 - 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
 - Raspberry Pi Zero 2W
 - Waveshare v4

**Additional context**
Removing the main.whitelist line from the config file and restarting resolves the problem.

jayofelony, 2024-01-19:
What is output of `sudo pwnagotchi --debug`?

halgski, 2024-01-19:
ERROR:root:There was an error processing the configuration file:

Duplicate keys! (line 121 column 1 char 4281)

halgski, 2024-01-19:
Note: That output pointed me in the right direction to find the duplicate
main.whitelist entry.  Thanks.

jayofelony, 2024-01-19:
Glad to know nothing was wrong with what I did or didn't do haha

closed 2024-01-19 by jayofelony