jayofelony/pwnagotchi #21: [BUG]

ChrisSewell, 2024-01-20:
**Describe the bug**
pwnagotchi-launcher (code=exited, status=1/FAILURE

**To Reproduce**
Flash 2.7.3 to SD card
Insert into Pi Zero 2 WH
Power Pi
sudo systemctl status pwnagotchi

**Expected behavior**
Pwnagotchi service starts properly

**Environment (please complete the following information):**
 - Pwnagotchi version 2.7.3
 - Type of hardware Pi Zero 2 WH

**Additional context**
Tried flashing a different SD card, same issue.
Downloaded previous version, same issue

Ravioli5, 2024-01-20:
This has happened to me on the torch release as well. Usually reflashing fixes it after a while but editing the config .toml fixes it too.

ChrisSewell, 2024-01-20:
> This has happened to me on the torch release as well. Usually reflashing fixes it after a while but editing the config .toml fixes it too.

What edit are you making to the config to resolve this?

Ravioli5, 2024-01-20:
I changed the whitelist to being blank and it worked but web config stopped working.

jayofelony, 2024-01-20:
`sudo pwnagotchi --debug` what is output from that command?

Ravioli5, 2024-01-22:
Usually it would return something about the config.toml not being done properly and shows where the error is being caused so sudo pwnagotchi --debug tells you where to fix it too.

jayofelony, 2024-01-22:
> Usually it would return something about the config.toml not being done properly and shows where the error is being caused so sudo pwnagotchi --debug tells you where to fix it too.

I know, I would like an output from that command while he is having issues.