jayofelony/pwnagotchi #22: [BUG] waveshare 213g display broken

Opalinus, 2024-01-20:
**Describe the bug**
Waveshare 213g (also named 2in13g) is broken

**To Reproduce**
Set config.toml to waveshare213g, stop pwnagotchi va systemd and restart /usr/bin/pwnagotchi-launcher manually to seee output:

```
pi@pwnagotchi:/tmp $ sudo bash /usr/bin/pwnagotchi-launcher 
[2023-12-11 05:03:57,985] [INFO] [update] plugin loaded.
[2023-12-11 05:03:57,986] [INFO] webcfg: Plugin loaded.
[2023-12-11 05:03:57,987] [INFO] [Fix_Services] plugin loaded.
[2023-12-11 05:03:57,987] [INFO] GPIO Button plugin loaded.
Exception ignored in thread started by: <function locked_cb at 0x7fb394f4c0>
Traceback (most recent call last):
[2023-12-11 05:03:57,995] [INFO] Session-stats plugin loaded.
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/__init__.py", line 85, in locked_cb
    cb(*args, *kwargs)
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/gpio_buttons.py", line 29, in on_loaded
    gpios = self.options['gpios']
            ~~~~~~~~~~~~^^^^^^^^^
KeyError: 'gpios'
[2023-12-11 05:03:58,011] [WARNING] ui.fps is 0, the display will only update for major changes
[2023-12-11 05:03:58,012] [INFO] initializing waveshare v2in13g display
Exception in thread Thread-7 (_render_thread):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 240, in _render_thread
    self._implementation.render(self._canvas_next)
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/waveshare2in13g.py", line 43, in render
    self._display.displayPartial(buf)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'EPD' object has no attribute 'displayPartial'
[2023-12-11 05:04:18,855] [INFO] pwnagotchi@8c2afad7831fcea67475242d61475a3feadc3c68accd64866b8d5a438c94a9ad (v2.7.3)
[2023-12-11 05:04:18,861] [INFO] entering manual mode ...
[2023-12-11 05:04:18,943] [INFO] web ui available at http://0.0.0.0:8080/
 * Serving Flask app 'pwnagotchi.ui.web.server'
 * Debug mode: off
[2023-12-11 05:04:18,979] [INFO] the last session lasted 8 minutes, 59 seconds (0 completed epochs, trained for 0), average reward:0.0 (min:1000 max:-1000)
```

**Environment (please complete the following information):**
 - Pwnagotchi version: 2.7.3
 - Type of hardware: Rpi Zero W 2
 - Any additional hardware used: Waveshare 213G

**Additional context**
temporary fix repaired the problem

jayofelony, 2024-01-20:
Merged! I actually done this same fix locally on my laptop!

jayofelony, 2024-01-21:
@Opalinus does it work for you like it should, I'm getting messages it doesn't fully work.

Opalinus, 2024-01-21:
> @Opalinus does it work for you like it should, I'm getting messages it doesn't fully work.

Since my changes you modified waveshare2in13g.py again in [Moved self._display = None to base.py for easier debugging display files.](https://github.com/jayofelony/pwnagotchi-bookworm/commit/e50cf04967942fadb1815ede5905617035c72a0f)

This change is too big to review and probably has nothing to do with my issue. I would prefer to see the files from externals sources like @waveshareteam intact, otherwise merges will complicate things.

jayofelony, 2024-01-21:
waveshare2in13g.py is not a waveshare file, its a pwnagotchi display file. It has the layout of the face etc in it. Down inside this file it loads epd2in13g which is the library file from Waveshare which is how they made it.

Opalinus, 2024-01-21:
> 
> waveshare2in13g.py is not a waveshare file, its a pwnagotchi display file. It has the layout of the face etc in it. Down inside this file it loads epd2in13g which is the library file from Waveshare which is how they made it.

does that mean all displays by waveshare are defunct now? I did not quite understand your commit 🥲

jayofelony, 2024-01-21:
No, my waveshare v2 works perfectly fine. My commit was for me, it loaded self._display as a None property. So I moved it to base.py. This way if I add a new display, I can debug the same error as it had with display quicker. Because my IDE will tell me there is a problem now, it did not before. Hence the bug you found went unnoticed for me.

Opalinus, 2024-01-21:
Ill try this as early as in two days. I ran their 1.2 epdconfig file.

I saw the flickering with the c code as well. I also ordered a b/w only for more testing

Opalinus, 2024-01-24:
I tried epdconfig.py file from waveshareteam and can confirm the original file does not work. The GPIO busy exception is thrown excactly [as you predicted](https://github.com/jayofelony/pwnagotchi-bookworm/files/14031536/ugly_exception.txt)

Would love some more input why the files differ. Any chance to add support for epd2in13g so it does only update when needed and not flash like crazy?

Also I find debugging by microusb (still on zero2w) quite tedious and I might look into the deploy feature in 
PyCharm. Till then my b/w epd2in13 should arrive, and I'm up for more testing this weekend.

closed 2024-03-27 by jayofelony