jayofelony/pwnagotchi #24: Common white list

jagp8, 2024-01-22:
It would be great if we could define a single whitelist so we only have to modify it in one place instead of five.

I understand that having a whitelist for each plugin gives you more flexibility, but if you plan to put the same networks on all the whitelists, it's just a hindrance.

It would be great if we could have a common one that applies to all whitelists and have additional values for plugins whose flexibility we want to maintain.

wpa-2, 2024-01-22:
Feel free to create a plugin I'm sure it would be welcome.

jayofelony, 2024-01-22:
I think this is an easy change, I can just make the 3 plugins that currently use the whitelist setting look for the `main.whitelist` setting in config.

Current 3 plugins are wigle, wpa-sec and onlinehashcrack.

jayofelony, 2024-01-22:
I forgot about grid, this also uses a whitelist. Called exclude, but I can change that.

closed 2024-01-23 by jayofelony