jayofelony/pwnagotchi #26: [BUG] Installed on Raspberry Pi Zero 2W and it's blind

rsun0525, 2024-01-23:
**Describe the bug**
Installed 2.7.3 on Raspberry Pi Zero 2W and it updated to 2.7.6, but still blind

At one point I had heard there's more than one WiFi chipset for Raspberry Pi Zero 2W, do you support both?

[pwnagotchi.log](https://github.com/jayofelony/pwnagotchi-bookworm/files/14016902/pwnagotchi.log)

jayofelony, 2024-01-23:
Yes

jagp8, 2024-01-24:
I have two Zeros2W and none of the is finding anithing since updated to 2.7.7 :S

jagp8, 2024-01-24:
I did a sudo apt update and full-upgrade to force the update, maybe it's related

jayofelony, 2024-01-24:
You shouldn't do that.. that might overwrite the custom wifi firmware. And probably did, as you say its blind now. Which has absolutely nothing to do with upgrading pwnagotchi.

You should run `sudo pwnagotchi --check-update`.

rsun0525, 2024-01-24:
Redoing it again with the 2.7.3 (currently running 2.7.7) and not doing and sudo apt update and full-upgrade solved it.

closed 2024-01-24 by rsun0525