jayofelony/pwnagotchi #30: [BUG] pwnagotchi plugins list command fails

spitfire, 2024-01-26:
**Describe the bug**

`pwnagotchi plugins list` command fails:

```
# pwnagotchi plugins list
Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 230, in <module>
    pwnagotchi_cli()
  File "/usr/local/bin/pwnagotchi", line 147, in pwnagotchi_cli
    rc = plugins_cmd.handle_cmd(args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/cmd.py", line 85, in handle_cmd
    return list_plugins(args, config)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/cmd.py", line 202, in list_plugins
    max_len = max(map(len, max_len_list))
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

**To Reproduce**
Steps to reproduce the behavior:
1. run pwnagotchi plugins list as root 

**Expected behavior**
It should list plugins

**Environment (please complete the following information):**
 - Pwnagotchi version: 2.7.7
 - OS version: bookworm
 - Type of hardware: rpi3a

jayofelony, 2024-01-26:
First do `sudo pwnagotchi plugins update`

closed 2024-01-26 by jayofelony

spitfire, 2024-01-26:
Tried, it - after that it's the same:

```
┌─[root@spitfire-pi3a] - [~] - [2024-01-26 23:10:34]
└─[130] pwnagotchi plugins update
[2024-01-26 23:10:47,781] [INFO] Downloading plugins from https://github.com/jayofelony/pwnagotchi-torch-plugins/archive/master.zip to /usr/local/share/pwnagotchi/available-plugins/plugins0.zip
[2024-01-26 23:10:50,546] [INFO] Unzipping...
[2024-01-26 23:10:51,115] [INFO] 2 file(s) were changed.
[2024-01-26 23:10:51,120] [INFO] Downloading plugins from https://github.com/evilsocket/pwnagotchi-plugins-contrib/archive/master.zip to /usr/local/share/pwnagotchi/available-plugins/plugins1.zip
[2024-01-26 23:10:53,382] [INFO] Unzipping...
[2024-01-26 23:10:53,583] [INFO] 2 file(s) were changed.
[2024-01-26 23:10:53,585] [INFO] Downloading plugins from https://github.com/tisboyo/pwnagotchi-pisugar2-plugin/archive/master.zip to /usr/local/share/pwnagotchi/available-plugins/plugins2.zip
[2024-01-26 23:10:55,958] [INFO] Unzipping...
[2024-01-26 23:10:56,373] [INFO] Downloading plugins from https://github.com/nullm0ose/pwnagotchi-plugin-pisugar3/archive/master.zip to /usr/local/share/pwnagotchi/available-plugins/plugins3.zip
[2024-01-26 23:10:58,415] [INFO] Unzipping...
[2024-01-26 23:10:58,833] [INFO] 1 file(s) were changed.
[2024-01-26 23:10:58,835] [INFO] Downloading plugins from https://github.com/Sniffleupagus/pwnagotchi_plugins/archive/master.zip to /usr/local/share/pwnagotchi/available-plugins/plugins4.zip
[2024-01-26 23:11:01,250] [INFO] Unzipping...
[2024-01-26 23:11:01,641] [INFO] 2 file(s) were changed.
┌─[root@spitfire-pi3a] - [~] - [2024-01-26 23:11:03]
└─[0] pwnagotchi plugins list
Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 230, in <module>
    pwnagotchi_cli()
  File "/usr/local/bin/pwnagotchi", line 147, in pwnagotchi_cli
    rc = plugins_cmd.handle_cmd(args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/cmd.py", line 85, in handle_cmd
    return list_plugins(args, config)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/cmd.py", line 202, in list_plugins
    max_len = max(map(len, max_len_list))
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: max() arg is an empty sequence
```

Ravioli5, 2024-01-28:
You have to reopen an issue if you want others to see it @spitfire

spitfire, 2024-01-28:
Thanks @Ravioli5 I didn't notice it was closed. I can see no way of reopening it, so just pinging @jayofelony.

jayofelony, 2024-01-28:
Try not as root, but as pi with `sudo pwnagotchi plugins list`

spitfire, 2024-01-28:
This time it doesn't throw exceptions, but it still doesn't work:

```
┌─[pi@spitfire-pi3a] - [~] - [2024-01-28 14:01:03]
└─[0] sudo pwnagotchi plugins list
---------------------------------
|Plugin| Version |  Active  |    Status     |
---------------------------------
---------------------------------
[2024-01-28 14:01:11,260] [INFO] Maybe try: pwnagotchi plugins update
┌─[pi@spitfire-pi3a] - [~] - [2024-01-28 14:01:12]
└─[1] sudo pwnagotchi plugins update
[2024-01-28 14:01:24,585] [INFO] Downloading plugins from https://github.com/jayofelony/pwnagotchi-torch-plugins/archive/master.zip to /usr/local/share/pwnagotchi/available-plugins/plugins0.zip
[2024-01-28 14:01:26,529] [INFO] Unzipping...
[2024-01-28 14:01:26,610] [INFO] 2 file(s) were changed.
[2024-01-28 14:01:26,612] [INFO] Downloading plugins from https://github.com/evilsocket/pwnagotchi-plugins-contrib/archive/master.zip to /usr/local/share/pwnagotchi/available-plugins/plugins1.zip
[2024-01-28 14:01:28,350] [INFO] Unzipping...
[2024-01-28 14:01:28,424] [INFO] 3 file(s) were changed.
[2024-01-28 14:01:28,426] [INFO] Downloading plugins from https://github.com/tisboyo/pwnagotchi-pisugar2-plugin/archive/master.zip to /usr/local/share/pwnagotchi/available-plugins/plugins2.zip
[2024-01-28 14:01:30,118] [INFO] Unzipping...
[2024-01-28 14:01:30,152] [INFO] Downloading plugins from https://github.com/nullm0ose/pwnagotchi-plugin-pisugar3/archive/master.zip to /usr/local/share/pwnagotchi/available-plugins/plugins3.zip
[2024-01-28 14:01:31,927] [INFO] Unzipping...
[2024-01-28 14:01:31,969] [INFO] 1 file(s) were changed.
[2024-01-28 14:01:31,970] [INFO] Downloading plugins from https://github.com/Sniffleupagus/pwnagotchi_plugins/archive/master.zip to /usr/local/share/pwnagotchi/available-plugins/plugins4.zip
[2024-01-28 14:01:33,803] [INFO] Unzipping...
[2024-01-28 14:01:33,865] [INFO] 2 file(s) were changed.
┌─[pi@spitfire-pi3a] - [~] - [2024-01-28 14:01:34]
└─[0] sudo pwnagotchi plugins list
---------------------------------
|Plugin| Version |  Active  |    Status     |
---------------------------------
---------------------------------
[2024-01-28 14:01:44,458] [INFO] Maybe try: pwnagotchi plugins update
┌─[pi@spitfire-pi3a] - [~] - [2024-01-28 14:01:45]
└─[1]
```

jayofelony, 2024-01-28:
and when you do `sudo pwnagotchi plugins update && sudo pwnagotchi plugins list` ?

jayofelony, 2024-01-28:
Okay thank you for the heads up, I will rollback some changes I did there.