jayofelony/pwnagotchi #31: White on Black Screen

xstrex, 2024-01-26:
Bit of an odd one, I'm hesitant to open an issue, because I seem to be the only one with it. 
Base install of v2.7.7 on a RPi 02w, functionally everything's working great, but the display is inverted. 
[image]

My config.toml contains the following:
```yml
ui.display.enabled = true
ui.display.type = "waveshare_2"
ui.display.color = "black"
ui.display.rotation = 180
```

If set `ui.display.color` to `white` I get a blank screen that flickers every 5sec or so. 

My pwnagotchi-plugins-contrib contains the following:
```
age.py          display-password.py  exp_stats.json    internet-connection.py  mycracked_pw.py  __pycache__  screen_refresh.py
auto_backup.py  exp.py               gpio_shutdown.py  memtemp-plus.py         pijuice_ups.py   quickdic.py  show_pwd.py
```

None of which, I believe, would cause the screen to invert. 

Using a waveshare v2. 

For testing I installed v2.7.6 on a different SD card, installed it in a RPi 4 with a waveshare 2.7" hat (the one with the 4 buttons), and didn't install any plugins- just base install, and following lines in `config.toml`
```yml
ui.display.enabled = true
ui.display.type = "waveshare27inch"
ui.display.color = "black"
```
It's display is also inverted. 

I've got no idea why this is happening, if it's a bug or a 'feature'. Can't seem to find any documentation on it, and figured I'd ask here. Any idea what's going on? 

Happy to drop configs/logs/whatever, if I have to reinstall it's not the end of the world, just inconvenient.

xstrex, 2024-01-26:
It seems that swapping these two values in this file seems to fix the issue. 
https://github.com/jayofelony/pwnagotchi-bookworm/blob/04435229febb882b0a05a0518cb1029a1fdb9ce7/pwnagotchi/ui/view.py#L19
https://github.com/jayofelony/pwnagotchi-bookworm/blob/04435229febb882b0a05a0518cb1029a1fdb9ce7/pwnagotchi/ui/view.py#L20

jayofelony, 2024-01-26:
Yep, its how I want my image to be.

closed 2024-01-26 by jayofelony