jayofelony/pwnagotchi #36: [BUG] 2.13in V4 display not drawing properly

xslendix, 2024-01-28:
**Describe the bug**
The display is not drawing properly. The display hardware is fine, checked with official waveshare demo.
**To Reproduce**
Steps to reproduce the behavior:
1. Connect waveshare V4 display
2. Flash image with ws4 display in it
3. dafaq?????

**Expected behavior**
The stuff to be drawn properly
**Screenshots**
[image]

**Environment (please complete the following information):**
 - Latest release at time of issue
 - RPi Zero 2 W

jayofelony, 2024-01-28:
Can you show me your config.toml `ui.display` part?

xslendix, 2024-01-28:
```toml
ui.display.enabled = true
ui.display.type = "waveshare_4"
ui.display.color = "black"
```

also tried color white and waveshare_3

jayofelony, 2024-01-28:
You could remove the .color setting that is not used with waveshare_3 or 4. Other than that it's ok. Are you sure it is the waveshare V4, not the 2.13b V4?

ProMate04, 2024-01-29:
[image]
I would like to second this. I cannot get my Waveshare 2.13in V4 to work either. I would also like to ask if there is any way to know if I have the B type display?

jayofelony, 2024-01-29:
> ![IMG_1581](https://private-user-images.githubusercontent.com/37308842/300545207-db6afcb6-2cef-4d60-9f1a-9ebc0aacc075.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY1NTY2NjgsIm5iZiI6MTcwNjU1NjM2OCwicGF0aCI6Ii8zNzMwODg0Mi8zMDA1NDUyMDctZGI2YWZjYjYtMmNlZi00ZDYwLTlmMWEtOWViYzBhYWNjMDc1LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTI5VDE5MjYwOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFjMjliOTE3OTBlOTRlMzc0ZDIyODUyYTgzZmQxYWJlN2U1NzE2YmVhNjg2ZTViODFjZGZmN2VjN2JkNGVmNDgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.TwVqd4wO_w-yKNu3rFNfZU1uazlwZqXl0Sj1CX37VNc) I would like to second this. I cannot get my Waveshare 2.13in V4 to work either. I would also like to ask if there is any way to know if I have the B type display?

Did you take into AUTO yet, because MANU does not do much with refreshing the display.

ProMate04, 2024-01-29:
Yes, I just tried, but the display is still not right. And I have noticed another issue, seems like my Pi cannot find the wlan0mon interface? Basically even in AUTO mode, it seems like it can't see any networks, and the program just freezes. I checked the logs and found this: 
[image]

jayofelony, 2024-01-30:
> Yes, I just tried, but the display is still not right. And I have noticed another issue, seems like my Pi cannot find the wlan0mon interface? Basically even in AUTO mode, it seems like it can't see any networks, and the program just freezes. I checked the logs and found this: ![image](https://private-user-images.githubusercontent.com/37308842/300613177-66a1f399-0aaf-4d96-b405-b872e65a5664.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY1OTUzMzgsIm5iZiI6MTcwNjU5NTAzOCwicGF0aCI6Ii8zNzMwODg0Mi8zMDA2MTMxNzctNjZhMWYzOTktMGFhZi00ZDk2LWI0MDUtYjg3MmU2NWE1NjY0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTMwVDA2MTAzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRhNDg5M2ZiNWVmOWJhZTEzNTU4ZmRlZDA0YjM4YzhkOGVlMTkwOTIzZDAyOTcyYWE3MGQ0ZGM4MmIxNTM1YTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Hjc1n-gPDVzGmz99Pwackh8vTqylDbMyUNitbg2BpEE)
> 
> This is a RPi Zero 2 W Rev 1.0 board

This error you can ignore this is thrown right after boot, because I delayed creation of monitor mode so the wifi firmware wont crash.

The notfinding anything, did you happen to try and update it by using apt-get? If so, good luck reflashing.

ProMate04, 2024-01-30:
I actually have not, all I did was get all the saved files from my old unit, and put those back on, but I have not updated anything. I am reflashing right now, since I realised that I overwritten the fingerprint file as well, which I did not want to since my old gotchi was using the stock firmware. 

> You could remove the .color setting that is not used with waveshare_3 or 4. 

Could you please explain what this means exactly? Can I not use a white background with black text on it anymore?

jayofelony, 2024-01-30:
> I actually have not, all I did was get all the saved files from my old unit, and put those back on, but I have not updated anything. I am reflashing right now, since I realised that I overwritten [...]

This .color setting is for a specific function to certain displays ie waveshare v1and 2. If you want to change the background you have to manually edit view.py file with each update I do.

ProMate04, 2024-01-30:
I see. I found the file, and edited it properly, it works like this but it is a bit jank. Eh, it happens, still really cool tho. Going back to the original problem tho, it actually works fine for me with waveshare_4, I don't know where the problem came from orogonally. Anyways, thank you for the help!

jayofelony, 2024-01-30:
As soon as waveshare have updated files I will update them ASAP. The libraries come straight from them and are unaltered.

xslendix, 2024-02-01:
> ![IMG_1581](https://private-user-images.githubusercontent.com/37308842/300545207-db6afcb6-2cef-4d60-9f1a-9ebc0aacc075.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoic [...]

Unrelated, but may I please know what case that is?

Also, I will try and manually drop in the library from waveshare into my installation of pwnagotchi when I get the chance.

ProMate04, 2024-02-01:
> Unrelated, but may I please know what case that is?

Sure thing! It's an old model from thingiverse, it has a bunch of options for different displays and configurations, like slimagotchi and stuff like that. 
Link: https://www.thingiverse.com/thing:3920904

xslendix, 2024-02-01:
Ok so I updated the library manually but it didn't work sadly. Same result.

xslendix, 2024-02-01:
Added some logger calls on the drawing function and clearing function, they are never called. initialize is tho

xslendix, 2024-02-01:
Updating worked, had bad name in config.

jayofelony, 2024-02-01:
There is no new library yet

xslendix, 2024-02-01:
I think I have an idea for a fix, call epd.display with a buffer full of the background colour when initialized. I think this will solve the issue because I've noticed the display to look good after a couple of draw calls.

jayofelony, 2024-02-12:
`ui.display.fps = 1` 

Try this

Tiebe, 2024-03-02:
I used waveshare_4 and modified the file a bit so it does a full refresh on startup instead of instantly starting with partial refreshes. It also does a full refresh every 50 partial refreshes (not entirely sure if this is needed, but i thought i read somewhere that that's good for the display).

`sudo nano /usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/waveshare2in13_V4.py`

```
import logging

import pwnagotchi.ui.fonts as fonts
from pwnagotchi.ui.hw.base import DisplayImpl

class WaveshareV4(DisplayImpl):
    def __init__(self, config):
        super(WaveshareV4, self).__init__(config, 'waveshare_4')
        self.sinceLastFullRefresh = 10000

    def layout(self):
        fonts.setup(10, 9, 10, 35, 25, 9)
        self._layout['width'] = 250
        self._layout['height'] = 122
        self._layout['face'] = (0, 40)
        self._layout['name'] = (5, 20)
        self._layout['channel'] = (0, 0)
        self._layout['aps'] = (28, 0)
        self._layout['uptime'] = (185, 0)
        self._layout['line1'] = [0, 14, 250, 14]
        self._layout['line2'] = [0, 108, 250, 108]
        self._layout['friend_face'] = (0, 92)
        self._layout['friend_name'] = (40, 94)
        self._layout['shakes'] = (0, 109)
        self._layout['mode'] = (225, 109)
        self._layout['status'] = {
            'pos': (125, 20),
            'font': fonts.status_font(fonts.Medium),
            'max': 20
        }
        return self._layout

    def initialize(self):
        logging.info("initializing waveshare v2in13_V4 display")
        from pwnagotchi.ui.hw.libs.waveshare.v2in13_V4.epd2in13_V4 import EPD
        self._display = EPD()
        self._display.init()
        self._display.Clear(0xFF)

    def render(self, canvas):
        buf = self._display.getbuffer(canvas)
        if self.sinceLastFullRefresh > 50:
            self._display.display(buf)
            self.sinceLastFullRefresh = 0
        else:
            self._display.displayPartial(buf)
            self.sinceLastFullRefresh += 1

    def clear(self):
        self._display.Clear(0xFF)

```

rai68, 2024-03-05:
> I used waveshare_4 and modified the file a bit so it does a full refresh on startup instead of instantly starting with partial refreshes. It also does a full refresh every 50 partial refreshes (not [...]

You should make PR with that if it works

Tiebe, 2024-03-05:
> You should make PR with that if it works

Will do later today!

jayofelony, 2024-03-05:
Awesome!

closed 2024-03-27 by jayofelony