jayofelony/pwnagotchi #39: [BUG] Since upgrading to 2.8.0. now it's not handling the wlan0mon disappearing

rsun0525, 2024-01-30:
**Describe the bug**
When it was running 2.7.9, it seemed to handle the condition of wlan0mon disappearing properly by restarting whatever, etc. but since going to 2.8.0, it seems to get stuck and I have to manually do it again to fix it.

Hopefully you'll be able to see something in the pwnagotchi.log file
[pwnagotchi.log](https://github.com/jayofelony/pwnagotchi-bookworm/files/14105358/pwnagotchi.log)

jayofelony, 2024-01-31:
Nothing changed in that regard, so can't help.

jayofelony, 2024-02-01:
@rsun0525 I am reopening this issue. When it is booted can you run `sudo dmesg | grep brcmf`for me please. ANd show me printscreen of the output.

rsun0525, 2024-02-01:
pi@pwnagotchi400:~ $ sudo dmesg | grep brcmf
pi@pwnagotchi400:~ $ sudo dmesg | grep brcm
[    1.214060] irq_brcmstb_l2: registered L2 intc (/soc/interrupt-controller@7ef00100, parent irq: 27)
[    1.224347] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.224410] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.224501] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    1.224592] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x00ffffffff -> 0x0400000000
[    1.225148] brcm-pcie fd500000.pcie: setting SCB_ACCESS_EN, READ_UR_MODE, MAX_BURST_SIZE
[    1.225599] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    1.340239] brcm-pcie fd500000.pcie: link up, 5.0 GT/s PCIe x1 (SSC)
[    1.343517] brcm-pcie fd500000.pcie: clkreq control enabled
[    2.469105] brcmstb-i2c fef04500.i2c:  @97500hz registered in polling mode
[    2.469983] brcmstb-i2c fef09500.i2c:  @97500hz registered in polling mode
[    9.470339] Bluetooth: hci0: BCM4345C5 'brcm/BCM4345C5.raspberrypi,400.hcd' Patch
pi@pwnagotchi400:~ $

Incognisco, 2024-02-01:
Not sure if it helps, but mine has the same issue sometimes.

[image]

jayofelony, 2024-02-01:
> Not sure if it helps, but mine has the same issue sometimes. [...]

You have the best wifi chip there is, the set channel failed can be ignored. You could even disable fix_services, it is set to restart at that error but it'll still work if you ignore it apparently.

jayofelony, 2024-02-01:
> pi@pwnagotchi400:~ $ sudo dmesg | grep brcmf pi@pwnagotchi400:~ $ sudo dmesg | grep brcm [ 1.214060] irq_brcmstb_l2: registered L2 intc (/soc/interrupt-controller@7ef00100, parent irq: 27) [ [...]

And what do you do to fix it?

jayofelony, 2024-02-01:
Because there is simply no wireless interface to begin with at this point.

jayofelony, 2024-02-01:
I will try a fresh flash with 2.7.9 asap on my pi 4.

rsun0525, 2024-02-01:
That's what's really puzzling, is that it got a lot more stable doing a fresh flash to 2.7.9 but after upgrading to 2.8.0, it became unstable, looking at your change log, don't see why this would of been a problem.  I can do a fresh flash again if you like and then manually kick off the 2.8.0 (unless you prefer me to let it sit there and automatically upgrade).

jayofelony, 2024-02-01:
I do know of an issue with the Pi 4 wifi firmware of nexmon, I am currently in talks with them about a fix hopefully. I will know more about that tomorrow.

jayofelony, 2024-02-01:
[image]
fresh flash of 2.7.9 works out the box with a buddy of mine, just flashed it.

Ravioli5, 2024-02-02:
> I do know of an issue with the Pi 4 wifi firmware of nexmon, I am currently in talks with them about a fix hopefully. I will know more about that tomorrow.

I have been having problems with this again and can't fix it. Sudo monstart isn't working and pwnkill works (for a few mins) . Im having a similar error as @rsun0525 as well and I'm on raspi 4.

Ravioli5, 2024-02-02:
I'm getting this.

pi@spark:~ $ sudo dmesg | grep brcmf
[    7.814908] brcmfmac: loading out-of-tree module taints kernel.
[    7.982876] brcmfmac: F1 signature read @0x18000000=0x15264345
[    8.038403] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    8.039409] usbcore: registered new interface driver brcmfmac
[    8.525091] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    8.539485] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan 26
2024 15:07:50 version 7.45.206 (nexmon.org: 2.2.2-514-gf195c-1) FWID 01-88ee44ea[    9.874484] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   10.402341] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   11.308401] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[   11.331508] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[   11.331529] brcmfmac: brcmf_mon_add_vif: brcmf_mon_add_vif called
[   11.331534] brcmfmac: brcmf_mon_add_vif: Adding vif "wlan0mon"
[  463.970708] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[  467.028351] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[  467.028385] brcmfmac: brcmf_vif_add_validate: ... there is already a monitor
interface, returning EOPNOTSUPP
[  467.028402] ieee80211 phy0: brcmf_cfg80211_add_iface: iface validation failed: err=-95
[  554.368570] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[  557.397637] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[  557.397655] brcmfmac: brcmf_vif_add_validate: ... there is already a monitor
interface, returning EOPNOTSUPP
[  557.397707] ieee80211 phy0: brcmf_cfg80211_add_iface: iface validation failed: err=-95

Ravioli5, 2024-02-02:
That's what I get at first... I have not done apt get or apt upgrade or update since flashing either. Bootloader is up to date too. Fix_services isn't working either...

jayofelony, 2024-02-02:
Like the error says, there is a monitor interface already it seems

Ravioli5, 2024-02-02:
This happens until the wlan0mon goes down (or nexmon I'm not really sure which one)

Ravioli5, 2024-02-02:
I found this in the thing as well. 

[  557.397707] ieee80211 phy0: brcmf_cfg80211_add_iface: iface validation failed: err=-95
[ 2712.856823] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
[ 2712.941511] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing
to do.
[ 2712.941529] ieee80211 phy0: brcmf_net_mon_stop: BRCMF_C_SET_MONITOR error (-5)
[ 2713.141342] brcmfmac: F1 signature read @0x18000000=0x15264345
[ 2713.145532] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[ 2713.474046] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[ 2713.477451] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan 26
2024 15:07:50 version 7.45.206 (nexmon.org: 2.2.2-514-gf195c-1) FWID 01-88ee44ea

jayofelony, 2024-02-02:
Yep that is the crash I am talking to the nexmon devs about

Ravioli5, 2024-02-02:
Ah, ok

jayofelony, 2024-02-02:
I will close this issue, as there is no fix and perhaps not for a while also. It is what it is I guess for now.

closed 2024-02-02 by jayofelony

Ravioli5, 2024-02-02:
Oh ok

Ravioli5, 2024-02-02:
Oh, I guess I'll have to live with it or wait for a while then.