jayofelony/pwnagotchi #40: [BUG] Can't see assoc face

jagp8, 2024-01-31:
Hi again, I was checking the documentation and I found this image

https://i.imgur.com/c7xh4hN.png

And I realased that on my pwnagotchi the little faco of the assoc doesn't appear, it just appear the 4 bars, the name (alpha in this image) and the numbers

Is that normal?

jayofelony, 2024-01-31:
Yes because I removed it from the display, as people tend to have custom faces. And if the peer doesnt have the exact same custom faces installed it would show an error in the place of the face

jayofelony, 2024-01-31:
I could place a default one, but whats the point of the face if it cant 'show' it's emotions.

Ravioli5, 2024-01-31:
I think default in this sense would mean what would be the emotion but with the standard pwnagotchi faces so that the peer's emotion would be sent to the pwnagotchi and the pwnagotchi would display the emotion as one of its built in faces as the peer face. This way even of it is a custom face a face would be shown along with its emotion. Also this is more of an enhancement rather than a bug.

jagp8, 2024-01-31:
Yep, exactly, I meant the built in emotions.
I didn't know it was disabled on pruporse, yes then it would be an enhancement

closed 2024-03-27 by jayofelony