jayofelony/pwnagotchi #42: [BUG] text problem

Ravioli5, 2024-02-01:
**Describe the bug**
Strange text can appear in the place of the manual or ai. 

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on pwnagotchi
2. open web ui
3. Wait for a few hours, repeat process several times to see this.

**Expected behavior**
A clear and concise description of what you expected to happen.
???Strange text appeared

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Environment (please complete the following information):**
 - Pwnagotchi version - latest, 2.8.0
 - OS version bookworm
 - Type of hardware rpi 4
 - Any additional hardware used ethernet cable, power source

**Additional context**
In the example it shows Tr01 but it has also shown other things and also has shown ai instead of AI. I know this isa very small bug and it's not even that annoying but I wondered why it was there.

jayofelony, 2024-02-01:
You have a plugin called Miyagi enabled, good day.

closed 2024-02-01 by jayofelony

Ravioli5, 2024-02-01:
Ohhh... sorry for wasting your time then

jayofelony, 2024-02-01:
All good, just happy to know there isnt anything wrong.. again hahaha

Ravioli5, 2024-02-01:
😅