jayofelony/pwnagotchi #43: Make the dark theme a choice in the config

xslendix, 2024-02-01:
Would be cool to have in the config something like `ui.colourscheme = 'dark'` or light instead of modifying the python file.

xslendix, 2024-02-01:
I think I will work on this as I really want this

jayofelony, 2024-02-01:
tried it, and failed. Cannot be done.

xslendix, 2024-02-01:
> tried it, and failed. Cannot be done.

Let me try my hand at it, see if I can do smth

jayofelony, 2024-02-01:
The display settings are loaded before anything of pwnagotchi is loaded, so you cannot make a call to config. As it will return a None type.

jagp8, 2024-02-02:
Hmmmm crazy idea here, if you can't call the config but you can call the clock yo could make day/night mode, I mean white background from 8am to 8pm and dark background from 8pm to 8am

jayofelony, 2024-02-02:
Yeah, but it would only change if you were to restart/reboot. As this part of the script is only used once at start of pwnagotchi.

jagp8, 2024-02-02:
I replaced the color options with this just for fun

if int(time.strftime("%M"))%2 == 0:
    WHITE = 0x00
    BLACK = 0xFF
else:
    WHITE = 0xFF
    BLACK = 0x00

It's working as expected, it changes from color depending on the minutes being odd or even, and just to point out that it even changes when I do a pwnkill, i don't know if it should be the expected behaviour or should it only change in reboots

Passarb, 2024-02-07:
Here is a script that I found to invert the colors.  https://github.com/tPayne0647/pwnagotchi_screen_color_invert

jayofelony, 2024-02-07:
I am not planning on implementing any scripts.

renewedbullet, 2024-02-08:
This should be closed the main purpose of the black background was for Jays image to stand out from others, if you want to change it you can edit view.py.

xslendix, 2024-02-09:
I do not think this issue should be closed because of that reason, as it is a selfish reason. Not everyone likes the dark theme, adding an option for it would be good. Default could still be dark theme but the choice being there matters in my opinion.

renewedbullet, 2024-02-09:
He has tried to add the option and it requires an astronomical amount of work to fix it correctly, plus Jay made it black to have his image be different than anyone else. You can google invert colors on Pwnagotchi and switch it back to white.

xslendix, 2024-02-09:
I do know how to switch the colour, but I do not wish to do it every time my pwnagotchi updates. This is the whole point of the issue, figure out and plan out how this configuration option should be and work.

PR #77 (2024-03-04): Adds invertable BLACK/WHITE to config

`ui.invert = true` to change from a BLACK default to WHITE (this image)
fixes  #43 
## How Has This Been Tested?
pi zero 2w

## Types of changes

- [ ] Bug fix (non-breaking change which fixes an issue)
- [x] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to change)
closed 2024-03-04 by jayofelony