jayofelony/pwnagotchi #49: [BUG] main.name will not update

lomelo-x, 2024-02-05:
**Describe the bug**
when updating main.name in config.toml, it is not updating on the device. I have updated via ssh as well as sftp neither are working

**To Reproduce:**
1. ssh into pwnagotchi
2. sudo nano /etc/pwnagotchi/config.toml
3. change main.name = "pwnagotchi" to main.name = "joe"
4. ctrl X, Yes, enter
5. sduo systemctl restart pwnagotchi
6. check device for name update
7. see name has not changed
8. cry

**Environment:**
 - Pwnagotchi version 2.8.1
 - pi02w

jayofelony, 2024-02-05:
`sudo pwnagotchi --debug`

lomelo-x, 2024-02-05:
> `sudo pwnagotchi --debug`
> 
> Please make a screenshot of the errors shown, or try to fix errors yourself.

@jayofelony Thanks, I had an underscore in the name 🤦‍♂️ I wasn't aware it was an invalid character.