jayofelony/pwnagotchi #59: [BUG] after update, reboots only in manual mode

Ravioli5, 2024-02-09:
Whenever the pwnagotchi is booted it boots into auto mode, and then says wifi recon flipped and then goes to manual mode, before it loads the ai. It then loads the ai IN MANUAL MODE! It just become a fusion of the two modes and I'm not sure what happened. The booting time is quite longer too. I'm not sure why this happened 

Steps to reproduce the behavior:
1. update pwnagotchi via auto update 
2. Open web ui
3. look at screen with eyes

**Expected behavior**
A strange thing happens on screen.

**Videos**

Manual mode problem
https://github.com/jayofelony/pwnagotchi-bookworm/assets/157115073/c6f6d704-3044-4f64-b7a3-6eeaa92060bd

Handshake problem(it's a long video it's not letting me upload, not sure where to upload it....)

**Environment (please complete the following information):**
 - 1.8.2 bookworm
 - Raspberry pi 4
 - ethernet cable?

**Additional context**
The pwnagotchi is unable to collect handshakes when this happens. Either this happens or the pwnagotchi just becomes unable to collect handshakes even though it was doing well before. Not sure why this is happening... I really hope this is a problem on my end but I've tried to fix it with no results. Plz help

jayofelony, 2024-02-09:
You are powering it from pc to otg usb, power it from pc to other usb port. Or unplug it completely from pc/laptop and use a battery to power it.

Ravioli5, 2024-02-09:
It's not plugged into a pc though... I'll try connecting it to a different outlet though.

Ravioli5, 2024-02-09:
Ok I tried it with the same outlet and it started working again. The boot time is still long for some reason but the wifi recon flipped message still comes before the ai is loaded (hasn't happened before) and happens everytime instead of after a few hours or so and only rarely. Here's a video of it happening. 

https://github.com/jayofelony/pwnagotchi-bookworm/assets/157115073/ab62740a-6fd4-4a31-bdc2-797dc92c09a6

jayofelony, 2024-02-09:
https://discord.gg/vrbMsCMJ

Please get on discord if you can. This is not a bug, you are the only one experiencing it. You must have done something out of the ordinary.

Ravioli5, 2024-02-12:
Ok, I will check on discord

Ravioli5, 2024-02-12:
I'll close for now

closed 2024-02-12 by Ravioli5