jayofelony/pwnagotchi #60: [Enhancement] Auto bt restart when internet connection is lost

jagp8, 2024-02-13:
Okay, so I've noticed that after some time of being connected to my phone's Bluetooth, the internet stops working.

The plugins for Bluetooth tethering (bt-tether) and internet connection (internet-connection) indicate that I'm still connected to Bluetooth and the internet, respectively. However, when I enter via SSH and try to ping Google, it returns a timeout.

The solution to that is simply turning off and then on the Bluetooth tethering. So, I thought, "How cool would it be to have a plugin or something that, every 15 minutes (or a configurable amount of time), does a ping to Google, and if it can't reach it, automatically turns off Bluetooth tethering, waits for 10 seconds, and then turns it back on."

So, I started testing with Python, importing os, and running os.system("ping -c 1 google.com"), which started to work as I wanted. After that, I looked for the bt-tether plugin to see if I could take something from there, and now I'm lost in a sea of hopelessness and almost 600 lines of code.

I'll keep trying to work something out, but I'm starting to doubt myself. I don't know if I can make a plugin that turns on and off another plugin, or if it would be better to improve that one, but in both cases, I don't think my programming skills are high enough yet to achieve this task ^^U

jayofelony, 2024-02-13:
```py
def is_connected():
    try:
        # check DNS
        host = 'https://api.opwngrid.xyz/api/v1/uptime'
        r = requests.get(host, headers=None, timeout=(30.0, 60.0))
        if r.json().get('isUp'):
            return True
    except:
        pass
    return False
```

It is easier to use this I think

jagp8, 2024-02-13:
Not for me ^^U

In fact I sent the problem also to chat gpt and it created me a couple of functions to add at the end of the bt-tether.py
`
    def is_connected(self):
        try:
            # Verify with proper logic
            r = os.system("ping -c 1 google.com")
            if r==0:
                return True
        except Exception as e:
            logging.error(f"BT-TETHER: Error checking connection: {e}")
        return False

    def check_connection_and_restart_bt(self):
        try:
            # Verify internet connection
            if not self.is_connected():
                logging.info('BT-TETHER: No internet connectivity. Restarting Bluetooth...')
                # Stop Bluetooth
                bt_unit = SystemdUnitWrapper('bluetooth.service')
                if not bt_unit.disable():
                    logging.error("BT-TETHER: Can't disable bluetooth.service")
                    return
                #Wait
                time.sleep(10)
                # Start Bluetooth
                if not bt_unit.enable():
                    logging.error("BT-TETHER: Can't enable bluetooth.service")
                    return
        except Exception as e:
            logging.error(f"BT-TETHER: Error checking connection and restarting Bluetooth: {e}")
        #Restart timer for next verification
        threading.Timer(self.options['check_restart_timer'], self.check_connection_and_restart_bt).start()
`

Yo also have to add the last line after def on_load(self)

`
    def on_loaded(self):
        threading.Timer(self.options['check_restart_timer'], self.check_connection_and_restart_bt).start()
`

And in the config the new parameter

main.plugins.bt-tether.check_restart_timer = 900

JD-2006, 2024-02-27:
@jagp8 How about this? Create a python script...
```import subprocess

def check_internet():
    try:
        subprocess.run(["ping", "-c", "1", "8.8.8.8"], stdout=subprocess.PIPE, stderr=subprocess.PIPE, check=True)
        return True
    except subprocess.CalledProcessError:
        return False

def restart_service():
    try:
        subprocess.run(["sudo", "systemctl", "restart", "bluetooth"], check=True, stdout=subprocess.PIPE, stderr=subprocess.PIPE)
    except subprocess.CalledProcessError:
        pass  # Handle the error as needed

if __name__ == "__main__":
    if not check_internet():
        restart_service()
```

Then add a cron job to run every 15 mins.
crontab -e

Add...
*/15 * * * * sudo python3 /path/to/check_internet_and_restart_service.py

jagp8, 2024-02-29:
I'll try, thanks!!

rai68, 2024-03-26:
pinging once is a bad idea, since ARP exists

closed 2024-03-27 by jayofelony