jayofelony/pwnagotchi #61: [BUG] RPi4b doesnt seem to setup correctly, webui also wont work. Tried flashing multiple times

RealDogGuy, 2024-02-14:
**Describe the bug**
I plug my RPi4b to my pc with the USB C cable and its detected as RNDIS gadget and I can SSH into it normally but when I try to open the webui it doesnt work. Other commands like curl 0.0.0.0:8080 output: curl: (7) Failed to connect to 0.0.0.0 port 8080: Connection refused, and PWNlog just says inf pwngrid starting in peers, but nothing happens. in my config my pwnagotchi is name das panzer but in the cmd it shows as pi. I dont know if my config is working correctly. Any help?

**To Reproduce**
Steps to reproduce the behavior:
1. Plug Pwnagotchi into pc
2. Ssh into it
3. try to open webui which results in it not working or the other mentioned commands.

**Expected behavior**
Its expected to open the webui and function like a normal pwnagotchi.

**Screenshots**
[image]
[image]

**Environment (please complete the following information):**
 - Pwnagotchi version: 2.8.2
 - OS version ?
 - Type of hardware: Raspberry pi 4b 
 - Any additional hardware used.

**Additional context**
Add any other context about the problem here.

jayofelony, 2024-02-14:
open http://10.0.0.2:8080
if that does not work use `sudo pwnagotchi --debug` and fix any errors that it gives you.

RealDogGuy, 2024-02-14:
> open http://10.0.0.2:8080 if that does not work use `sudo pwnagotchi --debug` and fix any errors that it gives you.

I did that and its still going but I think its just looping the same message, which is:
```
DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 127.0.0.1:8666
[2024-02-14 07:00:31,978] [DEBUG] Starting new HTTP connection (1): 127.0.0.1:8666
DEBUG:urllib3.connectionpool:http://127.0.0.1:8666 "GET /api/v1/mesh/peers HTTP/1.1" 200 2
[2024-02-14 07:00:31,980] [DEBUG] http://127.0.0.1:8666 "GET /api/v1/mesh/peers HTTP/1.1" 200 2
DEBUG:websockets.client:< PING '' [0 bytes]
[2024-02-14 07:00:33,737] [DEBUG] < PING '' [0 bytes]
DEBUG:websockets.client:> PONG '' [0 bytes]
[2024-02-14 07:00:33,737] [DEBUG] > PONG '' [0 bytes]
DEBUG:websockets.client:% sending keepalive ping
[2024-02-14 07:00:34,222] [DEBUG] % sending keepalive ping
DEBUG:websockets.client:> PING cb a5 52 e0 [binary, 4 bytes]
[2024-02-14 07:00:34,222] [DEBUG] > PING cb a5 52 e0 [binary, 4 bytes]
DEBUG:websockets.client:< PONG cb a5 52 e0 [binary, 4 bytes]
[2024-02-14 07:00:34,223] [DEBUG] < PONG cb a5 52 e0 [binary, 4 bytes]
DEBUG:websockets.client:% received keepalive pong
[2024-02-14 07:00:34,224] [DEBUG] % received keepalive pong
DEBUG:urllib3.connectionpool:http://127.0.0.1:8081 "POST /api/session HTTP/1.1" 400 75
[2024-02-14 07:00:34,974] [DEBUG] http://127.0.0.1:8081 "POST /api/session HTTP/1.1" 400 75
INFO:root:error 400: could not find interface wlan0mon: no interface matching 'wlan0mon' found.
[2024-02-14 07:00:34,976] [INFO] error 400: could not find interface wlan0mon: no interface matching 'wlan0mon' found.
ERROR:root:[Fix_Services]SYSLOG wifi.recon flip fail: error 400: could not find interface wlan0mon: no interface matching 'wlan0mon' found.
[2024-02-14 07:00:34,977] [ERROR] [Fix_Services]SYSLOG wifi.recon flip fail: error 400: could not find interface wlan0mon: no interface matching 'wlan0mon' found.
DEBUG:urllib3.util.retry:Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
[2024-02-14 07:00:34,978] [DEBUG] Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
DEBUG:urllib3.util.retry:Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
[2024-02-14 07:00:34,979] [DEBUG] Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 127.0.0.1:8666
DEBUG:root:polling pwngrid-peer for peers ...
[2024-02-14 07:00:34,983] [DEBUG] Starting new HTTP connection (1): 127.0.0.1:8666
[2024-02-14 07:00:34,983] [DEBUG] polling pwngrid-peer for peers ...
DEBUG:urllib3.util.retry:Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
[2024-02-14 07:00:34,986] [DEBUG] Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
DEBUG:urllib3.util.retry:Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
[2024-02-14 07:00:34,987] [DEBUG] Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
DEBUG:urllib3.connectionpool:http://127.0.0.1:8666 "POST /api/v1/mesh/data HTTP/1.1" 200 16
DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 127.0.0.1:8666
[2024-02-14 07:00:34,989] [DEBUG] http://127.0.0.1:8666 "POST /api/v1/mesh/data HTTP/1.1" 200 16
[2024-02-14 07:00:34,993] [DEBUG] Starting new HTTP connection (1): 127.0.0.1:8666
DEBUG:urllib3.connectionpool:http://127.0.0.1:8666 "GET /api/v1/mesh/peers HTTP/1.1" 200 2
[2024-02-14 07:00:35,012] [DEBUG] http://127.0.0.1:8666 "GET /api/v1/mesh/peers HTTP/1.1" 200 2
Device "wlan0mon" does not exist.
ERROR:root:[Fix_Services ip link show wlan0mon]: CalledProcessError(1, 'ip link show wlan0mon')
[2024-02-14 07:00:35,043] [ERROR] [Fix_Services ip link show wlan0mon]: CalledProcessError(1, 'ip link show wlan0mon')
DEBUG:urllib3.util.retry:Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
[2024-02-14 07:00:35,044] [DEBUG] Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
DEBUG:urllib3.util.retry:Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
[2024-02-14 07:00:35,045] [DEBUG] Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 127.0.0.1:8081
[2024-02-14 07:00:35,047] [DEBUG] Starting new HTTP connection (1): 127.0.0.1:8081
DEBUG:urllib3.connectionpool:http://127.0.0.1:8081 "POST /api/session HTTP/1.1" 400 27
[2024-02-14 07:00:35,050] [DEBUG] http://127.0.0.1:8081 "POST /api/session HTTP/1.1" 400 27
INFO:root:error 400: module wifi is not running
[2024-02-14 07:00:35,052] [INFO] error 400: module wifi is not running
ERROR:root:[Fix_Services wifi.recon off] error  Exception('error 400: module wifi is not running')
[2024-02-14 07:00:35,053] [ERROR] [Fix_Services wifi.recon off] error  Exception('error 400: module wifi is not running')
INFO:root:[Fix_Services] recon paused. Now trying wlan0mon reload
[2024-02-14 07:00:35,053] [INFO] [Fix_Services] recon paused. Now trying wlan0mon reload
wlan0mon: ERROR while getting interface flags: No such device
[... cut ...]
DEBUG:urllib3.util.retry:Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
[2024-02-14 07:00:36,172] [DEBUG] Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
DEBUG:urllib3.util.retry:Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
[2024-02-14 07:00:36,173] [DEBUG] Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
```

RealDogGuy, 2024-02-14:
after i closed the cmd and reopened I ran pwnlog and it gave me this:
```
/home/pi/logs/pwngrid-peer.log <==
04:55:02 inf pwngrid v1.11.1 starting in peer mode ...
04:55:26 inf pwngrid v1.11.1 starting in peer mode ...
05:13:35 inf pwngrid v1.11.1 starting in peer mode ...
06:25:57 inf pwngrid v1.11.1 starting in peer mode ...
06:25:56 inf pwngrid v1.11.1 starting in peer mode ...
07:21:07 inf pwngrid v1.11.1 starting in peer mode ...
07:21:09 inf /etc/pwnagotchi/id_rsa found
07:21:09 war error while setting interface wlan0mon in monitor mode: No such device exists
07:21:10 !!! error while starting signaling: error while activating handle: No Such Device
07:21:40 inf pwngrid v1.11.1 starting in peer mode ...
07:21:42 inf /etc/pwnagotchi/id_rsa found
07:21:42 inf started beacon discovery and message routing (0 known peers)
07:21:42 inf peer pwnagotchi@36e1a52c74ba53dadc3650f3456f7771ce1e034e8743ae92aa4ac5a756f781b1 signaling is ready
07:21:42 inf pwngrid api starting on 127.0.0.1:8666 ...

```

jayofelony, 2024-02-14:
Yeah I dont know what it is with the error 400. It turned up somehow

renewedbullet, 2024-02-15:
> after i closed the cmd and reopened I ran pwnlog and it gave me this:
>
> ```
> /home/pi/logs/pwngrid-peer.log <==
> 04:55:02 inf pwngrid v1.11.1 starting in peer mode ...
> 04:55:26 inf pwngrid v1.11.1 [...]

400 error has been narrowed down to bookworm changing firmware usage, further testing and a fix is being worked on

bryzz42o, 2024-02-27:
id try going into your network router setting directly with your gotchi connected to your pc & if its sharing the internet it'll be listed the router devices, use that IP with :8080 to see the webUi but had the same 400 issue with this repo on a rpi4, the brcmf_plugin enabled along with enable-assoc & enable-deauth with a proper power supply seemed to help it going blind every 2 seconds

closed 2024-03-27 by jayofelony