jayofelony/pwnagotchi #62: SSH doesn't accept default credentials

srmo, 2024-02-14:
Hi, first time user here.
Setup on Raspberry Pi Zero 2W. 
Started in AUTO and collected some stuff. Pulled the power plug.
Connected via USB Data to Windows 11 using a RNDIS driver from the windows catalog.
Configured ethernet interface.
bettercap UI and 8080 UI working with default credentials.

Trying to ssh into the IP (same as used for the web uis, 10.0.0.2) leads to permission denied.
I've used the default credentials from the official pwnagotchi.ai site
pi/raspberry

What am I missing?

jayofelony, 2024-02-14:
webui:
changeme/changeme

bettercap ui:
pwnagotchi/pwnagotchi

ssh:
pi/raspberry

srmo, 2024-02-14:
Yeah, as mentioned, webui and bettercap works flawlessly. But ssh gives me Permission denied with these credentials.
I don't see what I could be doing wrong. 
Any further help appreciated. Is there some other way I could debug/fix this?

For completeness sake the output, just to make sure I'm not accidentally using key auth.
```
 ssh -o PreferredAuthentications=password -o PubkeyAuthentication=no -vvv pi@10.0.0.2
OpenSSH_9.2p1 Debian-2+deb12u2, OpenSSL 3.0.11 19 Sep 2023
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 19: include /etc/ssh/ssh_config.d/*.conf matched no files
debug1: /etc/ssh/ssh_config line 21: Applying options for *
debug2: resolve_canonicalize: hostname 10.0.0.2 is address
debug3: expanded UserKnownHostsFile '~/.ssh/known_hosts' -> '/home/srm/.ssh/known_hosts'
debug3: expanded UserKnownHostsFile '~/.ssh/known_hosts2' -> '/home/srm/.ssh/known_hosts2'
debug3: ssh_connect_direct: entering
debug1: Connecting to 10.0.0.2 [10.0.0.2] port 22.
debug3: set_sock_tos: set socket 3 IP_TOS 0x10
debug1: Connection established.
debug1: identity file /home/srm/.ssh/id_rsa type 0
debug1: identity file /home/srm/.ssh/id_rsa-cert type -1
debug1: identity file /home/srm/.ssh/id_ecdsa type -1
debug1: identity file /home/srm/.ssh/id_ecdsa-cert type -1
debug1: identity file /home/srm/.ssh/id_ecdsa_sk type -1
debug1: identity file /home/srm/.ssh/id_ecdsa_sk-cert type -1
debug1: identity file /home/srm/.ssh/id_ed25519 type -1
debug1: identity file /home/srm/.ssh/id_ed25519-cert type -1
debug1: identity file /home/srm/.ssh/id_ed25519_sk type -1
debug1: identity file /home/srm/.ssh/id_ed25519_sk-cert type -1
debug1: identity file /home/srm/.ssh/id_xmss type -1
debug1: identity file /home/srm/.ssh/id_xmss-cert type -1
debug1: identity file /home/srm/.ssh/id_dsa type -1
debug1: identity file /home/srm/.ssh/id_dsa-cert type -1
debug1: Local version string SSH-2.0-OpenSSH_9.2p1 Debian-2+deb12u2
debug1: Remote protocol version 2.0, remote software version OpenSSH_9.2p1 Debian-2+deb12u2
debug1: compat_banner: match: OpenSSH_9.2p1 Debian-2+deb12u2 pat OpenSSH* compat 0x04000000
debug2: fd 3 setting O_NONBLOCK
debug1: Authenticating to 10.0.0.2:22 as 'pi'
debug3: record_hostkey: found key type ED25519 in file /home/srm/.ssh/known_hosts:10
debug3: load_hostkeys_file: loaded 1 keys from 10.0.0.2
debug1: load_hostkeys: fopen /home/srm/.ssh/known_hosts2: No such file or directory
debug1: load_hostkeys: fopen /etc/ssh/ssh_known_hosts: No such file or directory
debug1: load_hostkeys: fopen /etc/ssh/ssh_known_hosts2: No such file or directory
debug3: order_hostkeyalgs: have matching best-preference key type ssh-ed25519-cert-v01@openssh.com, using HostkeyAlgorithms verbatim
debug3: send packet: type 20
debug1: SSH2_MSG_KEXINIT sent
debug3: receive packet: type 20
debug1: SSH2_MSG_KEXINIT received
debug2: local client KEXINIT proposal
debug2: KEX algorithms: sntrup761x25519-sha512@openssh.com,curve25519-sha256,curve25519-sha256@libssh.org,ecdh-sha2-nistp256,ecdh-sha2-nistp384,ecdh-sha2-nistp521,diffie-hellman-group-exchange-sha256,diffie-hellman-group16-sha512,diffie-hellman-group18-sha512,diffie-hellman-group14-sha256,ext-info-c,kex-strict-c-v00@openssh.com
debug2: host key algorithms: ssh-ed25519-cert-v01@openssh.com,ecdsa-sha2-nistp256-cert-v01@openssh.com,ecdsa-sha2-nistp384-cert-v01@openssh.com,ecdsa-sha2-nistp521-cert-v01@openssh.com,sk-ssh-ed25519-cert-v01@openssh.com,sk-ecdsa-sha2-nistp256-cert-v01@openssh.com,rsa-sha2-512-cert-v01@openssh.com,rsa-sha2-256-cert-v01@openssh.com,ssh-ed25519,ecdsa-sha2-nistp256,ecdsa-sha2-nistp384,ecdsa-sha2-nistp521,sk-ssh-ed25519@openssh.com,sk-ecdsa-sha2-nistp256@openssh.com,rsa-sha2-512,rsa-sha2-256
debug2: ciphers ctos: chacha20-poly1305@openssh.com,aes128-ctr,aes192-ctr,aes256-ctr,aes128-gcm@openssh.com,aes256-gcm@openssh.com
debug2: ciphers stoc: chacha20-poly1305@openssh.com,aes128-ctr,aes192-ctr,aes256-ctr,aes128-gcm@openssh.com,aes256-gcm@openssh.com
debug2: MACs ctos: umac-64-etm@openssh.com,umac-128-etm@openssh.com,hmac-sha2-256-etm@openssh.com,hmac-sha2-512-etm@openssh.com,hmac-sha1-etm@openssh.com,umac-64@openssh.com,umac-128@openssh.com,hmac-sha2-256,hmac-sha2-512,hmac-sha1
debug2: MACs stoc: umac-64-etm@openssh.com,umac-128-etm@openssh.com,hmac-sha2-256-etm@openssh.com,hmac-sha2-512-etm@openssh.com,hmac-sha1-etm@openssh.com,umac-64@openssh.com,umac-128@openssh.com,hmac-sha2-256,hmac-sha2-512,hmac-sha1
debug2: compression ctos: none,zlib@openssh.com,zlib
debug2: compression stoc: none,zlib@openssh.com,zlib
debug2: languages ctos:
debug2: languages stoc:
debug2: first_kex_follows 0
debug2: reserved 0
debug2: peer server KEXINIT proposal
debug2: KEX algorithms: sntrup761x25519-sha512@openssh.com,curve25519-sha256,curve25519-sha256@libssh.org,ecdh-sha2-nistp256,ecdh-sha2-nistp384,ecdh-sha2-nistp521,diffie-hellman-group-exchange-sha256,diffie-hellman-group16-sha512,diffie-hellman-group18-sha512,diffie-hellman-group14-sha256,kex-strict-s-v00@openssh.com
debug2: host key algorithms: rsa-sha2-512,rsa-sha2-256,ecdsa-sha2-nistp256,ssh-ed25519
debug2: ciphers ctos: chacha20-poly1305@openssh.com,aes128-ctr,aes192-ctr,aes256-ctr,aes128-gcm@openssh.com,aes256-gcm@openssh.com
debug2: ciphers stoc: chacha20-poly1305@openssh.com,aes128-ctr,aes192-ctr,aes256-ctr,aes128-gcm@openssh.com,aes256-gcm@openssh.com
[... cut ...]
debug3: start over, passed a different list publickey,password
debug3: preferred password
debug3: authmethod_lookup password
debug3: remaining preferred:
debug3: authmethod_is_enabled password
debug1: Next authentication method: password
pi@10.0.0.2's password:
debug3: send packet: type 50
debug2: we sent a password packet, wait for reply
debug3: receive packet: type 51
debug1: Authentications that can continue: publickey,password
Permission denied, please try again.
pi@10.0.0.2's password:
```

jayofelony, 2024-02-14:
You must mispell it or something, it is working with pi and raspberry

srmo, 2024-02-15:
.... oh my god. I tried with the serial console to rule out any ssh client shenanigans. Same problem. Then it hit me.
For some reason, I deployed my user config when writing the image to SD via the raspbi imager. I'm 99.9999..% sure I disabled applying a custom config. Well, seems that 0.0000000...1% are enough to prove me wrong.
Sorry for the confusion.

closed 2024-02-15 by srmo