jayofelony/pwnagotchi #65: [BUG] Build Image Errors

hamzenis, 2024-02-19:
**Describe the bug**
I can't build the image from the make file and got stuck with this error:

```bash
root@ubuntu20.04:~/pwnagotchi-bookworm# make image
make: command: Command not found
make: Circular dist/pwnagotchi-2.8.4.tar.gz <- pwnagotchi dependency dropped.
cd builder && ~/pwnagotchi/packer init pwnagotchi.json.pkr.hcl && sudo  ~/pwnagotchi/packer build -var "pwn_hostname=pwnagotchi" -var "pwn_version=2.8.4" pwnagotchi.json.pkr.hcl
Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi: output will be in this color.

==> Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi: Retrieving rootfs_archive
==> Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi: Trying https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2023-12-11/2023-12-11-raspios-bookworm-arm64-lite.img.xz?archive=false
==> Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi: Trying https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2023-12-11/2023-12-11-raspios-bookworm-arm64-lite.img.xz?archive=false&checksum=file%3Ahttps%3A%2F%2Fdownloads.raspberrypi.org%2Fraspios_lite_arm64%2Fimages%2Fraspios_lite_arm64-2023-12-11%2F2023-12-11-raspios-bookworm-arm64-lite.img.xz.sha256
==> Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi: https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2023-12-11/2023-12-11-raspios-bookworm-arm64-lite.img.xz?archive=false&checksum=file%3Ahttps%3A%2F%2Fdownloads.raspberrypi.org%2Fraspios_lite_arm64%2Fimages%2Fraspios_lite_arm64-2023-12-11%2F2023-12-11-raspios-bookworm-arm64-lite.img.xz.sha256 => /root/.cache/packer/954fc95274fbf1dd9b273ffe9fa6e2280d09ec58.xz
    Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi: unpacking /root/.cache/packer/954fc95274fbf1dd9b273ffe9fa6e2280d09ec58.xz to ../../../pwnagotchi-rpi-bookworm-2.8.4-arm64.img
    Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi: unpacking with custom command: [unxz /tmp/image4064044028/954fc95274fbf1dd9b273ffe9fa6e2280d09ec58.xz]
    Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi: resizing the image file ../../../pwnagotchi-rpi-bookworm-2.8.4-arm64.img to 9G
    Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi: expanding partition no. 2 on the image ../../../pwnagotchi-rpi-bookworm-2.8.4-arm64.img
    Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi: mapping image ../../../pwnagotchi-rpi-bookworm-2.8.4-arm64.img to free loopback device
    Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi: image ../../../pwnagotchi-rpi-bookworm-2.8.4-arm64.img mapped to /dev/loop0
    Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi: running resize2fs on /dev/loop0p2
    Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi: mounting /dev/loop0p2 to /tmp/2461357824
    Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi: mounting /dev/loop0p1 to /tmp/2461357824/boot/firmware
    Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi: running extra setup
    Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi: mounting /dev with: [mount --bind /dev /tmp/2461357824/dev]
    Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi: mounting /devpts with: [mount -t devpts /devpts /tmp/2461357824/dev/pts]
    Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi: mounting proc with: [mount -t proc proc /tmp/2461357824/proc]
    Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi: mounting binfmt_misc with: [mount -t binfmt_misc binfmt_misc /tmp/2461357824/proc/sys/fs/binfmt_misc]
    Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi: mounting sysfs with: [mount -t sysfs sysfs /tmp/2461357824/sys]
==> Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi: Failed to find binfmt_misc for /usr/libexec/qemu-binfmt/aarch64-binfmt-P under /proc/sys/fs/binfmt_misc
==> Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi: remove /tmp/2461357824/usr/libexec/qemu-binfmt/aarch64-binfmt-P: no such file or directory
    Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi: optional (please ignore) `fuser -k` failed with exit status 1:
Build 'Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi' errored after 32 seconds 713 milliseconds: build was halted

==> Wait completed after 32 seconds 713 milliseconds

==> Some builds didn't complete successfully and had errors:
--> Raspberry Pi 64 Pwnagotchi.arm.rpi64-pwnagotchi: build was halted

==> Builds finished but no artifacts were created.
make: *** [Makefile:61: pwnagotchi] Error 1
```

**To Reproduce**
Steps to reproduce the behavior:
1. Just `make image`

**Expected behavior**
It should make a .img file

**Environment (please complete the following information):**
 - Pwnagotchi version 2.8.4
 - OS version Ubuntu 20..04
 - Type of hardware Amd64

hamzenis, 2024-02-19:
> un-sure about the error but you could just install the pre made image and then update to 2.8.4 https://github.com/jayofelony/pwnagotchi-bookworm/releases/tag/v2.8.3

@wpa-2 I wanted to get started with some coding and testing, but with the lack of documentation for building an image and now this problem. I wanted to build on my Mac, but some errors then switched to my linux server

hamzenis, 2024-02-19:
The pre build image is working fine.

wpa-2, 2024-02-19:
> @wpa-2 I wanted to get started with some coding and testing, but with the lack of documentation for building an image and now this problem. I wanted to build on my Mac, but some errors then switched [...]

qemu caused me loads of issues, I had to reinstall everything before it worked. 

> The pre build image is working fine.

It does, flawlessly

hamzenis, 2024-02-19:
> qemu caused me loads of issues, I had to reinstall everything before it worked.
>
>
> It does, flawlessly

Okay I try to reinstall emu completely and purge everything. Do you remember what you installed and on which OS?

jayofelony, 2024-02-19:
Please move this to discussions.

hamzenis, 2024-02-19:
> Please move this to discussions.

Sorry I moved it there.