jayofelony/pwnagotchi #69: [BUG] how to install on OrangePi Zero 2W

Ddataa, 2024-02-23:
**Describe the bug**
I am trying to use this with a OrangePi Zero 2W.

after flashing on to sd-card and booting the device I dont know how to connect

**To Reproduce**
Steps to reproduce the behavior:
1. flash sd card
2. boot device

**Environment (please complete the following information):**
 - Pwnagotchi version latest
 - OS version latest
 - Type of hardware OrangePi Zero 2W
 - Any additional hardware used 2''7 Sharp Memory LCD

**Additional context**

Ddataa, 2024-02-23:
considering I have a working bookworm on another sd-card for this device, is there a script to install from whithin ?

Ddataa, 2024-02-23:
like if I clone the repo and start setup.py ?

renewedbullet, 2024-02-23:
This is not for an orangepi zero 2w. An there is not a working image for a orange pi 2w they are barely supported by Armbian and likely won't get any Pwnagotchi release for a while if you are interested in other SBCs I recommend looking at Jayofelony repo for banana pi m4 zero.

Ddataa, 2024-02-23:
Thanks

this one ?
https://github.com/jayofelony/bananagotchi

Ddataa, 2024-02-23:
do you think I can run the setup.py directly ?

Ddataa, 2024-02-23:
ok thanks!

closed 2024-02-23 by Ddataa