jayofelony/pwnagotchi #72: error setting up waveshare1in54 e-paper display

bryzz42o, 2024-02-27:
no display after installing screen and changing config,
 --debug returns

Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 235, in <module>
    pwnagotchi_cli()
  File "/usr/local/bin/pwnagotchi", line 214, in pwnagotchi_cli
    display = Display(config=config, state={'name': '%s>' % pwnagotchi.name()})
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 18, in __init__
    self.init_display()
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 219, in init_display
    self._implementation.initialize()
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/waveshare1in54.py", line 37, in initialize
    self._display.init(0x00)
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/v1in54/epd1in54.py", line 201, in init
    for i in range(0, len(lut)):
                      ^^^^^^^^
TypeError: object of type 'int' has no len()
?????

jayofelony, 2024-02-27:
ok let me dive into that

bryzz42o, 2024-02-27:
appreciate you to the moon J! thought it may not be detecting the screen but "ls /dev/spi*" showed the screen occupying spi so might be in the code aka out of my league

jayofelony, 2024-02-27:
`sudo nano /usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/waveshare1in54.py`

Can you edit this file on line 37, where it says:
`self._display.init(0x00)` into `self._display.init(0)`

Can you do that real quick for me? Than I can start a new image build with this fix.

bryzz42o, 2024-02-27:
yep will have it done in 10 mins legend

bryzz42o, 2024-02-27:
sorted, did you want me to submit a pull for it or just run a debug & post output?

bryzz42o, 2024-02-27:
ui.display.type is waveshare1in54 
pwnagotchi --debug output: Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 235, in <module>
    pwnagotchi_cli()
  File "/usr/local/bin/pwnagotchi", line 214, in pwnagotchi_cli
    display = Display(config=config, state={'name': '%s>' % pwnagotchi.name()})
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 18, in __init__
    self.init_display()
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 219, in init_display
    self._implementation.initialize()
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/waveshare1in54.py", line 37, in initialize
    self._display.init(0)
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/v1in54/epd1in54.py", line 201, in init
    for i in range(0, len(lut)):
                      ^^^^^^^^
TypeError: object of type 'int' has no len()

bryzz42o, 2024-02-27:
Here's a possible modification to the init() method to explicitly accept lut as an argument:class EPD:
    def __init__(self):
        self.reset_pin = epdconfig.RST_PIN
        self.dc_pin = epdconfig.DC_PIN
        self.busy_pin = epdconfig.BUSY_PIN
        self.cs_pin = epdconfig.CS_PIN
        self.width = EPD_WIDTH
        self.height = EPD_HEIGHT

    lut_full_update = [
        0x02, 0x02, 0x01, 0x11, 0x12, 0x12, 0x22, 0x22,
        0x66, 0x69, 0x69, 0x59, 0x58, 0x99, 0x99, 0x88,
        0x00, 0x00, 0x00, 0x00, 0xF8, 0xB4, 0x13, 0x51,
        0x35, 0x51, 0x51, 0x19, 0x01, 0x00
    ]

    lut_partial_update = [
        0x10, 0x18, 0x18, 0x08, 0x18, 0x18, 0x08, 0x00,
        0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
        0x00, 0x00, 0x00, 0x00, 0x13, 0x14, 0x44, 0x12,
        0x00, 0x00, 0x00, 0x00, 0x00, 0x00
    ]

    def init(self, lut):
        if epdconfig.module_init() != 0:
            return -1
        # EPD hardware init start
        self.reset()

        self.send_command(0x01)  # DRIVER_OUTPUT_CONTROL
        self.send_data((EPD_HEIGHT - 1) & 0xFF)
        self.send_data(((EPD_HEIGHT - 1) >> 8) & 0xFF)
        self.send_data(0x00)  # GD = 0 SM = 0 TB = 0

        self.send_command(0x0C)  # BOOSTER_SOFT_START_CONTROL
        self.send_data(0xD7)
        self.send_data(0xD6)
        self.send_data(0x9D)

        self.send_command(0x2C)  # WRITE_VCOM_REGISTER
        self.send_data(0xA8)  # VCOM 7C

        self.send_command(0x3A)  # SET_DUMMY_LINE_PERIOD
        self.send_data(0x1A)  # 4 dummy lines per gate

        self.send_command(0x3B)  # SET_GATE_TIME
        self.send_data(0x08)  # 2us per line

        self.send_command(0x11)  # DATA_ENTRY_MODE_SETTING
        self.send_data(0x03)  # X increment Y increment

        # set the look-up table register
        self.send_command(0x32)
        for i in range(0, len(lut)):
            self.send_data(lut[i])
        # EPD hardware init end
        return 0

jayofelony, 2024-02-27:
hmm change it to a `1` and debug again :smile:

bryzz42o, 2024-02-27:
how it looks after change --- 
def initialize(self):
        logging.info("initializing waveshare v1in54 display")
        from pwnagotchi.ui.hw.libs.waveshare.v1in54.epd1in54 import EPD
        self._display = EPD()
        self._display.init(1)
        self._display.Clear() 

"sudo pwnagotchi --debug" output:
Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 235, in <module>
    pwnagotchi_cli()
  File "/usr/local/bin/pwnagotchi", line 214, in pwnagotchi_cli
    display = Display(config=config, state={'name': '%s>' % pwnagotchi.name()})
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 18, in __init__
    self.init_display()
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 219, in init_display
    self._implementation.initialize()
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/waveshare1in54.py", line 37, in initialize
    self._display.init(1)
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/v1in54/epd1in54.py", line 201, in init
    for i in range(0, len(lut)):
                      ^^^^^^^^
TypeError: object of type 'int' has no len()

here's my config for clarity:
main.name = "blipppy"
main.lang = "en"
main.whitelist = [
 "hedgewitch",
 "ANOTHER_EXAMPLE_NETWORK",
 "fo:od:ba:be:fo:od",
 "fo:od:ba",
]
main.confd = "/etc/pwnagotchi/conf.d/"
main.custom_plugin_repos = [
 "https://github.com/jayofelony/pwnagotchi-torch-plugins/archive/master.zip",
 "https://github.com/tisboyo/pwnagotchi-pisugar2-plugin/archive/master.zip",
 "https://github.com/nullm0ose/pwnagotchi-plugin-pisugar3/archive/master.zip",
 "https://github.com/Sniffleupagus/pwnagotchi_plugins/archive/master.zip",
 "https://github.com/NeonLightning/pwny/archive/master.zip",
 "https://github.com/marbasec/UPSLite_Plugin_1_3/archive/master.zip",
 "https://github.com/Kaska89/pwnagotchi-EXPv2-plugin/archive/master.zip",
]
main.custom_plugins = "/usr/local/share/pwnagotchi/custom-plugins/"
main.plugins.aircrackonly.enabled = true
main.plugins.aircrackonly.face = "(>.<)"

main.plugins.hashie.enabled = true

main.plugins.auto-update.enabled = true
main.plugins.auto-update.install = true
main.plugins.auto-update.interval = 1

main.plugins.bt-tether.enabled = true
main.plugins.bt-tether.devices.android-phone.enabled = false
main.plugins.bt-tether.devices.android-phone.search_order = 1
main.plugins.bt-tether.devices.android-phone.mac = ""
main.plugins.bt-tether.devices.android-phone.ip = "192.168.44.44"
main.plugins.bt-tether.devices.android-phone.netmask = 24
main.plugins.bt-tether.devices.android-phone.interval = 1
main.plugins.bt-tether.devices.android-phone.scantime = 10
main.plugins.bt-tether.devices.android-phone.max_tries = 10
main.plugins.bt-tether.devices.android-phone.share_internet = false
main.plugins.bt-tether.devices.android-phone.priority = 1

main.plugins.bt-tether.devices.ios-phone.enabled = false
main.plugins.bt-tether.devices.ios-phone.search_order = 2
main.plugins.bt-tether.devices.ios-phone.mac = ""
[... 197 lines cut ...]
 "ble.device.connected",
 "ble.device.service.discovered",
 "ble.device.characteristic.discovered",
 "wifi.client.new",
 "wifi.client.lost",
 "wifi.client.probe",
 "wifi.ap.new",
 "wifi.ap.lost",
 "mod.started",
]

fs.memory.enabled = true
fs.memory.mounts.log.enabled = true
fs.memory.mounts.log.mount = "/home/pi/logs"
fs.memory.mounts.log.size = "50M"
fs.memory.mounts.log.sync = 60
fs.memory.mounts.log.zram = true
fs.memory.mounts.log.rsync = true

fs.memory.mounts.data.enabled = true
fs.memory.mounts.data.mount = "/var/tmp/pwnagotchi"
fs.memory.mounts.data.size = "10M"
fs.memory.mounts.data.sync = 3600
fs.memory.mounts.data.zram = true
fs.memory.mounts.data.rsync = true

jayofelony, 2024-02-27:
Can you set it as this?
`self._display.init(self._display.lut_full_update)`

bryzz42o, 2024-02-27:
That fixed the len error, unfortunately screen still not displaying so I'll recheck all the pin connections today. Could possibly be the tweak_view changing the disp settings or something so will turn off all the plugins later and see if running bare bones helps it display the screen

bryzz42o, 2024-02-27:
No luck running bare bones and reseating the pins, If the repo supports piTfT screens I'll try one I have later to confirm this e-paper isn't faulty but is only a couple days old so unlikely. Appreciate all the help

jayofelony, 2024-02-27:
If you have a v2 b version you need to use different display type

bryzz42o, 2024-02-28:
changed & --debug: unsupported display type waveshare1in54b_v2

bryzz42o, 2024-02-28:
changed & success with waveshare1in54b <3 incase anyone else ends up here its the 1.54inch e-paper waveshare screen with a GREEN sticker on the front protector film

closed 2024-02-28 by bryzz42o