jayofelony/pwnagotchi #73: [BUG]

WolfgangBlack, 2024-02-28:
**Describe the bug**
Unable to install a RTC module

**To Reproduce**
Steps to reproduce the behavior:
1. Unable to install a RTC module, i2cdetect is not found when I run "run sudo i2cdetect -y 1"
2. Trying to install i2c-tools fails even when I do a
sudo apt-get update
sudo apt-get install i2c-tools 
It can not find the libraries needed to install the i2c-tools....

**Expected behavior**
I am able to use the pwnagotchi guide to install the RTC on my Pi 02W pwnagotchi build using aluminum-ice's 02w 32 bit OS but that has bugs with the UPS-Lite v1.2 and is not 64 bit.
Thought I could simply install i2c-tools and then the changes to make the RTC work.
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Pwnagotchi v2.8.5 from this github
 - UPS-lite v1.2, waveshare v4 screen, pi zero 2w

jayofelony, 2024-02-28:
[image]
Finds i2c-tools perfectly fine on my pi4.

WolfgangBlack, 2024-02-29:
Close it, my internet connection sharing is not working.
Sorry.

WolfgangBlack, 2024-02-29:
I am stupid.

closed 2024-02-29 by WolfgangBlack