jayofelony/pwnagotchi #76: [BUG] Update fails

simphide, 2024-03-02:
**Describe the bug**
I run the command sudo pwnagotchi --check-update and nothing happens
[image]

**To Reproduce**
Steps to reproduce the behavior:
1.  sudo pwnagotchi --check-update
2. pwnlog

**Expected behavior**
Pwnagotchi gets updated

**Screenshots**
[image]

[image]

**Environment (please complete the following information):**
 - Pwnagotchi 2.8.2
 - Pi Zero 2 W
 - nothing

**Additional context**
I don't know what should happen...

jayofelony, 2024-03-02:
You have to reflash sadly

jayofelony, 2024-03-02:
This one specifically because of repo name change. Otherwise I will tell in release notes if I recommend it.

closed 2024-03-02 by simphide

simphide, 2024-03-02:
the update was successful! Just a quick question @jayofelony - what is the new text in the bottom right cornor? "rst#1"

jayofelony, 2024-03-02:
Thats the counter how many times the wifi firmware is being reset. It will be removed in the next update.