jayofelony/pwnagotchi #82: Bettercap not starting.

b4k0s, 2024-03-06:
When running in AUTO mode, this message keeps flashing:

jayofelony, 2024-03-07:
Working on a fix, just restart bettercap with `sudo systemctl restart bettercap` for now.

Probably a 02w with code 23xx on the cpu die?

jagp8, 2024-03-07:
I don't know if this might be related, but when I run it into auto mode the screen shows up but after 5 seconds it get restarted again into manual (if connected to the pc) or it enters a restart loop

b4k0s, 2024-03-07:
> Working on a fix, just restart bettercap with `sudo systemctl restart bettercap` for now.
> 
> Probably a 02w with code 23xx on the cpu die?

I just wipe it all and flashed the image again. but thanks for your response! :)

ShiniGandhi, 2024-03-21:
apparently wifi and bluetooth just aren't even being recognized by the os.
that's probably bettercap isn't working.

wpa-2, 2024-03-21:
> apparently wifi and bluetooth just aren't even being recognized by the os. that's probably bettercap isn't working.

check debug 
`sudo pwnagotchi --debug `

ShiniGandhi, 2024-03-21:
> > apparently wifi and bluetooth just aren't even being recognized by the os. that's probably bettercap isn't working.
> 
> check debug `sudo pwnagotchi --debug `

```
[2024-03-12 02:19:52,602] [INFO] entering auto mode ...
DEBUG:urllib3.util.retry:Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
[2024-03-12 02:19:52,609] [DEBUG] Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
INFO:root:[AI] bootstrapping dependencies ...
[2024-03-12 02:19:52,641] [INFO] [AI] bootstrapping dependencies ...
DEBUG:urllib3.util.retry:Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
[2024-03-12 02:19:52,683] [DEBUG] Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): localhost:8081
[2024-03-12 02:19:52,984] [DEBUG] Starting new HTTP connection (1): localhost:8081
INFO:root:waiting for bettercap API to be available ...
[2024-03-12 02:19:53,133] [INFO] waiting for bettercap API to be available ...
DEBUG:urllib3.util.retry:Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
[2024-03-12 02:20:14,142] [DEBUG] Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
DEBUG:urllib3.util.retry:Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
[2024-03-12 02:20:14,187] [DEBUG] Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): localhost:8081
[2024-03-12 02:20:14,328] [DEBUG] Starting new HTTP connection (1): localhost:8081
INFO:root:waiting for bettercap API to be available ...
[2024-03-12 02:20:14,450] [INFO] waiting for bettercap API to be available ...
INFO:root:web ui available at http://[::]:8080/
[2024-03-12 02:20:14,715] [INFO] web ui available at http://[::]:8080/
 * Serving Flask app "pwnagotchi.ui.web.server" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
Exception ignored in thread started by: <bound method Server._http_serve of <pwnagotchi.ui.web.server.Server object at 0xb2228790>>
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/ui/web/server.py", line 50, in _http_serve
    app.run(host=self._address, port=self._port, debug=False)
  File "/usr/lib/python3/dist-packages/flask/app.py", line 990, in run
    run_simple(host, port, self, **options)
  File "/usr/lib/python3/dist-packages/werkzeug/serving.py", line 1052, in run_simple
    inner()
  File "/usr/lib/python3/dist-packages/werkzeug/serving.py", line 996, in inner
    srv = make_server(
  File "/usr/lib/python3/dist-packages/werkzeug/serving.py", line 847, in make_server
    return ThreadedWSGIServer(
  File "/usr/lib/python3/dist-packages/werkzeug/serving.py", line 740, in __init__
    HTTPServer.__init__(self, server_address, handler)
  File "/usr/lib/python3.9/socketserver.py", line 452, in __init__
    self.server_bind()
  File "/usr/lib/python3.9/http/server.py", line 138, in server_bind
    socketserver.TCPServer.server_bind(self)
  File "/usr/lib/python3.9/socketserver.py", line 466, in server_bind
    self.socket.bind(self.server_address)
OSError: [Errno 98] Address already in use
DEBUG:urllib3.util.retry:Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
[2024-03-12 02:20:15,508] [DEBUG] Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
DEBUG:urllib3.util.retry:Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
[2024-03-12 02:20:15,532] [DEBUG] Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): localhost:8081
[2024-03-12 02:20:15,647] [DEBUG] Starting new HTTP connection (1): localhost:8081
INFO:root:waiting for bettercap API to be available ...
[2024-03-12 02:20:15,718] [INFO] waiting for bettercap API to be available ...
DEBUG:urllib3.util.retry:Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
[2024-03-12 02:20:16,757] [DEBUG] Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
DEBUG:urllib3.util.retry:Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
[2024-03-12 02:20:16,792] [DEBUG] Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): localhost:8081
[2024-03-12 02:20:16,932] [DEBUG] Starting new HTTP connection (1): localhost:8081
INFO:root:waiting for bettercap API to be available ...
[2024-03-12 02:20:16,993] [INFO] waiting for bettercap API to be available ...
DEBUG:urllib3.util.retry:Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
[2024-03-12 02:20:18,042] [DEBUG] Converted retries value: 0 -> Retry(total=0, connect=None, read=None, redirect=None, status=None)
[... 60 lines cut ...]
[... cut ...]
  File "/usr/local/bin/pwnagotchi", line 335, in pwnagotchi_cli
    do_auto_mode(agent)
  File "/usr/local/bin/pwnagotchi", line 53, in do_auto_mode
^C  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/agent.py", line 133, in start
    self._wait_bettercap()
  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/agent.py", line 129, in _wait_bettercap
    time.sleep(1)
```

as i've said, bettercap is not starting since it can't find the wifi chip. when i do `ip a` it does not show the wifi card. same with when trying to run `bluetoothctl`

jayofelony, 2024-03-22:
should be fixed now

closed 2024-03-22 by jayofelony