jayofelony/pwnagotchi #83: how to change to "name>" from "name!!!"

TheMobas0, 2024-03-07:
it's a small visual thing that bothers me but I wish I could change it to be like the original firmware where it says "name>" instead of "name!!!".

OnlyNandan, 2024-03-07:
Check if your pwnagotchi has instaattack enabled. Instaattack adds that to the ui. Should be a very simple fix in the Python code if you don't want it to have `!!!` in the ui.

OnlyNandan, 2024-03-07:
Check if your pwnagotchi has instaattack plugin on. That plugin manipulates the UI to add `!!!` and remove `>` . Should be a very easy fix if you are looking to use that plugin and keep the pwnagotchi UI stock. 

[Instaatck.py](https://github.com/Sniffleupagus/pwnagotchi_plugins/blob/main/instattack.py)

The code you are looking would be on line 36.

TheMobas0, 2024-03-08:
Thanks that fixed it. I just commented the whole on_ui_update() function. Sorry for the confusion I thought that it was the firmware since it changed the color too but I was wrong. Thanks again.

closed 2024-03-08 by TheMobas0