jayofelony/pwnagotchi #86: Can't enter in auto mode [BUG]

jagp8, 2024-03-09:
I just flashed the 2.8.6 image but when I try to run it into auto mode it seems that it tries and restart the pwnagotchi service and in the screen the auto mode appears, but after 5 seconds the service gets restarted again and it goes back into manual mode

If I connect it to the power supply instead of the data usb it's a constat restart :S

jagp8, 2024-03-10:
Here are the logs
[logs.txt](https://github.com/jayofelony/pwnagotchi/files/14548287/logs.txt)

jayofelony, 2024-03-10:
Well this doesnt show an obvious error, try disabling plugins one by one. You have 19 plugins enabled, which is quite much.

jagp8, 2024-03-10:
I disabled all the plugins but still the same behaviour, here there are some [more_logs.txt](https://github.com/jayofelony/pwnagotchi/files/14550982/more_logs.txt) :S

jagp8, 2024-03-11:
Flashed back to 2.8.3 with the same config and all the plugins and it's working fine

jagp8, 2024-03-18:
Updated to latest version and still the same error, the last message con the systemctl status from pwnagotchi is
```
● pwnagotchi.service - pwnagotchi Deep Reinforcement Learning instrumenting bettercap for WiFI pwning.                       Loaded: loaded (/etc/systemd/system/pwnagotchi.service; enabled; preset: enabled)                                       Active: activating (auto-restart) since Mon 2024-03-18 21:12:39 CET; 24s ago                                              Docs: https://pwnagotchi.org                                                                                         Process: 40681 ExecStart=/usr/bin/pwnagotchi-launcher (code=exited, status=0/SUCCESS)                                   Process: 40966 ExecStopPost=/usr/bin/bash -c if egrep -qi  'personality.clear_on_exit[ =]*true' /etc/pwnagotchi/config.toml ; then /usr/local/bin/pwnagotchi --clear; fi (code=exited, status=0/SUCCESS)                                       Main PID: 40681 (code=exited, status=0/SUCCESS)                                                                              CPU: 16.255s            
```

jayofelony, 2024-03-27:
You have something wrong somewhere I do not know where. But everyone has no issues, so I will close this issue now.

closed 2024-03-27 by jayofelony

IT-Wizard, 2025-01-13:
> I just flashed the 2.8.6 image but when I try to run it into auto mode it seems that it tries and restart the pwnagotchi service and in the screen the auto mode appears, but after 5 seconds the [...]

I have the same issue with the latest version. The service keeps on looping in AUTO mode.
I have the plain version installed without any plugins.
Did you solve this situation ?

wpa-2, 2025-01-13:
> I have the same issue with the latest version. The service keeps on looping in AUTO mode. I have the plain version installed without any plugins. Did you solve this situation ?

AI was removed so auto is right?

IT-Wizard, 2025-01-13:
what ? Pwnagotchi is looping: having "new morning" message, reading logs, aso.
It is version [2.9.4-2] on Zero 32bits.
I just see there is a new version 2.9.5. I will try it this evening...

wpa-2, 2025-01-13:
> what ? Pwnagotchi is looping: having "new morning" message, reading logs, aso. It is version [2.9.4-2] on Zero 32bits. I just see there is a new version 2.9.5. I will try it this evening...

what do the logs say!

IT-Wizard, 2025-01-13:
Pretty much the same as thoses of jagp8. I am not home yet, so cannot pot it now.

wpa-2, 2025-01-13:
> Pretty much the same as thoses of jagp8. I am not home yet, so cannot pot it now. I remerber that I saw a message like : [INFO] error 400. I had mixed reaction; is that an info or an error, bat as it was on web interface, I was not that worried.

run debug and see whats up.