jayofelony/pwnagotchi #89: [BUG] Device keeps going blind

cr0bar, 2024-03-11:
**Describe the bug**
Pwnagotchi sees APs, but after about 10 minutes it switches to APS 0 (0) and complains about not seeing any networks. 

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on device
2. Finds networks successfully
3. 10 minutes elapsed
4. No longer sees networks

**Expected behavior**
Continue to see APs

**Screenshots**
[image]

**Environment (please complete the following information):**
 - Pwnagotchi version - [Pwnagotchi v2.8.6](https://github.com/jayofelony/pwnagotchi/releases/tag/v2.8.6)
 - OS version - As above
 - Type of hardware - Raspberry Pi Zero W
 - Any additional hardware used - PiSugar, Waveshare 213

**Additional context**
Device is static at the time and within reach of many networks. Tried re-imaging etc.

cr0bar, 2024-03-11:
Yes set up with plugin enabled and haven't run update or upgrade, clean image.

I've had this failure all morning, 10 minutes in and it would cut out, but... I've just taken the dog out and 1 hour 10 minutes in, it's still working as it has for the last couple of years. Strange.

I'll leave it open until this evening, but for now, please assume it's closed.

cr0bar, 2024-03-18:
Re-opened due to further investigation and potential fix.

- Edit /etc/network/interfaces.d/wlan0-cfg
- Add the following: wireless-power off

wireless-power off disables power management of the internal wifi module and seems to stop it going blind.

BarnOwlDown, 2024-03-27:
It seems there's no support for RPi Zero 2W with bcm43430a1/7_45_41_46 (43430/1). My device keeps going blind and reboot instantly ad boot.
Can you confirm to me that that chip isn't supported (yet)? The aluminum-ice fork works well with it, but I surely prefer this fork!

jayofelony, 2024-03-27:
This chip has been supported since forever, my images support all wireless RPi.

closed 2024-03-27 by jayofelony

JAKAMI99, 2025-10-03:
@cr0bar Had you any luck solving this? I have this issue for months. 
I tried different power sources to eliminate the probability of instable voltage.
At the end, I even bought a replacement pi0-2wh to make sure it ain't hardware.