jayofelony/pwnagotchi #94: [BUG] pwnagotchi not booting in manual mode

jjonojj, 2024-03-14:
**Describe the bug**
Today i upgraded my 1.5.6beta2 pwnagotchi to 2.8.6 and if i plug it into my computer now, it goes auto, and after a short time ai. it does not go manual. (32-bit version, rpi0w v1, waveshare v4)

**To Reproduce**
1. flash 2.6.8 on a raspberry pi zero. default config from pwnagotchi.ai website, only change display.type to waveshare_3.
2. plug the micro usb cable into the data plug of your rpi
3. enjoy

**Expected behavior**
i want the device to go into manual mode so i can access the web ui.

**Environment (please complete the following information):**
 - Pwnagotchi 2.8.6 32bit
 - Raspberry Pi zero V1
 - Waveshare V4 display

jayofelony, 2024-03-14:
You marked as completed, so I assume all is working now?

jjonojj, 2024-03-14:
kinda, it boots in manual mode but i still can't access the web ui / ssh into it. i also installed the rndis drivers and set a static ip

wpa-2, 2024-03-14:
> kinda, it boots in manual mode but i still can't access the web ui / ssh into it. i also installed the rndis drivers and set a static ip

Probably better off moving this to discord as its not really a bug for here!

jjonojj, 2024-03-14:
ok

closed 2024-03-14 by jjonojj