jayofelony/pwnagotchi #96: SSLCertVerificationError

Eception, 2024-03-16:
Hi,

why do i get this error on 2.8.6. It happens on sudo apt update aswell.

Downloading plugins from https://github.com/jayofelony/pwnagotchi-torch-plugins/archive/master.zip to /usr/local/share/pwnagotchi/available-plugins/plugins0.zip
[2024-01-26 22:43:41,494] [ERROR] Error while updating plugins: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /jayofelony/pwnagotchi-torch-plugins/archive/master.zip (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate is not yet valid (_ssl.c:992)')))

jayofelony, 2024-03-16:
No internet?

Eception, 2024-03-18:
Yeah, tried it with windows 10 and it worked.

Idk why it doesn´t let me connect to the internet with windows 11 tho.

closed 2024-03-26 by jayofelony