jayofelony/pwnagotchi #97: [BUG] v2.8.7 won't work

hunter40415, 2024-03-17:
After update to the new version of the pwnagotchi it work for a few minute and then start to boot loop in the logfile and in an attempt to reflash the updated firmware the pwnagotchi just wont boot anymore
I'm currently using RPi 2w with waveshare 2.13 v4 e-paper display

OnlyNandan, 2024-03-17:
+1. Altho im running a waveshare v4 and its just stuck on rebooting.

My Debug Logs
`Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 338, in <module>
    pwnagotchi_cli()
  File "/usr/local/bin/pwnagotchi", line 318, in pwnagotchi_cli
    display = Display(config=config, state={'name': '%s>' % pwnagotchi.name()})
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 18, in __init__
    self.init_display()
`

tiimjcb, 2024-03-17:
+1. exact same problem.

Shy-Doge, 2024-03-17:
also facing this, waveshare v4 and rpi 2w. checking the pwnlog it just keeps on loading the plugins, it does not connect to my phone or anything, screen is dead.

jayofelony, 2024-03-17:
Can someone copy paste contents of `/boot/firmware/config.txt`

tiimjcb, 2024-03-17:
```
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

dtoverlay=dwc2
dtoverlay=spi1-3cs
dtparam=i2c1=on
dtparam=i2c_arm=on
dtparam=spi=on
gpu_mem=1
```

there are you

jayofelony, 2024-03-17:
Is that a 0w or a 02w?

jayofelony, 2024-03-17:
In `/boot/firmware/config.txt` at the bottom change spi1-3cs to spi0-0cs, somehow this didnt get written to the image.

Miles8of9, 2024-03-17:
> Can someone copy paste contents of `/boot/firmware/config.txt`
Raspberry Zero 2W here.

I have that directory but it's empty... There is a config.txt file in /boot

I've reflashed sd card with 2.8.7 image. Connecting raspberry data port to the pc (ubuntu), the green led flashes for some seconds then stays green still.

The cpu is a little hot so it's doing something but the Raspberry doesn't even show in dmesg or syslog.

Have no display, i'm running headless.

Repeated flash 2 times, same result.

simphide, 2024-03-17:
> In `/boot/firmware/config.txt` at the bottom change spi1-3cs to spi0-0cs, somehow this didnt get written to the image.

works for me

simphide, 2024-03-17:
But now the error "
[2024-03-17 19:46:49,599] | [ERROR] | [update] 'tag_name'
-- | -- | --"
pops up constantly

jayofelony, 2024-03-17:
> But now the error "
> [2024-03-17 19:46:49,599] | [ERROR] | [update] 'tag_name'
> -- | -- | --"
> pops up constantly
> 
> edit: you have forgotten to add a "v" to you version numer...
> `info['available'] = latest_ver = latest['tag_name'].replace('v', '')`

You have a softban from github

Shy-Doge, 2024-03-17:
> In `/boot/firmware/config.txt` at the bottom change spi1-3cs to spi0-0cs, somehow this didnt get written to the image.

this fixed it for me!

jayofelony, 2024-03-17:
https://github.com/jayofelony/pwnagotchi/releases/tag/v2.8.7.1

closed 2024-03-17 by jayofelony