jayofelony/pwnagotchi #98: [BUG] bt-tether missing bus module.

benallard, 2024-03-17:
I just installed a fresh 2.8.7, on an old rpi zero.

The device is booting, but the display is somehow not updating at all (another issue).

And I cannot enable the bt-tether plugin:

```

[2024-03-12 01:37:54,931] [ERROR] Exception on /plugins/toggle [POST]
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/lib/python3/dist-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/lib/python3/dist-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/lib/python3/dist-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/usr/lib/python3/dist-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/lib/python3/dist-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/ui/web/handler.py", line 71, in wrapper
    return f(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/ui/web/handler.py", line 189, in plugins
    return 'success' if plugins.toggle_plugin(request.form['plugin'], checked) else 'failed'
  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/plugins/__init__.py", line 60, in toggle_plugin
    load_from_file(database[name])
  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/plugins/__init__.py", line 110, in load_from_file
    spec.loader.exec_module(instance)
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/plugins/default/bt-tether.py", line 7, in <module>
    import dbus
ModuleNotFoundError: No module named 'dbus'
```

jayofelony, 2024-03-17:
https://github.com/jayofelony/pwnagotchi/issues/97#issuecomment-2002573554

benallard, 2024-03-17:
Ok, thanks, but there is not even a firmware directory in there!

jayofelony, 2024-03-17:
I dont know why that dbus would be missing somehow but you can install it using

`sudo apt-get install python3-dbus` iirc

jayofelony, 2024-03-17:
For 32bit its /boot/config.txt correct.

benallard, 2024-03-17:
The dtoverlay line is correct in the 32-bits image.

chermax14, 2024-03-19:
_File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/plugins/default/bt-tether.py", line 7, in <module>
    import dbus
ModuleNotFoundError: No module named 'dbus'_

still existing on freshly installed 2.8.7.2 32bit using rpi zero 2 w.

gpio_buttons.py doesn't also work on 2.8.7.1 and 2.8.7.2 64bit (im using GPIO 23 for shutdown. i didn't try other pins. but it works perfectly on 32bit images)

closed 2024-03-26 by jayofelony