jayofelony/pwnagotchi #101: [BUG] Waveshare v4 fails to use display no log errors.

rai68, 2024-03-18:
Display does not respond or refresh or clear

Pi Zero W + Waveshare v4

tigerbalm369
With v4 in config, running `sudo pwnagotchi --clear --debug`
[logv4.txt](https://github.com/jayofelony/pwnagotchi/files/14630625/message.32.txt)

With v3 in config, running `sudo pwnagotchi --clear --debug`
[logv3.txt](https://github.com/jayofelony/pwnagotchi/files/14630632/message.33.txt)

leow149, 2024-03-18:
+1 from me
It auto-updated itself this morning and now I get no display output and no bluetooth connection.
Logs look like this:

```
[2024-03-18 09:03:45,630] [INFO] [update] plugin loaded.
[2024-03-18 09:03:45,632] [INFO] webcfg: Plugin loaded.
[2024-03-18 09:03:45,633] [INFO] [EXP Plugin] Plugin Loaded
[2024-03-18 09:03:45,634] [INFO] grid plugin loaded.
[2024-03-18 09:03:45,635] [INFO] Logtail plugin loaded.
[2024-03-18 09:03:45,636] [INFO] [Fix_Services] plugin loaded.
[2024-03-18 09:03:45,636] [INFO] OHC: OnlineHashCrack plugin loaded.
[2024-03-18 09:03:45,638] [INFO] WPA_SEC: plugin loaded
[2024-03-18 09:03:45,640] [INFO] aircrackonly plugin loaded
[2024-03-18 09:03:45,641] [INFO] Session-stats plugin loaded.
[2024-03-18 09:03:45,642] [INFO] memtemp plugin loaded.
[2024-03-18 09:03:45,669] [INFO] initializing waveshare v2in13_V4 display
[2024-03-18 09:03:45,716] [INFO] aircrackonly: Found aircrack-ng 1:1.7-5
[2024-03-18 09:03:45,760] [INFO] BT-TETHER: Successfully loaded ...
[2024-03-18 09:03:50,729] [INFO] [update] plugin loaded.
[2024-03-18 09:03:50,730] [INFO] OHC: OnlineHashCrack plugin loaded.
[2024-03-18 09:03:50,732] [INFO] Logtail plugin loaded.
[2024-03-18 09:03:50,733] [INFO] WPA_SEC: plugin loaded
[2024-03-18 09:03:50,732] [INFO] grid plugin loaded.
[2024-03-18 09:03:50,734] [INFO] [EXP Plugin] Plugin Loaded
[2024-03-18 09:03:50,735] [INFO] aircrackonly plugin loaded
[2024-03-18 09:03:50,736] [INFO] webcfg: Plugin loaded.
[2024-03-18 09:03:50,738] [INFO] [Fix_Services] plugin loaded.
[2024-03-18 09:03:50,738] [INFO] memtemp plugin loaded.
[2024-03-18 09:03:50,741] [INFO] Session-stats plugin loaded.
[2024-03-18 09:03:50,764] [INFO] initializing waveshare v2in13_V4 display
[2024-03-18 09:03:50,780] [WARNING] non fatal error while updating view: dictionary changed size during iteration
[2024-03-18 09:03:50,833] [INFO] aircrackonly: Found aircrack-ng 1:1.7-5
[2024-03-18 09:03:50,914] [INFO] BT-TETHER: Successfully loaded ...
[2024-03-18 09:04:25,730] [INFO] [update] plugin loaded.
[2024-03-18 09:04:25,731] [INFO] grid plugin loaded.
[2024-03-18 09:04:25,732] [INFO] Logtail plugin loaded.
[2024-03-18 09:04:25,732] [INFO] OHC: OnlineHashCrack plugin loaded.
[2024-03-18 09:04:25,733] [INFO] Session-stats plugin loaded.
[2024-03-18 09:04:25,734] [INFO] webcfg: Plugin loaded.
[2024-03-18 09:04:25,735] [INFO] WPA_SEC: plugin loaded
[2024-03-18 09:04:25,738] [INFO] [EXP Plugin] Plugin Loaded
[2024-03-18 09:04:25,740] [INFO] aircrackonly plugin loaded
[2024-03-18 09:04:25,740] [INFO] [Fix_Services] plugin loaded.
[2024-03-18 09:04:25,741] [INFO] memtemp plugin loaded.
[2024-03-18 09:04:25,770] [INFO] initializing waveshare v2in13_V4 display
[2024-03-18 09:04:25,823] [INFO] aircrackonly: Found aircrack-ng 1:1.7-5
[2024-03-18 09:04:25,918] [INFO] BT-TETHER: Successfully loaded ...
[2024-03-18 09:04:30,581] [INFO] [update] plugin loaded.
[2024-03-18 09:04:30,582] [INFO] Logtail plugin loaded.
[2024-03-18 09:04:30,583] [INFO] aircrackonly plugin loaded
[2024-03-18 09:04:30,584] [INFO] [EXP Plugin] Plugin Loaded
[2024-03-18 09:04:30,586] [INFO] [Fix_Services] plugin loaded.
[2024-03-18 09:04:30,587] [INFO] OHC: OnlineHashCrack plugin loaded.
[2024-03-18 09:04:30,587] [INFO] WPA_SEC: plugin loaded
[2024-03-18 09:04:30,588] [INFO] grid plugin loaded.
[2024-03-18 09:04:30,592] [INFO] memtemp plugin loaded.
[2024-03-18 09:04:30,593] [INFO] Session-stats plugin loaded.
[2024-03-18 09:04:30,597] [INFO] webcfg: Plugin loaded.
[2024-03-18 09:04:30,622] [INFO] initializing waveshare v2in13_V4 display
[2024-03-18 09:04:30,666] [INFO] aircrackonly: Found aircrack-ng 1:1.7-5
[2024-03-18 09:04:30,751] [INFO] BT-TETHER: Successfully loaded ...
[2024-03-18 09:05:05,505] [INFO] [update] plugin loaded.
[2024-03-18 09:05:05,506] [INFO] grid plugin loaded.
[2024-03-18 09:05:05,507] [INFO] Logtail plugin loaded.
[2024-03-18 09:05:05,507] [INFO] OHC: OnlineHashCrack plugin loaded.
[2024-03-18 09:05:05,508] [INFO] WPA_SEC: plugin loaded
[2024-03-18 09:05:05,508] [INFO] aircrackonly plugin loaded
[2024-03-18 09:05:05,510] [INFO] Session-stats plugin loaded.
[2024-03-18 09:05:05,511] [INFO] [EXP Plugin] Plugin Loaded
[... 178 lines cut ...]
[2024-03-18 09:09:13,857] [INFO] OHC: OnlineHashCrack plugin loaded.
[2024-03-18 09:09:13,857] [INFO] WPA_SEC: plugin loaded
[2024-03-18 09:09:13,860] [INFO] webcfg: Plugin loaded.
[2024-03-18 09:09:13,861] [INFO] aircrackonly plugin loaded
[2024-03-18 09:09:13,862] [INFO] [EXP Plugin] Plugin Loaded
[2024-03-18 09:09:13,866] [INFO] Session-stats plugin loaded.
[2024-03-18 09:09:13,867] [INFO] memtemp plugin loaded.
[2024-03-18 09:09:13,891] [INFO] initializing waveshare v2in13_V4 display
[2024-03-18 09:09:13,943] [INFO] aircrackonly: Found aircrack-ng 1:1.7-5
[2024-03-18 09:09:14,038] [INFO] BT-TETHER: Successfully loaded ...
[2024-03-18 09:09:48,723] [INFO] [update] plugin loaded.
[2024-03-18 09:09:48,724] [INFO] grid plugin loaded.
[2024-03-18 09:09:48,724] [INFO] Logtail plugin loaded.
[2024-03-18 09:09:48,725] [INFO] OHC: OnlineHashCrack plugin loaded.
[2024-03-18 09:09:48,725] [INFO] WPA_SEC: plugin loaded
[2024-03-18 09:09:48,726] [INFO] aircrackonly plugin loaded
[2024-03-18 09:09:48,726] [INFO] [EXP Plugin] Plugin Loaded
[2024-03-18 09:09:48,728] [INFO] webcfg: Plugin loaded.
[2024-03-18 09:09:48,731] [INFO] memtemp plugin loaded.
[2024-03-18 09:09:48,732] [INFO] [Fix_Services] plugin loaded.
[2024-03-18 09:09:48,736] [INFO] Session-stats plugin loaded.
[2024-03-18 09:09:48,758] [INFO] initializing waveshare v2in13_V4 display
[2024-03-18 09:09:48,807] [INFO] aircrackonly: Found aircrack-ng 1:1.7-5
[2024-03-18 09:09:48,856] [INFO] BT-TETHER: Successfully loaded ...
```

jayofelony, 2024-03-18:
yes im working on it!

leow149, 2024-03-18:
Great, no pressure.
This update probably requires a reflash, right? How would I backup when I can't access the pi anymore?

renewedbullet, 2024-03-18:
> Great, no pressure. This update probably requires a reflash, right? How would I backup when I can't access the pi anymore?

Boot from a linux copy /root/handshakes, and /etc/pwnagotchi

jayofelony, 2024-03-19:
2.8.7.2 should have fixed all issues

closed 2024-03-19 by jayofelony