jayofelony/pwnagotchi #103: [BUG] cannot connect to phone: `Failed to connect: org.bluez.Error.NotAvailable br-connection-profile-unavailable`

formula-spectre, 2024-03-18:
**Describe the bug**
I have been bugged by this for a long time; I put the mac address of my phone in the config, but I saw that it connected once, disconnected and never reconnected anymore. so I did it manually with bluetoothctl; I tried to clean the slate, pair, connect and I get this:
```
[bluetooth]# connect MY_MAC_ADDRESS 
Attempting to connect to MY_MAC_ADDRESS
[CHG] Device MY_MAC_ADDRESS Connected: yes
[CHG] Device MY_MAC_ADDRESS Modalias: bluetooth:v00E0p1200d1436
[CHG] Device MY_MAC_ADDRESS UUIDs: 00001105-0000-1000-8000-00805f9b34fb
[CHG] Device MY_MAC_ADDRESS UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Device MY_MAC_ADDRESS UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Device MY_MAC_ADDRESS UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Device MY_MAC_ADDRESS UUIDs: 00001112-0000-1000-8000-00805f9b34fb
[CHG] Device MY_MAC_ADDRESS UUIDs: 00001115-0000-1000-8000-00805f9b34fb
[CHG] Device MY_MAC_ADDRESS UUIDs: 00001116-0000-1000-8000-00805f9b34fb
[CHG] Device MY_MAC_ADDRESS UUIDs: 0000111f-0000-1000-8000-00805f9b34fb
[CHG] Device MY_MAC_ADDRESS UUIDs: 0000112d-0000-1000-8000-00805f9b34fb
[CHG] Device MY_MAC_ADDRESS UUIDs: 0000112f-0000-1000-8000-00805f9b34fb
[CHG] Device MY_MAC_ADDRESS UUIDs: 00001132-0000-1000-8000-00805f9b34fb
[CHG] Device MY_MAC_ADDRESS UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device MY_MAC_ADDRESS UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Device MY_MAC_ADDRESS UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Device MY_MAC_ADDRESS UUIDs: a49eb41e-cb06-495c-9f4f-aa80a90cdf4a
[CHG] Device MY_MAC_ADDRESS UUIDs: a49eb41e-cb06-495c-9f4f-bb80a90cdf00
[CHG] Device MY_MAC_ADDRESS UUIDs: a82efa21-ae5c-3dde-9bbc-f16da7b16c5a
[CHG] Device MY_MAC_ADDRESS ServicesResolved: yes
Failed to connect: org.bluez.Error.NotAvailable br-connection-profile-unavailable
[CHG] Device MY_MAC_ADDRESS ServicesResolved: no
[CHG] Device MY_MAC_ADDRESS Connected: no
[bluetooth]# 
```
my phone is a pixel 6a, with grapheneOS on it;

**To Reproduce**
Steps to reproduce the behavior:
1. plug pwnagotchi in data port, and go into `manu` mode
2. issue `bluetoothctl`, then in the repl pair the device
- `agent on`
- `scan on`
- `pair phone_address`
- `trust phone_address`
- `connect phone_address`

**Expected behavior**
I should be able to pair the two, and access the webui from my phone or ssh into the pwnagotchi

**Environment (please complete the following information):**
 - Pwnagotchi version: v2.8.7.1
 - OS version: Debian GNU/Linux 12 (bookworm)
 - Type of hardware: Raspberry pi02 W, with waveshare eink hat v4
 - Google pixel 6a with grapheneOS

**Additional context**
I tried to search around and I tried to install pulseaudio or pipewire libraries, saw on some forums that I needed some libs like pulseaudio-bluetooth or libspa-0.2-bluetooth. no luck. I tried to put controller = bredr in `/etc/bluetooth/main.conf`, but nothing. i'm quite puzzled by this. on my pixel the pwnagotchi is registered as an audio device so.. I don't know honestly, I never encountered this error and I am kinda clueless.

jayofelony, 2024-03-19:
Simple things first, have you tried rebooting your phone. After boot turn on bt tethering on the phone, and after that reboot the rpi?

Miles8of9, 2024-03-19:
I can confirm there is some problem with BT tethering (raspberry pi zero 2w, image version 2.8.7.2)
Manual pairing is successful, connection is trusted but trying to connect returns this error.

[bluetooth]# connect MY:MAC:ADDRESS
Attempting to connect to MY:MAC:ADDRESS
[CHG] Device MY:MAC:ADDRESS Connected: yes
[CHG] Device MY:MAC:ADDRESS ServicesResolved: yes
Failed to connect: org.bluez.Error.NotAvailable br-connection-profile-unavailable
[CHG] Device MY:MAC:ADDRESS ServicesResolved: no
[CHG] Device MY:MAC:ADDRESS Connected: no

jayofelony, 2024-03-19:
The br profile seems to be something audio related, which we don't use. BT has always been a pain in the ass to set up. You will see it'll connect at some point.

formula-spectre, 2024-03-19:
> Simple things first, have you tried rebooting your phone. After boot turn on bt tethering on the phone, and after that reboot the rpi?

tried, rebooted phone, plugged pwnagotchi in the data port, enabled bt tether. manually tried bluetoothctl, connect MY:PHONE:MAC, same error

formula-spectre, 2024-03-19:
> The br profile seems to be something audio related, which we don't use. BT has always been a pain in the ass to set up. You will see it'll connect at some point.

can't one change bluetooth profile/agent then? because this thing is drawing me nuts; I tried to clean the slate and re-pair them, but it stayed connected for like one second and then disconnected

formula-spectre, 2024-03-19:
ah I forgot to put my config for the bt-tether, sorry!
```
main.plugins.bt-tether.enabled = true
main.plugins.bt-tether.devices.android-phone.enabled = true
main.plugins.bt-tether.devices.android-phone.search_order = 1
main.plugins.bt-tether.devices.android-phone.mac = ""
main.plugins.bt-tether.devices.android-phone.ip = "192.168.44.44"
main.plugins.bt-tether.devices.android-phone.netmask = 24
main.plugins.bt-tether.devices.android-phone.interval = 1
main.plugins.bt-tether.devices.android-phone.scantime = 10
main.plugins.bt-tether.devices.android-phone.max_tries = 10
main.plugins.bt-tether.devices.android-phone.share_internet = true
main.plugins.bt-tether.devices.android-phone.priority = 1
main.plugins.bt-tether.devices.ios-phone.enabled = false
main.plugins.bt-tether.devices.ios-phone.search_order = 2
main.plugins.bt-tether.devices.ios-phone.mac = ""
main.plugins.bt-tether.devices.ios-phone.ip = "172.20.10.6"
main.plugins.bt-tether.devices.ios-phone.netmask = 24
main.plugins.bt-tether.devices.ios-phone.interval = 5
main.plugins.bt-tether.devices.ios-phone.scantime = 20
main.plugins.bt-tether.devices.ios-phone.max_tries = 0
main.plugins.bt-tether.devices.ios-phone.share_internet = true
main.plugins.bt-tether.devices.ios-phone.priority = 999
```  
could it be the ip address?

jayofelony, 2024-03-19:
main.plugins.bt-tether.devices.android-phone.mac = ""

Did you leave this empty, you need to put your android mac address here.

formula-spectre, 2024-03-19:
> main.plugins.bt-tether.devices.android-phone.mac = ""
> 
> Did you leave this empty, you need to put your android mac address here.

I censored it, but it's there in the config.toml

jayofelony, 2024-03-19:
Ok and youre 100% sure its the BT MAC and not the WiFi MAC as they are directly underneath eachother on the phone info screen haha

formula-spectre, 2024-03-19:
UPDATE: just for giggles I put `main.plugins.bt-tether.devices.android-phone.max_tries = 0`, and.. it worked? like, now it connects, I can ssh into it and go into the webui.  
I don't know tbh. I have to reproduce this issue, because that was very random.

formula-spectre, 2024-03-19:
> Ok and youre 100% sure its the BT MAC and not the WiFi MAC as they are directly underneath eachother on the phone info screen haha

1000% sure

Miles8of9, 2024-03-19:
Tried the max_tries = 0 setting but not luck.
Looking at journalctl however i've found something that may be interesting:

Mar 15 16:09:04 pwnagotchi bluetoothd[444]: Starting SDP server
Mar 15 16:09:04 pwnagotchi bluetoothd[444]: Excluding (cli) sap
Mar 15 16:09:04 pwnagotchi bluetoothd[444]: Excluding (cli) a2dp
Mar 15 16:09:04 pwnagotchi systemd[1]: Started bluetooth.service - Bluetooth service.
Mar 15 16:09:04 pwnagotchi bluetoothd[444]: profiles/audio/vcp.c:vcp_init() D-Bus experimental not enabled
Mar 15 16:09:04 pwnagotchi bluetoothd[444]: src/plugin.c:plugin_init() Failed to init vcp plugin
Mar 15 16:09:04 pwnagotchi bluetoothd[444]: profiles/audio/mcp.c:mcp_init() D-Bus experimental not enabled
Mar 15 16:09:04 pwnagotchi bluetoothd[444]: src/plugin.c:plugin_init() Failed to init mcp plugin
Mar 15 16:09:04 pwnagotchi bluetoothd[444]: profiles/audio/bap.c:bap_init() D-Bus experimental not enabled
Mar 15 16:09:04 pwnagotchi bluetoothd[444]: src/plugin.c:plugin_init() Failed to init bap plugin
Mar 15 16:09:04 pwnagotchi kernel: Bluetooth: BNEP (Ethernet Emulation) ver 1.3
Mar 15 16:09:04 pwnagotchi kernel: Bluetooth: BNEP filters: protocol multicast
Mar 15 16:09:04 pwnagotchi kernel: Bluetooth: BNEP socket layer initialized
Mar 15 16:09:04 pwnagotchi bluetoothd[444]: Bluetooth management interface 1.22 initialized

These lines are repeated few times (yellow-red-yellow-red etc):
Mar 15 16:09:29 raspberry bluetoothd[530]: profiles/audio/vcp.c:vcp_init() D-Bus experimental not enabled
Mar 15 16:09:29 raspberry bluetoothd[530]: src/plugin.c:plugin_init() Failed to init vcp plugin
Mar 15 16:09:29 raspberry bluetoothd[530]: profiles/audio/mcp.c:mcp_init() D-Bus experimental not enabled
Mar 15 16:09:29 raspberry bluetoothd[530]: src/plugin.c:plugin_init() Failed to init mcp plugin
Mar 15 16:09:29 raspberry bluetoothd[530]: profiles/audio/bap.c:bap_init() D-Bus experimental not enabled
Mar 15 16:09:29 raspberry bluetoothd[530]: src/plugin.c:plugin_init() Failed to init bap plugin

On every bluetooth connection attempt (lots of them) these lines appear, no red or yellow however:
Mar 15 20:18:32 raspberry bluetoothd[522]: bnep0 connected
Mar 15 20:18:32 raspberry bluetoothd[522]: bnep0 disconnected
Mar 15 20:31:36 raspberry ModemManager[536]: <info>  [base-manager] port bnep0 released by device '/sys/devices/platform/soc'
Mar 15 20:31:36 raspberry ModemManager[536]: <info>  [base-manager] couldn't check support for device '/sys/devices/platform/soc': Operation was cancelled

Does this mean something to you?
Using 2.8.5 image and same pwnagotchi config.toml file i could connect to the phone without any problem.

Thanks for your help.

jayofelony, 2024-03-19:
Since 2.8.7 it is using kernel 6.6 that may be related. I don't know honestly. I'll look into this some other time.

Nixitov, 2024-03-20:
I'm having the same issue on all versions, it randomly connects, sometimes it says - C and it didnt connect, sometimes it says BE and sometimes NF

jayofelony, 2024-03-27:
My 02w works flawlessly with BT so I will close this issue. BT has always been finnicky.

closed 2024-03-27 by jayofelony

formula-spectre, 2024-03-27:
yeah, I could not replicate the error anymore, so good call closing this issue.  
kind OT but can I setup multiple bt devices for the pwnagotchi to connect to? I tried basically duplicate the bt-tether block but changing the mac address and ip but it says duplicate lines..

jayofelony, 2024-03-27:
No cannot

formula-spectre, 2024-03-27:
me sad :(