jayofelony/pwnagotchi #106: [BUG] RTL8814au Crashes Pwnagotchi

cr0bar, 2024-03-20:
**Describe the bug**
I have my RTL8814au in monitor mode with the interface name set to wlan0mon and the internal wifi disabled, but the Pwnagotchi service seems to crash and restart endlessly.

**To Reproduce**
Steps to reproduce the behavior:
1. Put device in monitor mode
2. Start pwnagotchi service

**Expected behavior**
Service starts and continues to find APs.

**Screenshots**

[image]

[image]

**Environment (please complete the following information):**
 - Pwnagotchi version - 2.8.6
 - OS version - raspberry pi image
 - Type of hardware - pi zero 2w
 - Any additional hardware used - Alfa AWUS1900

**Additional context**
The RTL8814au interface name doesn't change to wlan0mon when it goes into monitor by default so the name has to be overridden to remain compatible with the internal WiFi (when enabled). Using morrownr 8814au driver.

cr0bar, 2024-03-20:
Just to add to the above. If i stop the Pwnagotchi service and log into the bettercap web ui, I can see 38+ APs, so I know the wifi adapter is working ok.

jayofelony, 2024-03-20:
I cannot, and will not debug 3rd party hardware/drivers/firmware. You're on your own here, sorry.

closed 2024-03-20 by jayofelony

ceisma, 2024-04-12:
I'm well aware that the dev will not fix this but I want to share that all of my Realtek adapters will not work/crash with Jayofelony's image (they have RTL8814AU and RT8812AU chipsets). The Aluminum Ice image will not crash with these chipsets so I recommend using that image instead when using Realtek based adapters.