jayofelony/pwnagotchi #107: [BUG] Waveshare2in13b_V4

jojozada, 2024-03-20:
Hello, I ask for help.
I'm having a bug in waveshare V4

In the new update V2.8.7.2

When the screen updates it keeps changing the background from black to white.
This occurs with each new frame update.

I'm using.

Screen: Waveshare V4, 2.13inch e-Paper HAT

Respberry OW

driver:
waveshare2in13b_v4

ISO 32bit

I tested several other versions but they are only working but with this bug

config.toml:

main.name = "ENIAC"
main.lang = "en"
ui.display.enabled = true
ui.display.type = "waveshare2in13b_v4"
ui.display.color = "Black"

jayofelony, 2024-03-20:
Use waveshare_4 as display type

jojozada, 2024-03-20:
waveshare_4  don't start the screen

jayofelony, 2024-03-21:
Run `sudo pwnagotchi --debug` and look for any errors

Atlant87, 2024-03-22:
I had same problem, everething been fine i correcting plugins coordinats and some label to specific symbols like heart in "age" plugin, and then sometging went bad. Pwnagotchi start rebooted cyclically, the screen periodically flickers from white to black. I removed 

`pi@10.0.0.2:~ $ sudo pwnagotchi --debug
DEBUG:root:[FS] Daemonized...
DEBUG:root:[FS] Syncing /var/tmp/pwnagotchi -> /run/pwnagotchi/disk/pwnagotchi
DEBUG:root:loading plugins from /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default - enabled: ['age', 'aircrackonly', 'auto-update', 'bt-tether', 'exp', 'fix_services', 'grid', 'internet-conection', 'logtail', 'memtemp-plus', 'powerutils', 'pwnmenu', 'session-stats', 'webcfg', 'wpa-sec']
[2024-03-22 22:47:48,135] [DEBUG] loading plugins from /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default - enabled: ['age', 'aircrackonly', 'auto-update', 'bt-tether', 'exp', 'fix_services', 'grid', 'internet-conection', 'logtail', 'memtemp-plus', 'powerutils', 'pwnmenu', 'session-stats', 'webcfg', 'wpa-sec']
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/session-stats.py
[2024-03-22 22:47:48,141] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/session-stats.py
DEBUG:root:loaded plugin session-stats as <session-stats.SessionStats object at 0x7fae5bf150>
[2024-03-22 22:47:48,162] [DEBUG] loaded plugin session-stats as <session-stats.SessionStats object at 0x7fae5bf150>
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/webcfg.py
[2024-03-22 22:47:48,163] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/webcfg.py
DEBUG:root:loaded plugin webcfg as <webcfg.WebConfig object at 0x7fae5bfb50>
[2024-03-22 22:47:48,175] [DEBUG] loaded plugin webcfg as <webcfg.WebConfig object at 0x7fae5bfb50>
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/fix_services.py
[2024-03-22 22:47:48,176] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/fix_services.py
DEBUG:root:loaded plugin fix_services as <fix_services.FixServices object at 0x7fae5d8a10>
[2024-03-22 22:47:48,199] [DEBUG] loaded plugin fix_services as <fix_services.FixServices object at 0x7fae5d8a10>
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/logtail.py
[2024-03-22 22:47:48,200] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/logtail.py
DEBUG:root:loaded plugin logtail as <logtail.Logtail object at 0x7fae5d9d90>
[2024-03-22 22:47:48,238] [DEBUG] loaded plugin logtail as <logtail.Logtail object at 0x7fae5d9d90>
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/grid.py
[2024-03-22 22:47:48,239] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/grid.py
DEBUG:root:loaded plugin grid as <grid.Grid object at 0x7fae5dabd0>
[2024-03-22 22:47:48,264] [DEBUG] loaded plugin grid as <grid.Grid object at 0x7fae5dabd0>
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/wpa-sec.py
[2024-03-22 22:47:48,265] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/wpa-sec.py
DEBUG:root:loaded plugin wpa-sec as <wpa-sec.WpaSec object at 0x7fae5db490>
[2024-03-22 22:47:48,342] [DEBUG] loaded plugin wpa-sec as <wpa-sec.WpaSec object at 0x7fae5db490>
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/bt-tether.py
[2024-03-22 22:47:48,344] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/bt-tether.py
DEBUG:root:loaded plugin bt-tether as <bt-tether.BTTether object at 0x7fae62d9d0>
[2024-03-22 22:47:48,553] [DEBUG] loaded plugin bt-tether as <bt-tether.BTTether object at 0x7fae62d9d0>
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/auto-update.py
[2024-03-22 22:47:48,554] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/auto-update.py
DEBUG:root:loaded plugin auto-update as <auto-update.AutoUpdate object at 0x7fae62ec50>
[2024-03-22 22:47:48,558] [DEBUG] loaded plugin auto-update as <auto-update.AutoUpdate object at 0x7fae62ec50>
DEBUG:root:loading plugins from /usr/local/share/pwnagotchi/custom-plugins/ - enabled: ['age', 'aircrackonly', 'auto-update', 'bt-tether', 'exp', 'fix_services', 'grid', 'internet-conection', 'logtail', 'memtemp-plus', 'powerutils', 'pwnmenu', 'session-stats', 'webcfg', 'wpa-sec']
[2024-03-22 22:47:48,559] [DEBUG] loading plugins from /usr/local/share/pwnagotchi/custom-plugins/ - enabled: ['age', 'aircrackonly', 'auto-update', 'bt-tether', 'exp', 'fix_services', 'grid', 'internet-conection', 'logtail', 'memtemp-plus', 'powerutils', 'pwnmenu', 'session-stats', 'webcfg', 'wpa-sec']
DEBUG:root:loading /usr/local/share/pwnagotchi/custom-plugins/age.py
[2024-03-22 22:47:48,561] [DEBUG] loading /usr/local/share/pwnagotchi/custom-plugins/age.py
DEBUG:root:loaded plugin age as <age.Age object at 0x7fae62f9d0>
[2024-03-22 22:47:48,564] [DEBUG] loaded plugin age as <age.Age object at 0x7fae62f9d0>
DEBUG:root:loading /usr/local/share/pwnagotchi/custom-plugins/memtemp-plus.py
[2024-03-22 22:47:48,566] [DEBUG] loading /usr/local/share/pwnagotchi/custom-plugins/memtemp-plus.py
DEBUG:root:loaded plugin memtemp-plus as <memtemp-plus.MemTempPlus object at 0x7fae638b50>
[... cut ...]
    pin = self.pin_class(self, info)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/pins/lgpio.py", line 126, in __init__
    lgpio.gpio_claim_input(
  File "/usr/lib/python3/dist-packages/lgpio.py", line 755, in gpio_claim_input
    return _u2i(_lgpio._gpio_claim_input(handle&0xffff, lFlags, gpio))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/lgpio.py", line 458, in _u2i
    raise error(error_text(v))
lgpio.error: 'GPIO busy'
`

closed 2024-03-24 by jojozada