jayofelony/pwnagotchi #109: [BUG] "GPIO Busy" error when running --debug flag

greaterthanstar, 2024-03-22:
**Describe the bug**
I wasn't getting a plugin called wanmon.py to work (shows internet status on UI) so i tried running `sudo pwnagotchi --debug` and it is crashing with an eventual error of "GPIO Busy"

I'm not sure if the plugin has anything to do with it. My other plugins load fine, but when it gets to the wanmon.py area it seems to not load it. However, I can `*.enable = false` for that plugin and it still crashes the same way. 

[This issue is also on Reddit,](https://www.reddit.com/r/pwnagotchi/comments/1bhf5qf/gpio_busy_jayofelony_2871/) so other people are experiencing it. It seems wpa2 thought it was fixed in a prior version but it doesn't appear so. 

**To Reproduce**
Steps to reproduce the behavior:
1. Use waveshare_v2, rpi zero v2, pisugar3
2. jayofelony 64bit arm64 version, v 2.8.7.2
3. execute `sudo pwnagotchi --debug`

**Expected behavior**
able to execute the command without an exception being thrown

** Logs:
```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 408, in pin
    pin = self.pins[info]
          ~~~~~~~~~^^^^^^
KeyError: PinInfo(number=11, name='GPIO17', names=frozenset({'17', 17, 'BOARD11', 'GPIO17', 'BCM17', 'WPI0', 'J8:11'}), pull='', row=6, col=1, interfaces=frozenset({'gpio', '', 'spi', 'dpi', 'uart'}))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 338, in <module>
    pwnagotchi_cli()
  File "/usr/local/bin/pwnagotchi", line 318, in pwnagotchi_cli
    display = Display(config=config, state={'name': '%s>' % pwnagotchi.name()})
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 18, in __init__
    self.init_display()
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 240, in init_display
    self._implementation.initialize()
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/waveshare2in13_V4.py", line 35, in initialize
    from pwnagotchi.ui.hw.libs.waveshare.v2in13_V4.epd2in13_V4 import EPD
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/v2in13_V4/epd2in13_V4.py", line 32, in <module>
    from .. import epdconfig
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/epdconfig.py", line 127, in <module>
    implementation = RaspberryPi()
                     ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/epdconfig.py", line 52, in __init__
    self.GPIO_RST_PIN = gpiozero.LED(self.RST_PIN)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 103, in __call__
    self = super().__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 192, in __init__
    super().__init__(pin, active_high=active_high,
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 74, in __init__
    super().__init__(pin, pin_factory=pin_factory)
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 75, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 549, in __init__
    pin = self.pin_factory.pin(pin)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 410, in pin
    pin = self.pin_class(self, info)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/pins/lgpio.py", line 126, in __init__
    lgpio.gpio_claim_input(
  File "/usr/lib/python3/dist-packages/lgpio.py", line 755, in gpio_claim_input
    return _u2i(_lgpio._gpio_claim_input(handle&0xffff, lFlags, gpio))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/lgpio.py", line 458, in _u2i
    raise error(error_text(v))
lgpio.error: 'GPIO busy'
```

greaterthanstar, 2024-03-22:
Hmm I just noticed in the reddit thread that wpa2 said it requires a reflash. I think my current setup may have started with 2.8.6 and auto-updated itself to 2.8.7.2. Perhaps that is why I am still seeing it?

Atlant87, 2024-03-22:
Same error. Problem with "AGE" plugin, i uninstalled it, And vuala :)  (when try to install, start rebooted cyclically)

greaterthanstar, 2024-03-22:
> Same error. Problem with "AGE" plugin, i uninstalled it, And vuala :) (when try to install, start rebooted cyclically)

So you think it is plugin related? I specifically disabled the plugin (in config.toml) I thought was causing the issue but it still occurred. I am going to try flashing 2.8.7.2 to a new card and then restoring my old setup and see what happens.

Atlant87, 2024-03-22:
> So you think it is plugin related? I specifically disabled the plugin (in config.toml) I thought was causing the issue but it still occurred. I am going to try flashing 2.8.7.2 to a new card and [...]

frirstly try to uninstall.

I fllashed, and still when install plugin age forever reboot....

greaterthanstar, 2024-03-22:
ok I went back to my original image, deleted the wanmon.py plugin file from `custom-plugins` and removed all references to it in config.toml and tried to run with the --debug flag. same issue. so it wasn't from the plugin. now i'm going to try with a freshly flashed 2.8.7.2

greaterthanstar, 2024-03-22:
alright I flashed the 2.8.7.2 (stock, unmodified) image on my pi0v2+pisugar3+waveshare_v2 pwny and tried running the --debug cmd and it throws the same GPIO Busy error. Specifically it's after "initializing v2in13_V2 display" so that's what it appears to be related to, not a plugin.

jayofelony, 2024-03-23:
> **Describe the bug** I wasn't getting a plugin called wanmon.py to work (shows internet status on UI) so i tried running `sudo pwnagotchi --debug` and it is crashing with an eventual error of "GPIO [...]

You are using waveshare_4 as your display type, while you say you have a V2 display. You should use waveshare_2 as your display type in that case.

greaterthanstar, 2024-03-23:
@jayofelony actually I do have `waveshare_v2` in my config.toml. The log I posted was from the reddit post by someone else. I thought it was identical to mine, but it's not. Here's my debug log:
```
2024-03-15 15:28:39,413] [INFO] [update] plugin loaded.
[2024-03-15 15:28:39,415] [INFO] Session-stats plugin loaded.
INFO:root:initializing waveshare v2in13_V2 display
DEBUG:root:Got here
[2024-03-15 15:28:39,436] [INFO] initializing waveshare v2in13_V2 display
[2024-03-15 15:28:39,436] [DEBUG] Got here
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 408, in pin
    pin = self.pins[info]
          ~~~~~~~~~^^^^^^
KeyError: PinInfo(number=11, name='GPIO17', names=frozenset({'17', 'J8:11', 17, 'BOARD11', 'GPIO17', 'WPI0', 'BCM17'}), pull='', row=6, col=1, interfaces=frozenset({'', 'uart', 'dpi', 'gpio', 'spi'}))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 338, in <module>
    pwnagotchi_cli()
  File "/usr/local/bin/pwnagotchi", line 318, in pwnagotchi_cli
    display = Display(config=config, state={'name': '%s>' % pwnagotchi.name()})
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 18, in __init__
    self.init_display()
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 240, in init_display
    self._implementation.initialize()
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/waveshare2in13_V2.py", line 35, in initialize
    from pwnagotchi.ui.hw.libs.waveshare.v2in13_V2.epd2in13_V2 import EPD
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/v2in13_V2/epd2in13_V2.py", line 32, in <module>
    from .. import epdconfig
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/epdconfig.py", line 127, in <module>
    implementation = RaspberryPi()
                     ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/epdconfig.py", line 52, in __init__
    self.GPIO_RST_PIN = gpiozero.LED(self.RST_PIN)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 103, in __call__
INFO:root:BT-TETHER: Successfully loaded ...
[2024-03-15 15:28:39,936] [INFO] BT-TETHER: Successfully loaded ...
    self = super().__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 192, in __init__
    super().__init__(pin, active_high=active_high,
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 74, in __init__
    super().__init__(pin, pin_factory=pin_factory)
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 75, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 549, in __init__
    pin = self.pin_factory.pin(pin)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 410, in pin
    pin = self.pin_class(self, info)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/pins/lgpio.py", line 126, in __init__
    lgpio.gpio_claim_input(
  File "/usr/lib/python3/dist-packages/lgpio.py", line 755, in gpio_claim_input
    return _u2i(_lgpio._gpio_claim_input(handle&0xffff, lFlags, gpio))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/lgpio.py", line 458, in _u2i
    raise error(error_text(v))
lgpio.error: 'GPIO busy'
```
sorry about the confusion. it was just really hard to get the log onto a computer i could post from.

jayofelony, 2024-03-23:
I have a waveshare v2 as well and my pi02w works perfectly normal on 2.8.8.

Did you auto update or flash to the latest?

greaterthanstar, 2024-03-23:
I let it auto-update from an original install of 2.8.6. 

I also tried flashing 2.8.7.2 fresh and running that with no plugins or any changes other than the basic setup and it still had the same error. 

I haven't tried 2.8.8. I could try that later. is It possible it's due to the Pisugar3?

greaterthanstar, 2024-03-23:
I donno if this helps but here's the `gpioinfo` output
```
gpiochip0 - 54 lines:
        line   0:     "ID_SDA"       unused   input  active-high 
        line   1:     "ID_SCL"       unused   input  active-high 
        line   2:       "SDA1"       unused   input  active-high 
        line   3:       "SCL1"       unused   input  active-high 
        line   4:  "GPIO_GCLK"       unused   input  active-high 
        line   5:      "GPIO5"       unused   input  active-high 
        line   6:      "GPIO6"       unused   input  active-high 
        line   7:  "SPI_CE1_N"       unused   input  active-high 
        line   8:  "SPI_CE0_N"         "lg"  output  active-high [used]
        line   9:   "SPI_MISO"       unused   input  active-high 
        line  10:   "SPI_MOSI"       unused   input  active-high 
        line  11:   "SPI_SCLK"       unused   input  active-high 
        line  12:     "GPIO12"       unused   input  active-high 
        line  13:     "GPIO13"       unused   input  active-high 
        line  14:       "TXD1"       unused   input  active-high 
        line  15:       "RXD1"       unused   input  active-high 
        line  16:     "GPIO16"       unused   input  active-high 
        line  17:     "GPIO17"         "lg"  output  active-high [used]
        line  18:     "GPIO18"         "lg"  output  active-high [used]
        line  19:     "GPIO19"       unused   input  active-high 
        line  20:     "GPIO20"       unused   input  active-high 
        line  21:     "GPIO21"       unused   input  active-high 
        line  22:     "GPIO22"       unused   input  active-high 
        line  23:     "GPIO23"       unused   input  active-high 
        line  24:     "GPIO24"         "lg"   input  active-high [used pull-down]
        line  25:     "GPIO25"         "lg"  output  active-high [used]
        line  26:     "GPIO26"       unused   input  active-high 
        line  27:     "GPIO27"       unused   input  active-high 
        line  28: "HDMI_HPD_N"        "hpd"   input   active-low [used]
        line  29: "STATUS_LED_N" "ACT" output active-low [used]
        line  30:       "CTS0"       unused   input  active-high 
        line  31:       "RTS0"       unused   input  active-high 
        line  32:       "TXD0"       unused   input  active-high 
        line  33:       "RXD0"       unused   input  active-high 
        line  34:    "SD1_CLK"       unused   input  active-high 
        line  35:    "SD1_CMD"       unused   input  active-high 
        line  36:  "SD1_DATA0"       unused   input  active-high 
        line  37:  "SD1_DATA1"       unused   input  active-high 
        line  38:  "SD1_DATA2"       unused   input  active-high 
        line  39:  "SD1_DATA3"       unused   input  active-high 
        line  40:  "CAM_GPIO1" "cam1_regulator" output active-high [used]
        line  41:      "WL_ON"       unused  output  active-high 
        line  42:      "BT_ON"   "shutdown"  output  active-high [used]
        line  43:   "WIFI_CLK"       unused   input  active-high 
        line  44:       "SDA0"       unused   input  active-high 
        line  45:       "SCL0"       unused   input  active-high 
        line  46:   "SMPS_SCL"       unused   input  active-high 
        line  47:   "SMPS_SDA"       unused  output  active-high 
        line  48:   "SD_CLK_R"       unused   input  active-high 
        line  49:   "SD_CMD_R"       unused   input  active-high 
        line  50: "SD_DATA0_R"       unused   input  active-high 
        line  51: "SD_DATA1_R"       unused   input  active-high 
        line  52: "SD_DATA2_R"       unused   input  active-high 
        line  53: "SD_DATA3_R"       unused   input  active-high 
```

greaterthanstar, 2024-03-23:
it's working now. the problem was the `pwnagotchi` service was already running when i ran the command with the debug flag. so it must have been trying to access the display twice. pretty dumb move. maybe a "fix" would be to have it detect if it's already running when it first executes. i'm embarrassed.

closed 2024-03-23 by greaterthanstar