jayofelony/pwnagotchi #112: [BUG] Pwnagotchi doesn't boot anymore

julien-stiker, 2024-03-27:
Hello,

since this night my pwnagotchi doesn't boot anymore... I think, he downloaded the last release...

The screen blink... The connection doesn't establish if I use micro-usb...

My config : 
Piz 2 WH
Waveshare 4
PiSugar 3

Salam9000, 2024-03-27:
Same here, 
Piz WH
Waveshare 4
PiSugar S

wpa-2, 2024-03-28:
> Hello,
> 
> since this night my pwnagotchi doesn't boot anymore... I think, he downloaded the last release 2.8.9 this night...
> 
> The screen blink... The connection doesn't establish if I use micro-usb...
> 
> My config : Piz 2 WH Waveshare 4 PiSugar 3

> Same here, Piz WH Waveshare 4 PiSugar S

`sudo pwnagotchi --debug`

rsun0525, 2024-04-03:
This is related to the bug I filed, where my manual change to uncomment disable-wifi was overwritten, after I modified it again to remove the comment, everything started working.

closed 2024-05-26 by jayofelony