jayofelony/pwnagotchi #115: [BUG] RPI02W Waveshare v3 screen not updating

abonforti, 2024-03-28:
**Describe the bug**
Screen is frozen, however accessing via http://10.0.0.2:8080 it shows changes

**To Reproduce**

1. Flash the latest version
2. Connect the Waveshare v2

**Expected behavior**
Screen should auto update

**Environment (please complete the following information):**
 - Pwnagotchi version 2.8.9 64bit
 - Type of hardware RPI02W

**Additional context**
pwnagotchi --debug output can be found here: https://nopaste.net/W8amOjrQJN

jayofelony, 2024-03-28:
main.plugins.bt-tether.devices.ale-iPhone13.enabled

You cannot rename it to ale-iPhone, keep it ios.

abonforti, 2024-03-28:
I have replaced that, and noticed some more stability within the network, but the issue on the screen persists.

abonforti, 2024-03-28:
> I don't have a waveshare v2 so I can't personally test this, but in your config.toml try changing the line ui.display.type = "waveshare_3"config.toml to be ui.display.type = "waveshare_2"

But that is strange because with a pizero w and the 32bit image, I had no problem…

Sent from [Proton Mail](https://proton.me/mail/home) for iOS

abonforti, 2024-03-28:
My bad, that was just a typo, I’ve corrected this and confirm I am using a v3 screen.

jayofelony, 2024-03-29:
Ok you can access the webui over usb or through bt tether? And it worked on a pi0w,so that means you have bad soldering on the gpio pins. Because the software has no issues with e-ink displays.

abonforti, 2024-03-29:
Just to be on the safer side, I've re-soldered all the pins. But it's strange because as soon as I start the pwnagotchi daemon, the screen works and updates (I can see the little pipe character blinking), then it kind of freezes...

closed 2024-03-29 by abonforti