jayofelony/pwnagotchi #119: [BUG]

rsun0525, 2024-04-01:
**Describe the bug**
I modify /boot/firmware/config.txt to remove the comment of dtoverlay=disable-wifi under [all]

but sometime later, I notice it's commented again, almost like this file is getting overwritten by some default, is there a more proper way to enable disable-wifi in your config?  I wasn't expecting to change itself.

jayofelony, 2024-04-03:
It get overwritten with every auto update

rsun0525, 2024-04-03:
yes, I would think once it's set, shouldn't touch it in an autoupdate, but from now on I know to check it.

jayofelony, 2024-04-03:
RPi tends to change stuffso I might need to change stuff in it to keep stuff working and to avoid unnecessary reflashing.

closed 2024-04-03 by jayofelony