jayofelony/pwnagotchi #120: [DISPLAY SUPPORT] Inky PHAT v2

cviper13, 2024-04-02:
**Is your feature request related to a problem? Please describe.**
At "Unoffical Pwnagotchi Discord Server" a user came to the help channel with their problema bout Inky PHAT v2. I found out it wasnt supported but can be added easily and solved the user's problem.

**Describe the solution you'd like**
https://github.com/evilsocket/pwnagotchi/issues/979#issuecomment-825268356

**Describe alternatives you've considered**
Using the normal inkyphat config makes you unable to use full of your screen

**Additional context**
I have an fork ready to fix this I can open a PR if you want.

Alex-Smailes, 2024-04-25:
Hi,
I wrote an update for the Inky PHAT that allows it to remain compatible with both v1 and the higher resolution v2 displays and can auto-detect the variant of the display for v2 PHAT's that feature an eeprom. It should also work with the larger Inky WHAT's.
[Alex-Smailes/pwnagotchi/commit/32946e5](https://github.com/Alex-Smailes/pwnagotchi/commit/32946e5b4348ac0777d4475344308e27d96caf3b)

cviper13, 2024-04-29:
That is awesome! Is it merged to main repo?

closed 2024-04-29 by cviper13