jayofelony/pwnagotchi #124: UART GPS modules no longer work with v2.8.9

BilboByteSeeker, 2024-04-05:
UART GPS module no longer works with v2.8.9

First of all, thank you very much for your awesome work with the pwnagotchi project.

With the new version 2.8.9 GPS modules (NEO-6M) are no longer recognized which are controlled via UART (GPIO).
Tested with a Raspberry PI zero W and 3 Raspberry PI zero 2 W and 4 different NEO-6M modules

With version 2.8.6 the GPS modules are immediately recognized as /dev/ttyS0 as soon as you update to version 2.8.9 the GPS modules are no longer recognized

i have freshly flashed version 2.8.9 and version 2.8.6 and activated the serial interface with sudo raspi-config after a restart the recognition of the module works with version 2.8.6 but not with version 2.8.9

is there a solution for this?

jayofelony, 2024-04-06:
That is probably because GPIO settings changed as of 2.8.7 and up. To make displays work on the pi 5. I am unable to diagnose this as I don't have the 3rd party hardware. So you are dependent of the community to help on this to come up with a fix.

BilboByteSeeker, 2024-04-08:
I have found the solution

The problem is that when you enable the serial port with sudo raspi-config, enable_uart=1 is added at the end of /boot/firmware/config.txt under [pi5]

I added enable_uart=1 under the [all] tab and now it works.

jayofelony, 2024-04-08:
Okay, nice job. I will add that next update.

closed 2024-05-26 by jayofelony