jayofelony/pwnagotchi #131: [BUG]

j0p3s, 2024-04-17:
**Describe the bug**
Bluetooth fails to connect to android phone (gives error: bluetooth error: br-connection-profile-unavailable)

**To Reproduce**
Steps to reproduce the behavior:
1. Have a Pi zero 2w running jayofelony 2.8.9 64 bits with a waveshare v4 e-ink display
2. Samsung Note 20 Ultra 5G
3. Run the wizard and let the pwnagotchi fully boot, boot into ssh and as root run bluetoothctl and pair the phone with pair *mac*
trust *mac*
connect *mac* gives error [bluetooth error: br-connection-profile-unavailable]

**Expected behavior**
Expecting the pwnagotchi to connect to the phone automatically after the wizard is run but it doesn't

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Pwnagotchi version: 2.8.9
 - OS version 2.8.9
 - Type of hardware: Pi zero 2 w with waveshark v4
 - Any additional hardware used: Samsung Note 20 Ultra 5G

**Additional context**
Installing the lib "libspa-0.2-bluetooth" seems to fix the issue
Please include it?

j0p3s, 2024-04-17:
My config.toml:

# Do not edit this file if you do not know what you are doing!!!

main.name = "pwnagotchi"
main.whitelist = [
	"",
	"",
]

main.plugins.internet-connection.enabled = true

main.custom_plugin_repos = [
    "https://github.com/itsdarklikehell/pwnagotchi-plugins/archive/master.zip",
]

main.custom_plugins = "/usr/local/share/pwnagotchi/custom-plugins/"

# Configuration for Android Phone
main.plugins.bt-tether.enabled = true
main.plugins.bt-tether.devices.android-phone.enabled = true
main.plugins.bt-tether.devices.android-phone.search_order = 1
main.plugins.bt-tether.devices.android-phone.ip = "192.168.44.44"  # Static IP of the Pwnagotchi
main.plugins.bt-tether.devices.android-phone.netmask = 24  # Netmask of the PAN
main.plugins.bt-tether.devices.android-phone.interval = 1  # Search interval in minutes
main.plugins.bt-tether.devices.android-phone.scantime = 10  # Duration of each search in seconds
main.plugins.bt-tether.devices.android-phone.max_tries = 10  # Maximum attempts to find the phone
main.plugins.bt-tether.devices.android-phone.share_internet = true  # Enable internet sharing via Bluetooth
main.plugins.bt-tether.devices.android-phone.priority = 1  # Priority level for tethering
main.plugins.bt-tether.devices.android-phone.mac = "70:1f:********"

ui.display.enabled = true
ui.display.type = "waveshare_4"
ui.invert = true

ui.web.enabled = true
ui.web.address = "0.0.0.0"
ui.web.username = "j0p3s"
ui.web.password = "*********"
ui.web.origin = ""
ui.web.port = 8080
ui.web.on_frame = ""

bettercap.scheme = "http"
bettercap.hostname = "localhost"
bettercap.port = 8081
bettercap.username = "j0p3s"
bettercap.password = "********"
bettercap.handshakes = "/root/handshakes"
bettercap.silence = [
  "ble.device.new",
  "ble.device.lost",
  "ble.device.disconnected",
  "ble.device.connected",
  "ble.device.service.discovered",
  "ble.device.characteristic.discovered",
  "wifi.client.new",
  "wifi.client.lost",
  "wifi.client.probe",
  "wifi.ap.new",
  "wifi.ap.lost",
  "mod.started"
]

fs.memory.enabled = true
fs.memory.mounts.log.enabled = true
fs.memory.mounts.log.mount = "/var/log"
fs.memory.mounts.log.size = "50M"
fs.memory.mounts.log.sync = 60
fs.memory.mounts.log.zram = true
fs.memory.mounts.log.rsync = true

fs.memory.mounts.data.enabled = true
fs.memory.mounts.data.mount = "/var/tmp/pwnagotchi"
fs.memory.mounts.data.size = "10M"
fs.memory.mounts.data.sync = 3600
fs.memory.mounts.data.zram = true
fs.memory.mounts.data.rsync = true

Miles8of9, 2024-04-19:
Nice try... have same problem. Tried installing the lib you suggested, rebooted, unpaired, paired again successfully etc etc.. BUT...

connect MY:ANDROID:BT:MAC:ADDRESS
Attempting to connect to MY:ANDROID:BT:MAC:ADDRESS
[CHG] Device MY:ANDROID:BT:MAC:ADDRESS Connected: yes
[CHG] Device MY:ANDROID:BT:MAC:ADDRESS ServicesResolved: yes
Failed to connect: org.bluez.Error.NotAvailable br-connection-profile-unavailable
[CHG] Device MY:ANDROID:BT:MAC:ADDRESS ServicesResolved: no
[CHG] Device MY:ANDROID:BT:MAC:ADDRESS Connected: no

Journal

bluetoothd[666]: bnep0 connected
bluetoothd[666]: bnep0 disconnected
ModemManager[504]: <info>  [base-manager] port bnep0 released by device '/sys/devices/platform/soc'
ModemManager[504]: <warn>  [plugin-manager] task 2,bnep0: failed: Operation was cancelled
ModemManager[504]: <info>  [base-manager] couldn't check support for device '/sys/devices/platform/soc' : Operation was cancelled

j0p3s, 2024-04-20:
@Miles8of9 have no clue why it didnt work the second time..

closed 2024-04-20 by j0p3s