jayofelony/pwnagotchi #134: [BUG] Waveshare v4 error (debug info inside)

j0p3s, 2024-04-20:
**Describe the bug**
did the wizard and enabled FTP and now i'm stuck in AI Mode

**To Reproduce**
Steps to reproduce the behavior:
1. Enable FTP
2. Do the wizard without BT tether and only enable waveshare_4
3. Stuck in AI Mode

**Expected behavior**
wanted the pwnagotchi to boot and its stuck in AI mode

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Pwnagotchi version: pi zero 2 w
 - OS version 2.8.9
 - Type of hardware pi zero 2 w with waveshare v4
 - Any additional hardware used

**Additional context**
```
DEBUG:root:[FS] Daemonized...
[2024-03-16 14:22:41,713] [DEBUG] [FS] Daemonized...
DEBUG:root:[FS] Syncing /var/tmp/pwnagotchi -> /run/pwnagotchi/disk/pwnagotchi
[2024-03-16 14:22:41,716] [DEBUG] [FS] Syncing /var/tmp/pwnagotchi -> /run/pwnagotchi/disk/pwnagotchi
DEBUG:root:loading plugins from /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default - enabled: ['auto-update', 'fix_services', 'grid', 'session-stats', 'webcfg']
[2024-03-16 14:22:42,583] [DEBUG] loading plugins from /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default - enabled: ['auto-update', 'fix_services', 'grid', 'session-stats', 'webcfg']
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/session-stats.py
[2024-03-16 14:22:43,138] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/session-stats.py
DEBUG:root:loaded plugin session-stats as <session-stats.SessionStats object at 0x7f81820950>
[2024-03-16 14:22:44,660] [DEBUG] loaded plugin session-stats as <session-stats.SessionStats object at 0x7f81820950>
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/webcfg.py
[2024-03-16 14:22:44,662] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/webcfg.py
DEBUG:root:loaded plugin webcfg as <webcfg.WebConfig object at 0x7f81821290>
[2024-03-16 14:22:45,211] [DEBUG] loaded plugin webcfg as <webcfg.WebConfig object at 0x7f81821290>
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/fix_services.py
[2024-03-16 14:22:45,212] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/fix_services.py
DEBUG:root:loaded plugin fix_services as <fix_services.FixServices object at 0x7f81822010>
[2024-03-16 14:22:45,409] [DEBUG] loaded plugin fix_services as <fix_services.FixServices object at 0x7f81822010>
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/grid.py
[2024-03-16 14:22:45,411] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/grid.py
DEBUG:root:loaded plugin grid as <grid.Grid object at 0x7f81823210>
[2024-03-16 14:22:45,543] [DEBUG] loaded plugin grid as <grid.Grid object at 0x7f81823210>
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/auto-update.py
[2024-03-16 14:22:45,544] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/auto-update.py
DEBUG:root:loaded plugin auto-update as <auto-update.AutoUpdate object at 0x7f7adf0b10>
[2024-03-16 14:22:45,660] [DEBUG] loaded plugin auto-update as <auto-update.AutoUpdate object at 0x7f7adf0b10>
DEBUG:root:loading plugins from /usr/local/share/pwnagotchi/custom-plugins/ - enabled: ['auto-update', 'fix_services', 'grid', 'session-stats', 'webcfg']
[2024-03-16 14:22:45,661] [DEBUG] loading plugins from /usr/local/share/pwnagotchi/custom-plugins/ - enabled: ['auto-update', 'fix_services', 'grid', 'session-stats', 'webcfg']
INFO:root:[Fix_Services] plugin loaded.
[2024-03-16 14:22:45,782] [INFO] [Fix_Services] plugin loaded.
INFO:root:grid plugin loaded.
INFO:root:[update] plugin loaded.
[2024-03-16 14:22:45,783] [INFO] grid plugin loaded.
INFO:root:webcfg: Plugin loaded.
[2024-03-16 14:22:45,787] [INFO] [update] plugin loaded.
INFO:root:Session-stats plugin loaded.
[2024-03-16 14:22:45,789] [INFO] webcfg: Plugin loaded.
[2024-03-16 14:22:45,791] [INFO] Session-stats plugin loaded.
DEBUG:root:INVERT BLACK/WHITES:True
[2024-03-16 14:22:45,841] [DEBUG] INVERT BLACK/WHITES:True
INFO:root:initializing waveshare v2in13_V4 display
DEBUG:root:Got here
[2024-03-16 14:22:45,890] [INFO] initializing waveshare v2in13_V4 display
[2024-03-16 14:22:45,890] [DEBUG] Got here
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 408, in pin
    pin = self.pins[info]
          ~~~~~~~~~^^^^^^
KeyError: PinInfo(number=11, name='GPIO17', names=frozenset({'BOARD11', 17, 'WPI0', '17', 'J8:11', 'BCM17', 'GPIO17'}), pull='', row=6, col=1, interfaces=frozenset({'', 'gpio', 'dpi', 'spi', 'uart'}))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 337, in <module>
    pwnagotchi_cli()
  File "/usr/local/bin/pwnagotchi", line 316, in pwnagotchi_cli
    display = Display(config=config, state={'name': '%s>' % pwnagotchi.name()})
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 18, in __init__
    self.init_display()
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 279, in init_display
    self._implementation.initialize()
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/waveshare2in13_V4.py", line 35, in initialize
    from pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13_V4.epd2in13_V4 import EPD
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/epaper/v2in13_V4/epd2in13_V4.py", line 32, in <module>
    from pwnagotchi.ui.hw.libs.waveshare.epaper import epdconfig
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/epaper/epdconfig.py", line 127, in <module>
    implementation = RaspberryPi()
                     ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/epaper/epdconfig.py", line 52, in __init__
    self.GPIO_RST_PIN = gpiozero.LED(self.RST_PIN)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 103, in __call__
    self = super().__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 192, in __init__
    super().__init__(pin, active_high=active_high,
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 74, in __init__
    super().__init__(pin, pin_factory=pin_factory)
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 75, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 549, in __init__
    pin = self.pin_factory.pin(pin)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 410, in pin
    pin = self.pin_class(self, info)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/pins/lgpio.py", line 126, in __init__
    lgpio.gpio_claim_input(
  File "/usr/lib/python3/dist-packages/lgpio.py", line 755, in gpio_claim_input
    return _u2i(_lgpio._gpio_claim_input(handle&0xffff, lFlags, gpio))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/lgpio.py", line 458, in _u2i
    raise error(error_text(v))
lgpio.error: 'GPIO busy'
```

Get this error and in the screen i have AI mode only and won't move from there

j0p3s, 2024-04-20:
Changing the line: waveshare_4 to waveshare_3 and got his on debug:

```
DEBUG:root:[FS] Daemonized...
DEBUG:root:[FS] Syncing /var/tmp/pwnagotchi -> /run/pwnagotchi/disk/pwnagotchi
[2024-03-16 14:36:46,128] [DEBUG] [FS] Syncing /var/tmp/pwnagotchi -> /run/pwnagotchi/disk/pwnagotchi
DEBUG:root:loading plugins from /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default - enabled: ['auto-update', 'fix_services', 'grid', 'session-stats', 'webcfg']
[2024-03-16 14:36:46,142] [DEBUG] loading plugins from /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default - enabled: ['auto-update', 'fix_services', 'grid', 'session-stats', 'webcfg']
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/session-stats.py
[2024-03-16 14:36:46,144] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/session-stats.py
DEBUG:root:loaded plugin session-stats as <session-stats.SessionStats object at 0x7f8e0acc10>
[2024-03-16 14:36:46,148] [DEBUG] loaded plugin session-stats as <session-stats.SessionStats object at 0x7f8e0acc10>
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/webcfg.py
[2024-03-16 14:36:46,150] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/webcfg.py
DEBUG:root:loaded plugin webcfg as <webcfg.WebConfig object at 0x7f8e0ad510>
[2024-03-16 14:36:46,153] [DEBUG] loaded plugin webcfg as <webcfg.WebConfig object at 0x7f8e0ad510>
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/fix_services.py
[2024-03-16 14:36:46,154] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/fix_services.py
DEBUG:root:loaded plugin fix_services as <fix_services.FixServices object at 0x7f8e0ae290>
[2024-03-16 14:36:46,161] [DEBUG] loaded plugin fix_services as <fix_services.FixServices object at 0x7f8e0ae290>
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/grid.py
[2024-03-16 14:36:46,162] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/grid.py
DEBUG:root:loaded plugin grid as <grid.Grid object at 0x7f8e0af490>
[2024-03-16 14:36:46,170] [DEBUG] loaded plugin grid as <grid.Grid object at 0x7f8e0af490>
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/auto-update.py
[2024-03-16 14:36:46,171] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/auto-update.py
DEBUG:root:loaded plugin auto-update as <auto-update.AutoUpdate object at 0x7f876a4dd0>
[2024-03-16 14:36:46,176] [DEBUG] loaded plugin auto-update as <auto-update.AutoUpdate object at 0x7f876a4dd0>
DEBUG:root:loading plugins from /usr/local/share/pwnagotchi/custom-plugins/ - enabled: ['auto-update', 'fix_services', 'grid', 'session-stats', 'webcfg']
[2024-03-16 14:36:46,179] [DEBUG] loading plugins from /usr/local/share/pwnagotchi/custom-plugins/ - enabled: ['auto-update', 'fix_services', 'grid', 'session-stats', 'webcfg']
INFO:root:grid plugin loaded.
INFO:root:[Fix_Services] plugin loaded.
INFO:root:webcfg: Plugin loaded.
INFO:root:[update] plugin loaded.
[2024-03-16 14:36:46,190] [INFO] grid plugin loaded.
[2024-03-16 14:36:46,195] [INFO] [Fix_Services] plugin loaded.
INFO:root:Session-stats plugin loaded.
[2024-03-16 14:36:46,201] [INFO] webcfg: Plugin loaded.
[2024-03-16 14:36:46,200] [INFO] [update] plugin loaded.
[2024-03-16 14:36:46,206] [INFO] Session-stats plugin loaded.
DEBUG:root:INVERT BLACK/WHITES:True
[2024-03-16 14:36:46,215] [DEBUG] INVERT BLACK/WHITES:True
INFO:root:initializing waveshare v2in13_V3 display
DEBUG:root:Got here
[2024-03-16 14:36:46,227] [INFO] initializing waveshare v2in13_V3 display
[2024-03-16 14:36:46,228] [DEBUG] Got here
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 408, in pin
    pin = self.pins[info]
          ~~~~~~~~~^^^^^^
KeyError: PinInfo(number=11, name='GPIO17', names=frozenset({'17', 'J8:11', 17, 'WPI0', 'BCM17', 'GPIO17', 'BOARD11'}), pull='', row=6, col=1, interfaces=frozenset({'', 'spi', 'uart', 'dpi', 'gpio'}))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 337, in <module>
    pwnagotchi_cli()
  File "/usr/local/bin/pwnagotchi", line 316, in pwnagotchi_cli
    display = Display(config=config, state={'name': '%s>' % pwnagotchi.name()})
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 18, in __init__
    self.init_display()
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 279, in init_display
    self._implementation.initialize()
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/waveshare2in13_V3.py", line 36, in initialize
    from pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13_V3.epd2in13_V3 import EPD
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/epaper/v2in13_V3/epd2in13_V3.py", line 32, in <module>
    from pwnagotchi.ui.hw.libs.waveshare.epaper import epdconfig
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/waveshare/epaper/epdconfig.py", line 127, in <module>
    implementation = RaspberryPi()
[... 3 lines cut ...]
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...]
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/pins/lgpio.py", line 126, in __init__
    lgpio.gpio_claim_input(
  File "/usr/lib/python3/dist-packages/lgpio.py", line 755, in gpio_claim_input
    return _u2i(_lgpio._gpio_claim_input(handle&0xffff, lFlags, gpio))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/lgpio.py", line 458, in _u2i
    raise error(error_text(v))
lgpio.error: 'GPIO busy'
```

j0p3s, 2024-04-20:
my config.toml is just this:

# Do not edit this file if you do not know what you are doing!!!

main.name = "pwnagotchi"
ui.display.enabled = true
ui.display.type = "waveshare_3"
ui.invert = true
ui.fps = 1

j0p3s, 2024-04-20:
Just to clear things out:

I flashed 2.8.9 from zero! Let the pwnagotchi run for 48 hours straight... 

Disconnected it to use a powerbank and out of the blue gives that error in debug and its stuck in AI mode!

j0p3s, 2024-04-20:
Only ran the wizard and:

```
Do you want to restore the previous configuration?

[Y/N]: n
This will create a new configuration file and overwrite your current backup, are you sure?

[Y/N]: y
Welcome to the interactive installation of your personal Pwnagotchi configuration!
My name is Jayofelony, how may I call you?

Pwnagotchi name (no spaces): pwnagotchi
I shall go by pwnagotchi from now on!
How many networks do you want to whitelist? We will also ask a MAC for each network?
Each SSID and BSSID count as 1 network.

Be sure to use digits as your answer.

Amount of networks: 0
Do you want to enable BT-Tether?

[Y/N] n
Do you want to enable a display?

[Y/N]: y
What display do you use?

Be sure to check for the correct display type @
https://github.com/jayofelony/pwnagotchi/blob/master/pwnagotchi/utils.py#L240-L431

Display type: waveshare_4
Do you want to invert the display colors?
N = Black background
Y = White background

[Y/N]: y
Your configuration is done, and I will restart in 5 seconds.
```

jayofelony, 2024-04-20:
The debug can only be ran if you stopped pwn service;

`sudo systemctl stop pwnagotchi`

macedonga, 2024-05-06:
I'm having this issue as well.
```
[2024-03-15 15:16:58,959] [INFO] initializing waveshare 2.13inb v2in13_V4 display
INFO:root:BT-TETHER: Successfully loaded ...
[2024-03-15 15:16:59,110] [INFO] BT-TETHER: Successfully loaded ...
DEBUG:pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13b_v4.epd2in13b_V4:e-Paper busy
[2024-03-15 15:16:59,199] [DEBUG] e-Paper busy
DEBUG:pwnagotchi.ui.hw.libs.waveshare.epaper.v2in13b_v4.epd2in13b_V4:e-Paper busy release
```

This is the output of the pwnagotchi process after stopping it with systemctl and starting it with the debug flag. The screen is blank, but I know it works just fine. I didn't do the slim mod or anything, just plugged it in in the RPi Zero 2W.

closed 2024-05-26 by jayofelony