jayofelony/pwnagotchi #135: [BUG] Display hat mini error

iiTzTune, 2024-04-22:
**Describe the bug**
Go through config build wizard on fresh image flash, board reboots then blank screen.

**Expected behavior**
Device to boot up

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Latest release 2.8.9 64bit
 - Displayhatmini, Pi zero 2w
 - PiSugar2

**Additional context**

Attached debug

mount: /run/pwnagotchi/disk/pwnagotchi: mount point does not exist.
       dmesg(1) may have more information after failed mount system call.
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
DEBUG:root:loading plugins from /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default - enabled: ['auto-update', 'fix_services', 'grid', 'session-stats', 'webcfg']
[2024-04-22 09:33:39,149] [DEBUG] loading plugins from /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default - enabled: ['auto-update', 'fix_services', 'grid', 'session-stats', 'webcfg']
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/session-stats.py
[2024-04-22 09:33:39,151] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/session-stats.py
DEBUG:root:loaded plugin session-stats as <session-stats.SessionStats object at 0x7f856c0a10>
[2024-04-22 09:33:39,155] [DEBUG] loaded plugin session-stats as <session-stats.SessionStats object at 0x7f856c0a10>
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/webcfg.py
[2024-04-22 09:33:39,157] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/webcfg.py
DEBUG:root:loaded plugin webcfg as <webcfg.WebConfig object at 0x7f856c1310>
[2024-04-22 09:33:39,159] [DEBUG] loaded plugin webcfg as <webcfg.WebConfig object at 0x7f856c1310>
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/fix_services.py
[2024-04-22 09:33:39,160] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/fix_services.py
DEBUG:root:loaded plugin fix_services as <fix_services.FixServices object at 0x7f856c2050>
[2024-04-22 09:33:39,168] [DEBUG] loaded plugin fix_services as <fix_services.FixServices object at 0x7f856c2050>
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/grid.py
[2024-04-22 09:33:39,170] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/grid.py
DEBUG:root:loaded plugin grid as <grid.Grid object at 0x7f856c3250>
[2024-04-22 09:33:39,173] [DEBUG] loaded plugin grid as <grid.Grid object at 0x7f856c3250>
DEBUG:root:loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/auto-update.py
[2024-04-22 09:33:39,175] [DEBUG] loading /usr/local/lib/python3.11/dist-packages/pwnagotchi/plugins/default/auto-update.py
DEBUG:root:loaded plugin auto-update as <auto-update.AutoUpdate object at 0x7f834d4b50>
[2024-04-22 09:33:39,180] [DEBUG] loaded plugin auto-update as <auto-update.AutoUpdate object at 0x7f834d4b50>
DEBUG:root:loading plugins from /usr/local/share/pwnagotchi/custom-plugins/ - enabled: ['auto-update', 'fix_services', 'grid', 'session-stats', 'webcfg']
[2024-04-22 09:33:39,182] [DEBUG] loading plugins from /usr/local/share/pwnagotchi/custom-plugins/ - enabled: ['auto-update', 'fix_services', 'grid', 'session-stats', 'webcfg']
INFO:root:webcfg: Plugin loaded.
INFO:root:[update] plugin loaded.
[2024-04-22 09:33:39,192] [INFO] webcfg: Plugin loaded.
INFO:root:grid plugin loaded.
[2024-04-22 09:33:39,193] [INFO] [update] plugin loaded.
INFO:root:[Fix_Services] plugin loaded.
[2024-04-22 09:33:39,194] [INFO] grid plugin loaded.
INFO:root:Session-stats plugin loaded.
[2024-04-22 09:33:39,199] [INFO] [Fix_Services] plugin loaded.
[2024-04-22 09:33:39,206] [INFO] Session-stats plugin loaded.
WARNING:root:ui.fps is 0, the display will only update for major changes
DEBUG:root:Got here
[2024-04-22 09:33:39,225] [WARNING] ui.fps is 0, the display will only update for major changes
INFO:root:initializing Display Hat Mini
[2024-04-22 09:33:39,225] [DEBUG] Got here
[2024-04-22 09:33:39,227] [INFO] initializing Display Hat Mini
Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 337, in <module>
    pwnagotchi_cli()
  File "/usr/local/bin/pwnagotchi", line 316, in pwnagotchi_cli
    display = Display(config=config, state={'name': '%s>' % pwnagotchi.name()})
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 18, in __init__
    self.init_display()
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 279, in init_display
    self._implementation.initialize()
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/displayhatmini.py", line 37, in initialize
    self._display = ST7789(0, 1, 9, 13)
                    ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/pimoroni/displayhatmini/ST7789.py", line 123, in __init__
    self._spi = spidev.SpiDev(port, cs)
                ^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory

iiTzTune, 2024-04-22:
> You need to comment all the spi0 inside - the /boot/firmware/config.txt. and in the ST7789 you need to set the rotation to 0 instead of 90. The width should be at 320 instead of 240, and I changed the speed to 60_1000_1000 instead of 4000000.
> 
> There is a problem with the gpio buttons too. So for now the buttons can't work.
> 
> But at least, with this fix, the display can work.

Making these changes i was able to get the display to light up but nothing to display, just the back light activating. I edited the ST7789 in "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/pimoroni/displayhatmini/ST7789.py", and commented all the spi0 lines out in the config.

V0r-T3x, 2024-04-22:
The line 93 in `/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/hw/libs/pimoroni/displayhatmini/ST7789.py` look like this:
```
    def __init__(self, port, cs, dc, backlight=None, rst=None, width=320,
                 height=240, rotation=0, invert=True, spi_speed_hz=60 * 1000 * 1000,
                 offset_left=0,
                 offset_top=0):
```

And my `/boot/firmware/config.txt` contain those options:
```
dtparam=audio=on
camera_auto_detect=1
display_auto_detect=1
auto_initramfs=1
dtoverlay=vc4-kms-v3d
max_framebuffers=2
disable_fw_kms_setup=1
arm_64bit=1
disable_overscan=1
arm_boost=1

[cm4]
otg_mode=1

[all]
dtparam=i2c1=on
dtparam=i2c_arm=on
dtparam=spi=on
gpu_mem=1
dtoverlay=dwc2
```
And my `/etc/pwnagotchi/config.toml`
 ```
 ui.display.enabled = true
ui.display.rotation = 0
ui.display.type = "displayhatmini"
```
My display hat mini is working with those settings, I didn't change anything else to make it work. hope this can help you.

V0r-T3x, 2024-04-22:
@iiTzTune did you tried to boot your pwnagotchi without the pisugar2?

just saw another issue about the pwnagotchi not booting and seems that the /boot/firmware/config.txt can be overwritten... you can check this too. https://github.com/jayofelony/pwnagotchi/issues/112

iiTzTune, 2024-04-22:
@V0r-T3x copying and pasting your settings over mine resolved my issue, i did have my speed set to "the I changed the speed to 6010001000 instead of 4000000." i wonder if having it setup like you do at 60 * 1000 * 1000, makes the difference cause thats the only difference i seen between mine and yours.

closed 2024-04-23 by jayofelony