jayofelony/pwnagotchi #140: [BUG] SyntaxError: cannot assign to attribute here. Maybe you meant '==' instead of '='? after newest update (2.9.0)

formula-spectre, 2024-04-26:
**Describe the bug**
after `sudo pwnagotchi --check-update` and installing the new update, I get this error:
```
sudo pwnagotchi
Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 337, in <module>
    pwnagotchi_cli()
  File "/usr/local/bin/pwnagotchi", line 303, in pwnagotchi_cli
    from pwnagotchi.ui.display import Display
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 7, in <module>
    from pwnagotchi.ui.view import View
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/view.py", line 15, in <module>
    from pwnagotchi.ui.components import *
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/components.py", line 22
    if self.color = 0xFF:
       ^^^^^^^^^^
SyntaxError: cannot assign to attribute here. Maybe you meant '==' instead of '='?
```
I think it's python being python; for reference, I have a waveshare eink v4

**To Reproduce**
Steps to reproduce the behavior:
1. update pwnagotchi to version 2.9.0
2. enable the eink display in config.toml

**Expected behavior**
this should not be happening 

**Screenshots**
I don't have any, since it's a log; but here is the full log:
```
pi@max-pwnroom:~ $ systemctl status pwnagotchi
● pwnagotchi.service - pwnagotchi Deep Reinforcement Learning instrumenting bettercap for WiFI pwning.
     Loaded: loaded (/etc/systemd/system/pwnagotchi.service; enabled; preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Fri 2024-04-26 07:33:58 BST; 8s ago
       Docs: https://pwnagotchi.org
    Process: 655 ExecStart=/usr/bin/pwnagotchi-launcher (code=exited, status=1/FAILURE)
    Process: 660 ExecStopPost=/usr/bin/bash -c if egrep -qi  'personality.clear_on_exit[ =]*true' /etc/pwnagotchi/co>
   Main PID: 655 (code=exited, status=1/FAILURE)
        CPU: 8.892s

pi@max-pwnroom:~ $ sudo pwnagotchi
Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 337, in <module>
    pwnagotchi_cli()
  File "/usr/local/bin/pwnagotchi", line 303, in pwnagotchi_cli
    from pwnagotchi.ui.display import Display
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/display.py", line 7, in <module>
    from pwnagotchi.ui.view import View
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/view.py", line 15, in <module>
    from pwnagotchi.ui.components import *
  File "/usr/local/lib/python3.11/dist-packages/pwnagotchi/ui/components.py", line 22
    if self.color = 0xFF:
       ^^^^^^^^^^
SyntaxError: cannot assign to attribute here. Maybe you meant '==' instead of '='?
```

**Environment (please complete the following information):**
 - Pwnagotchi version: 2.9.0
 - OS version: gentoo linux (if it means the os my computer)
 - Type of hardware: RPI02W
 - Any additional hardware used: waveshare eink screen v4, UPSLite V1.3

**Additional context**
i'll try to fork and make a PR ASAP, so I can help the project!

formula-spectre, 2024-04-26:
huh, I just cloned the repo and opened the file but there is already a `==` at line 22? did something go wrong when I updated my pwnagotchi?

jayofelony, 2024-04-26:
Change the version of your pwn to 2.8.9 manually and let it auto update.

formula-spectre, 2024-04-26:
> Change the version of your pwn to 2.8.9 manually and let it auto update. I forgot to add 2.9.1 as a new version lol 😂

ah okay. do I just edit `/etc/os-release`? and you mean that you need to release 2.9.1 or do I just update to 2.9.0? sorry if I did not understand what you meant

jayofelony, 2024-04-26:
`/usr/local/lib/python3.11/dist-packages/pwnagotchi/_version.py` 

edit this file to 2.8.9

formula-spectre, 2024-04-26:
> `/usr/local/lib/python3.11/dist-packages/pwnagotchi/_version.py`
> 
> edit this file to 2.8.9

okay, then I wait for you to release 2.9.1?

jayofelony, 2024-04-26:
yeah, but until then yours should work

formula-spectre, 2024-04-26:
gotcha!

closed 2024-04-26 by formula-spectre

sakkarose, 2024-04-27:
oh ty for the fix xD, i was panic bc i was thinking like my python version is updated for like a month on there already and i don't find any current issue atm also