jayofelony/pwnagotchi #141: [BUG]

jfhenriques, 2024-04-26:
**Describe the bug**
The image is lacking wpasupplicant and cryptsetup

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the steps to activate encryption
2. cryptsetup is not installed
3. after reboot, hotspot is not launched, wpasupplicant package is also not installed

**Expected behavior**
Cryptsetup and wpa supplicant is installed.

**Environment (please complete the following information):**
 - 2.8.9

jayofelony, 2024-04-26:
Because wpasupplicant interferes with monitor mode

jfhenriques, 2024-04-26:
Oh, it makes sense. I was following the decryption guide, but it was not mentioned.

I installed cryptsetup, wpa supplicant and dnsmasq, disabled the startup service for all and it seems to be working so far, do you know in which context the interference occurs?

jayofelony, 2024-04-26:
It can hijack the wlan0 before it gets to monitor mode, and its said on nexmon gh to remove it.

jfhenriques, 2024-04-26:
Ok thanks for the explanation.

I think that, at least for the rpi zero with no any other external network devices or any changes  in the network files, disabling and masking the services is doing the trick, as it is preventing them from being loaded or started by any other service.

The only thing I still don't understand is the relation between wpa_supplicant and dbus, but I also see that masking the wpa_supplicant service, also creates a link from "dbus-fi.w1.wpa_supplicant1.service -> /dev/null".

I'll close the ticket, at least if someone else comes looking for the same, it will find the explanation to why it was not included.

closed 2024-04-26 by jfhenriques