jayofelony/pwnagotchi #145: [BUG]

Sharkrider120, 2024-04-26:
**Describe the bug**
A clear and concise description of what the bug is.
 icant get my waveshare v4 to connect like it wont update the screen you do the thing where it flashes at boot
**To Reproduce**
Steps to reproduce the behavior:
1. ...
2. ...
3. ...
i dont know how to reproduces it i just have to turn the unit on
**Expected behavior**
A clear and concise description of what you expected to happen.
i expected it to work
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Pwnagotchi version 2.9.1
 - OS version what is the os version i though the pwnagothi software was
 - Type of hardware RP0, v4 waveshare, and a pi suger s 1200mAh battery
 - Any additional hardware used

**Additional context**
Add any other context about the problem here.

config.toml
main.name = "jellogotchi"
main.lang = "en"
main.whitelist = [
 "EXAMPLE_NETWORK",
 "ANOTHER_EXAMPLE_NETWORK",
 "fo:od:ba:be:fo:od",
 "fo:od:ba",
]
main.plugins.grid.enabled = true
main.plugins.grid.report = true
main.plugins.grid.exclude = [ "Plopcorn1",]

main.plugins.auto-update.enabled = true
main.plugins.auto-update.install = true
main.plugins.auto-update.interval = 1

main.plugins.net-pos.enabled = true
main.plugins.net-pos.api_key = "test"

main.plugins.gps.enabled = false
main.plugins.gps.speed = 19200
main.plugins.gps.device = "/dev/ttyUSB0"

main.plugins.webgpsmap.enabled = false

main.plugins.onlinehashcrack.enabled = true
main.plugins.onlinehashcrack.email = "blocked out"
main.plugins.onlinehashcrack.dashboard = ""
main.plugins.onlinehashcrack.single_files = false
main.plugins.onlinehashcrack.whitelist = []

main.plugins.wpa-sec.enabled = true
main.plugins.wpa-sec.api_key = "blocked out"
main.plugins.wpa-sec.api_url = "https://wpa-sec.stanev.org"
main.plugins.wpa-sec.download_results = true
main.plugins.wpa-sec.whitelist = []

main.plugins.wigle.enabled = true
main.plugins.wigle.api_key = "blocked out"
main.plugins.wigle.whitelist = []
main.plugins.wigle.donate = true

main.plugins.bt-tether.enabled = false
main.plugins.bt-tether.devices.android-phone.enabled = true
main.plugins.bt-tether.devices.android-phone.search_order = 1
[... 210 lines cut ...]
 "ble.device.connected",
 "ble.device.service.discovered",
 "ble.device.characteristic.discovered",
 "wifi.client.new",
 "wifi.client.lost",
 "wifi.client.probe",
 "wifi.ap.new",
 "wifi.ap.lost",
 "mod.started",
]

fs.memory.enabled = false
fs.memory.mounts.log.enabled = false
fs.memory.mounts.log.mount = "/var/log"
fs.memory.mounts.log.size = "50M"
fs.memory.mounts.log.sync = 60
fs.memory.mounts.log.zram = true
fs.memory.mounts.log.rsync = true

fs.memory.mounts.data.enabled = false
fs.memory.mounts.data.mount = "/var/tmp/pwnagotchi"
fs.memory.mounts.data.size = "10M"
fs.memory.mounts.data.sync = 3600
fs.memory.mounts.data.zram = false
fs.memory.mounts.data.rsync = true

jayofelony, 2024-04-26:
yep 2.9.1 was just removed as was 2.9.0, just revert back to 2.8.9

Sharkrider120, 2024-04-27:
i now downgraded to 2.8.9 and the screen works but now when i put it into auto mode it just shows New day, new hunt, new pwns! (v2. 8. 9)

Sharkrider120, 2024-04-27:
it also is just saying in the logs over and over again 

[2024-04-27 01:36:05,286] | [INFO] | starting monitor interface ...

Sharkrider120, 2024-04-27:
I fixed it i just ran the wizard again and now everything works

closed 2024-04-27 by Sharkrider120